projectacrn/acrn-hypervisor #7252: When runing board_inspector.py return "ERROR:MSR 0x1ad not support in this platform!"

Waiting-For-Wind, 2022-04-04:
# ERROR:MSR 0x1ad not support in this platform!
When I runing `sudo board_inspector.py my_board`. it is return following:
```
ERROR:MSR 0x1ad not support in this platform!
Command '[`/usr/bin/python3','/usr/bin/legacy/board_parser.py','my_board','--out','myboard.xml']' returned non-zeroexit status 1.
```
The  plantform info:
```
intel NUIC8i7BEH
CPU 8559U
RAM 8G

Ubuntu 20.04
kernel 5.13
```

And the content of the my_board.xml is:
```
<acrn-config board="my_board">
	<BIOS_INFO>
	BIOS Information
	Vendor: Intel Corp.
	Version: BECFL357.86A.0077.2019.1127.1452
	Release Date: 11/27/2019
	BIOS Revision: 5.6
	</BIOS_INFO>

	<BASE_BOARD_INFO>
	Base Board Information
	Manufacturer: Intel Corporation
	Product Name: NUC8BEB
	Version: J72688-308
	</BASE_BOARD_INFO>

	<PCI_DEVICE>
	00:00.0 Host bridge: Intel Corporation 8th Gen Core Processor Host Bridge/DRAM Registers (rev 08)
	00:02.0 VGA compatible controller: Intel Corporation Iris Plus Graphics 655 (rev 01)
	Region 0: Memory at a7000000 (64-bit, non-prefetchable) [size=16M]
	Region 2: Memory at 80000000 (64-bit, prefetchable) [size=256M]
	00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model
	Region 0: Memory at 4022c1a000 (64-bit, non-prefetchable) [disabled] [size=4K]
	00:12.0 Signal processing controller: Intel Corporation Cannon Point-LP Thermal Controller (rev 30)
	Region 0: Memory at 4022c19000 (64-bit, non-prefetchable) [size=4K]
	00:14.0 USB controller: Intel Corporation Cannon Point-LP USB 3.1 xHCI Controller (rev 30)
	Region 0: Memory at 4022c00000 (64-bit, non-prefetchable) [size=64K]
	00:14.2 RAM memory: Intel Corporation Cannon Point-LP Shared SRAM (rev 30)
	Region 0: Memory at 4022c14000 (64-bit, non-prefetchable) [disabled] [size=8K]
	Region 2: Memory at 4022c18000 (64-bit, non-prefetchable) [disabled] [size=4K]
	00:14.3 Network controller: Intel Corporation Cannon Point-LP CNVi [Wireless-AC] (rev 30)
	Region 0: Memory at 4022c10000 (64-bit, non-prefetchable) [size=16K]
	00:16.0 Communication controller: Intel Corporation Cannon Point-LP MEI Controller #1 (rev 30)
	Region 0: Memory at 4022c17000 (64-bit, non-prefetchable) [size=4K]
	00:17.0 SATA controller: Intel Corporation Cannon Point-LP SATA Controller [AHCI Mode] (rev 30)
	Region 0: Memory at a8a24000 (32-bit, non-prefetchable) [size=8K]
	Region 1: Memory at a8a27000 (32-bit, non-prefetchable) [size=256]
	Region 5: Memory at a8a26000 (32-bit, non-prefetchable) [size=2K]
	00:1c.0 PCI bridge: Intel Corporation Cannon Point-LP PCI Express Root Port #1 (rev f0)
	00:1c.4 PCI bridge: Intel Corporation Cannon Point-LP PCI Express Root Port #5 (rev f0)
	00:1d.0 PCI bridge: Intel Corporation Cannon Point-LP PCI Express Root Port #9 (rev f0)
	00:1d.6 PCI bridge: Intel Corporation Cannon Point-LP PCI Express Root Port #15 (rev f0)
	00:1f.0 ISA bridge: Intel Corporation Cannon Point-LP LPC Controller (rev 30)
	00:1f.3 Audio device: Intel Corporation Cannon Point-LP High Definition Audio Controller (rev 30)
	Region 0: Memory at a8a20000 (64-bit, non-prefetchable) [size=16K]
	Region 4: Memory at 4022b00000 (64-bit, non-prefetchable) [size=1M]
	00:1f.4 SMBus: Intel Corporation Cannon Point-LP SMBus Controller (rev 30)
	Region 0: Memory at 4022c16000 (64-bit, non-prefetchable) [size=256]
	00:1f.5 Serial bus controller [0c80]: Intel Corporation Cannon Point-LP SPI Controller (rev 30)
	Region 0: Memory at fe010000 (32-bit, non-prefetchable) [size=4K]
	00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (6) I219-V (rev 30)
	Region 0: Memory at a8a00000 (32-bit, non-prefetchable) [size=128K]
	3c:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS522A PCI Express Card Reader (rev 01)
	Region 0: Memory at a8000000 (32-bit, non-prefetchable) [size=4K]
	</PCI_DEVICE>

	<PCI_VID_PID>
	00:00.0 0600: 8086:3ed0 (rev 08)
	00:02.0 0300: 8086:3ea5 (rev 01)
	00:08.0 0880: 8086:1911
	00:12.0 1180: 8086:9df9 (rev 30)
	00:14.0 0c03: 8086:9ded (rev 30)
	00:14.2 0500: 8086:9def (rev 30)
	00:14.3 0280: 8086:9df0 (rev 30)
	00:16.0 0780: 8086:9de0 (rev 30)
	00:17.0 0106: 8086:9dd3 (rev 30)
	00:1c.0 0604: 8086:9db8 (rev f0)
	00:1c.4 0604: 8086:9dbc (rev f0)
	00:1d.0 0604: 8086:9db0 (rev f0)
	00:1d.6 0604: 8086:9db6 (rev f0)
	00:1f.0 0601: 8086:9d84 (rev 30)
	00:1f.3 0403: 8086:9dc8 (rev 30)
	00:1f.4 0c05: 8086:9da3 (rev 30)
	00:1f.5 0c80: 8086:9da4 (rev 30)
	00:1f.6 0200: 8086:15be (rev 30)
	3c:00.0 ff00: 10ec:522a (rev 01)
	</PCI_VID_PID>

	<WAKE_VECTOR_INFO>
	#define WAKE_VECTOR_32          0x6F3E608CUL
	#define WAKE_VECTOR_64          0x6F3E6098UL
	</WAKE_VECTOR_INFO>

	<RESET_REGISTER_INFO>
	#define RESET_REGISTER_ADDRESS  0xCF9UL
	#define RESET_REGISTER_SPACE_ID SPACE_SYSTEM_IO
	#define RESET_REGISTER_VALUE    0x6U
	</RESET_REGISTER_INFO>

	<PM_INFO>
	#define PM1A_EVT_SPACE_ID       SPACE_SYSTEM_IO
	#define PM1A_EVT_BIT_WIDTH      0x20U
	#define PM1A_EVT_BIT_OFFSET     0x0U
	#define PM1A_EVT_ADDRESS        0x1800UL
	#define PM1A_EVT_ACCESS_SIZE    0x2U
	#define PM1B_EVT_SPACE_ID       SPACE_SYSTEM_IO
	#define PM1B_EVT_BIT_WIDTH      0x0U
	#define PM1B_EVT_BIT_OFFSET     0x0U
	#define PM1B_EVT_ADDRESS        0x0UL
	#define PM1B_EVT_ACCESS_SIZE    0x2U
	#define PM1A_CNT_SPACE_ID       SPACE_SYSTEM_IO
	#define PM1A_CNT_BIT_WIDTH      0x10U
	#define PM1A_CNT_BIT_OFFSET     0x0U
	#define PM1A_CNT_ADDRESS        0x1804UL
	#define PM1A_CNT_ACCESS_SIZE    0x2U
	#define PM1B_CNT_SPACE_ID       SPACE_SYSTEM_IO
	#define PM1B_CNT_BIT_WIDTH      0x0U
	#define PM1B_CNT_BIT_OFFSET     0x0U
	#define PM1B_CNT_ADDRESS        0x0UL
	#define PM1B_CNT_ACCESS_SIZE    0x2U
	</PM_INFO>

	<S3_INFO>
	#define S3_PKG_VAL_PM1A         0x5U
	#define S3_PKG_VAL_PM1B         0U
	#define S3_PKG_RESERVED         0x0U
	</S3_INFO>

	<S5_INFO>
	#define S5_PKG_VAL_PM1A         0x7U
	#define S5_PKG_VAL_PM1B         0U
	#define S5_PKG_RESERVED         0x0U
	</S5_INFO>

	<DRHD_INFO>
	#define DRHD_COUNT              2U

	#define DRHD0_DEV_CNT           0x1U
	#define DRHD0_SEGMENT           0x0U
	#define DRHD0_FLAGS             0x0U
	#define DRHD0_REG_BASE          0xFED90000UL
	#define DRHD0_IGNORE            true
	#define DRHD0_DEVSCOPE0_TYPE    0x1U
	#define DRHD0_DEVSCOPE0_ID      0x0U
	#define DRHD0_DEVSCOPE0_BUS     0x0U
	#define DRHD0_DEVSCOPE0_PATH    0x10U

	#define DRHD1_DEV_CNT           0x2U
	#define DRHD1_SEGMENT           0x0U
	#define DRHD1_FLAGS             0x1U
	#define DRHD1_REG_BASE          0xFED91000UL
	#define DRHD1_IGNORE            false
	#define DRHD1_DEVSCOPE0_TYPE    0x3U
	#define DRHD1_DEVSCOPE0_ID      0x2U
	#define DRHD1_DEVSCOPE0_BUS     0x0U
	#define DRHD1_DEVSCOPE0_PATH    0xf7U
	#define DRHD1_DEVSCOPE1_TYPE    0x4U
	#define DRHD1_DEVSCOPE1_ID      0x0U
	#define DRHD1_DEVSCOPE1_BUS     0x0U
	#define DRHD1_DEVSCOPE1_PATH    0xf6U

	</DRHD_INFO>

	<CPU_BRAND>
	"Intel(R) Core(TM) i7-8559U CPU @ 2.70GHz"
	</CPU_BRAND>

	<CX_INFO>
	{{SPACE_FFixedHW, 0x00U, 0x00U, 0x00U, 0x00UL}, 0x01U, 0x01U, 0x00U},	/* C1 */
	{{SPACE_SYSTEM_IO, 0x08U, 0x00U, 0x00U, 0x1816UL}, 0x02U, 0x97U, 0x00U},	/* C2 */
	{{SPACE_SYSTEM_IO, 0x08U, 0x00U, 0x00U, 0x1819UL}, 0x03U, 0x40AU, 0x00U},	/* C3 */
	</CX_INFO>

	<PX_INFO>
```

fuzhongl, 2022-04-10:
@Waiting-For-Wind Could you please help to try with master branch?
Thanks!

fuzhongl, 2022-07-07:
Close it since no feedback long time.

closed 2022-07-07 by fuzhongl