projectacrn/acrn-hypervisor #7259: make configurator pollutes hypervisor repo code tree

dbkinder, 2022-04-05:
Running ``make configurator`` should not write to files outside of the build folder within the repo.  After running ``make configurator``
the ``scenario.json`` file is modified, polluting the checked out code tree:
```
~/projectacrn/acrn-hypervisor$ git status
On branch master
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   misc/config_tools/configurator/src/assets/schema/scenario.json

no changes added to commit (use "git add" and/or "git commit -a")
```
Looks like the problem is scenario.json is checked in with Windows line endings (CR LF):
```
00000000  7b 0d 0a 09 22 24 73 63  68 65 6d 61 22 3a 20 22  |{..."$schema": "|
```
but after running ``make configurator`` the file has Linux line endings (only LF):
```
00000000  7b 0a 09 22 24 73 63 68  65 6d 61 22 3a 20 22 68  |{.."$schema": "h|
```
The updated file is also missing a final line ending at the end of the file.

fuzhongl, 2022-04-07:
[External_System_ID] ACRN-8732

dbkinder, 2022-04-07:
Looks like there's another file being written to outside of the build folders.  ``misc/config_tools/configurator/src-tauri/tauri.conf.json`` is created by the build:
```
~/projectacrn/acrn-hypervisor$ git status
On branch master
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   misc/config_tools/configurator/src/assets/schema/scenario.json

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        misc/config_tools/configurator/src-tauri/tauri.conf.json

no changes added to commit (use "git add" and/or "git commit -a")
```

Weiyi-Feng, 2022-04-25:
`misc/config_tools/configurator/src-tauri/tauri.conf.json` cannot add to the `.gitignore` file, otherwise will build fail due to unknown reasons. 

the `scenario.json` might can be added to the `.gitignore` file.

gvancuts, 2022-04-25:
> `misc/config_tools/configurator/src-tauri/tauri.conf.json` cannot add to the `.gitignore` file, otherwise will build fail due to unknown reasons.
> 
> the `scenario.json` might can be added to the `.gitignore` file.

But that's hiding those files to `git`. Can we instead make sure these are not created in the tree (and put instead in the `build` folder for example)?

Weiyi-Feng, 2022-04-25:
> But that's hiding those files to `git`. Can we instead make sure these are not created in the tree (and put instead in the `build` folder for example)?

build tool will auto clean the `build` folder before build... instead, maybe we can add a `build_stage` dir to store these files?

gvancuts, 2022-04-25:
> build tool will auto clean the `build` folder before build... instead, maybe we can add a `build_stage` dir to store these files?

Which build tool, the `gen_acrn_deb.py` script? Is this something we could avoid?

Weiyi-Feng, 2022-04-25:
> > build tool will auto clean the `build` folder before build... instead, maybe we can add a `build_stage` dir to store these files?
> 
> Which build tool, the `gen_acrn_deb.py` script? Is this something we could avoid?

In the web build step, we use [vite](https://vitejs.dev/) (like webpack, another famous web build toolkit) to build out our webpage.
The `build.emptyOutDir` ([link](https://vitejs.dev/config/#build-emptyoutdir)) option seems might work, but if we disable it, we need to handle the empty out dir task by ourselves.

dbkinder, 2022-04-25:
Note that after the make, there are two problems here.  One existing file in the repo is modified by the build:   misc/config_tools/configurator/src/assets/schema/scenario.json, and one file is newly created: misc/config_tools/configurator/src-tauri/tauri.conf.json.