projectacrn/acrn-hypervisor #7265: About 20s after booting uaag with usb mediator,usb disk isn't recognized

liudlong, 2022-04-08:
Environment

HW/Board

ADL-S_CRB

Build link

http://sit-image.sh.intel.com/ubuntu/daily-build/2022ww03.5-1/ubuntu-adl-s-crb-shared/

Image info

sos kernel:5.10.83-acrn-service-vm-dirty

uaag kernel: 5.11.0-16-generic

uaag rootfs: http://sit-image.sh.intel.com/ubuntu_E2E/uos_21.04/zwl_uos_image/

DM version is: 2.8-unstable-ad7ccc28a (daily tag:acrn-2022w03.2-180000p), build by acrn@2022-01-14 00:22:43

HV 2.8-unstable-2022-01-14 00:22:21-ad7ccc28a DBG (daily tag: acrn-2022w03.2-180000p) shared@adl-s-crb build by acrn

Steps

1.insert a usb disk and boot sos

2. boot uaag with usb mediator

3. check usb disk

Expected result

input "lsblk" can find usb disk

Actual result

when booting uaag, can find usb disk  by inputting "lsblk", but after about 20s,can't recognize the usb disk

Reproduce rate

100%

Debugging info

use yaag rootfs to test,this issue is not reproduced

yaag rootfs:http://sit-image.sh.intel.com/yocto/e2e/5.10/yocto/ww44_bullpen_sos_uos_rt_65/uos/

liudlong, 2022-04-08:
[External_System_ID] ACRN-7590

NanlinXie, 2022-05-22:
Closed with no regression.

closed 2022-05-22 by NanlinXie