projectacrn/acrn-hypervisor #7291: [packaging] error while loading shared libraries: libSDL2-2.0.so.0

NanlinXie, 2022-04-16:
When I deploy the Debian packages to my target and try to launch the Windows VM, I can see an error message “acrn-dm: error while loading shared libraries: libSDL2-2.0.so.0: cannot open shared object file: No such file or directory”.

I can fix the error by manually installing the library by means of “sudo apt install libsdl2-2.0-0”. Shouldn’t this be installed automatically when I install the ACRN package?

NanlinXie, 2022-04-16:
[External_System_ID] ACRN-8721

dbkinder, 2022-04-17:
What Debian packages were you deploying? How were they built?

gvancuts, 2022-04-19:
We also need to update the tutorial on how to install [ACRN using a tarfile](https://projectacrn.github.io/latest/tutorials/tar_installation.html) to reflect the runtime dependencies, i.e. `libcjson1` and `libsdl2-2.0`

NanlinXie, 2022-06-05:
Close with no regression.

NanlinXie, 2022-06-05:
reopen to track document update

closed 2022-07-05 by fuzhongl