projectacrn/acrn-hypervisor #7301: [Virtio-GPU]Not enough free memory reserved in SOS

NanlinXie, 2022-04-18:
Test step:

1: BOOT to ACRN and SOS 

2.check memory in SOS found the free memory is only more than 900M, and when log in local UI , system will crash;

root@ubuntu20-04-adl-s-gpu:~# free -h
              total        used        free      shared  buff/cache   available
Mem:           30Gi        28Gi       956Mi        45Mi       511Mi       1.1Gi
Swap:         2.0Gi          0B       2.0Gi
root@ubuntu20-04-adl-s-gpu:~# sync

 __________________________

Expect result: enough free memory for used in SOS and no crash happened 

Actual Result: No enough free memory in SOS and that sometimes cause crash when login SOS local UI

Fail Rate: 100%

NanlinXie, 2022-04-18:
[External_System_ID] ACRN-8696

dbkinder, 2022-04-18:
Please use "Service VM" instead of "SOS" :)

NanlinXie, 2022-05-22:
Closed with no regression

closed 2022-05-22 by NanlinXie