projectacrn/acrn-hypervisor #7330: [PCI UART] Fail to build hypervisor without pci uart bdf value

junjiemao1, 2022-04-20:
HV tag:
acrn-2022w15.5-180000p

Steps:
1. Modify shared xml as below:

```
<vuart_connections>
<vuart_connection>
<name>vUART connection 1</name>
<type>pci</type>
<endpoint>
<vm_name>ACRN_Service_VM</vm_name>
<io_port>0x2F8</io_port>
</endpoint>
<endpoint>
<vm_name>POST_RT_VM1</vm_name>
<io_port>0x2F8</io_port>
</endpoint>
</vuart_connection>
<vuart_connection>
<name>vUART connection 2</name>
<type>pci</type>
<endpoint>
<vm_name>POST_RT_VM1</vm_name>
<io_port>0x2F8</io_port>
</endpoint>
<endpoint>
<vm_name>POST_STD_VM1</vm_name>
<io_port>0x2F8</io_port>
</endpoint>
</vuart_connection>
<vuart_connection>
<name>vUART connection 3</name>
<type>pci</type>
<endpoint>
<vm_name>POST_STD_VM1</vm_name>
<io_port>0x2F8</io_port>
</endpoint>
<endpoint>
<vm_name>POST_STD_VM2</vm_name>
<io_port>0x2F8</io_port>
</endpoint>
</vuart_connection>
</vuart_connections>
```

2. make hypervisor

Actual Result:
Compile should be successful.

Expected Result:
Getting bdf empty error.

Debug info:
Tried generating shared xml via UI without set bdf value, unfortunately no default bdf value was given.

junjiemao1, 2022-04-21:
The fix has been merged.

junjiemao1, 2022-05-30:
[External_System_ID] ACRN-8766

NanlinXie, 2022-06-05:
Close with no regression.

closed 2022-06-05 by NanlinXie