projectacrn/acrn-hypervisor #7342: Run ACRN Service VM is getting struck at WARNING: no console will be available to OS

shrikantuhallur, 2022-04-22:
**Describe the bug**
Launching ACRN Service OS is getting struck at loading ACRN kernel and displaying "WARNING: no console will be available to OS"

**Platform**
Dell Laptop
Processor: 11th Gen Intel Core i7 Tiger Lake
Graphics: Mesa Intel Xe Graphics (TGL GT2)

**Codebase**
Reference Link: https://projectacrn.github.io/latest/getting-started/getting-started.html
Code Branch: release_2.7
cd ~/acrn-work
git clone https://github.com/projectacrn/acrn-hypervisor.git
cd acrn-hypervisor
git checkout v2.7
cd ..
git clone --depth 1 --branch release_2.7 https://github.com/projectacrn/acrn-kernel.git

**Scenario**
Describe the scenario: Hybrid, Industry or logical_partition

**To Reproduce**
Follow steps as per link https://projectacrn.github.io/latest/getting-started/getting-started.html
Target Machine: Dell Laptop with Ubuntu 20.04 with 11th Gen Intel Core i7 Tiger Lake (As mentioned above in platform section)
Development Machine: Any machine with Ubuntu 20.04

**Expected behavior**
Expected to Launch ACRN Service OS.

**Additional context**
Attach the following logs if available:
my_board.xml
Senario:
shared.xml
shared_launch_6user_vm.xml
launch_user_vm_id3.sh

[my_board.xml.txt](https://github.com/projectacrn/acrn-hypervisor/files/8537227/my_board.xml.txt)
[shared.xml.txt](https://github.com/projectacrn/acrn-hypervisor/files/8537248/shared.xml.txt)
[shared_launch_6user_vm.xml.txt](https://github.com/projectacrn/acrn-hypervisor/files/8537249/shared_launch_6user_vm.xml.txt)
[launch_user_vm_id3.sh.txt](https://github.com/projectacrn/acrn-hypervisor/files/8537271/launch_user_vm_id3.sh.txt)

Note: Uploaded with .txt as not allowing me to upload with other extension.

fuzhongl, 2022-04-22:
@shrikantuhallur 
Using [shared.xml](https://github.com/projectacrn/acrn-hypervisor/files/8537248/shared.xml.txt); build will fail with v2.7.
`lib.error.ResourceError:  is not in the native environment! The ttyS available are: dict_keys(['ttyS4'])`

Build passed with following modification:
` <SERIAL_CONSOLE>/dev/ttyS4</SERIAL_CONSOLE>`

If ACRN still can't boot up, please share console log.
Thanks!

shrikantuhallur, 2022-04-22:
@fuzhongl Thank you for helping with this issue.

I was doing below change to resolve error "lib.error.ResourceError: is not in the native environment! The ttyS available are: dict_keys(['ttyS4'])"
misc/config_tools/static_allocators/lib/lib.py:  Change from ttys_n = ' ' to ttys_n = 'ttyS4' in parse_hv_console function.

I tried with shared.xml change as per your suggestion and still stuck at "WARNING: no console will be available to OS".

I do not have console logs and sharing requested results.
lspci -vvvvxxxx -> lspci-vvvvxxxx.txt
cat  /proc/iomem -> iomem.txt
dmesg -> dmesg.txt
lspci -t -> lspci-t.txt

[lspci-t.txt](https://github.com/projectacrn/acrn-hypervisor/files/8541290/lspci-t.txt)
[lspci-vvvvxxxx.txt](https://github.com/projectacrn/acrn-hypervisor/files/8541291/lspci-vvvvxxxx.txt)
[dmesg.txt](https://github.com/projectacrn/acrn-hypervisor/files/8541292/dmesg.txt)
[iomem.txt](https://github.com/projectacrn/acrn-hypervisor/files/8541294/iomem.txt)

fuzhongl, 2022-04-23:
@shrikantuhallur 
One potential issue about  memory size, there is 32G memory on your board.
Please help to add "nokaslr" in grub menu for workaround: 
`multiboot2 /boot/acrn.shared.my_board.bin root=PARTUUID=073352d9-588e-4045-80da-c9ce39ae3384 **nokaslr**`
If the system can boot up with this kernel cmdline, you can update the shared.xml as following for formal fix:
```
From
      <PLATFORM_RAM_SIZE>0x400000000</PLATFORM_RAM_SIZE> 
To
      <PLATFORM_RAM_SIZE>0x800000000</PLATFORM_RAM_SIZE>
The 
```
If the system still struck with "nokaslr" in kernel cmdline, please share following log again with root right.
```
sudo lspci -vvvvxxxx 
sudo cat /proc/iomem
```
Thanks!

shrikantuhallur, 2022-04-25:
@fuzhongl Thank you for your suggestion.
I tried with "nokaslr" and hitting the same issue.

grub.cfg changes with addition of **nokaslr**,
menuentry 'ACRN multiboot2 ' --id ACRN_deb_multiboot2 {
    load_video
    insmod gzio
    insmod part_gpt
    insmod ext2
    search --no-floppy --fs-uuid  --set a6c0ff99-ae09-4dda-8370-8e42748375eb
multiboot2 /boot/acrn.shared.my_board.bin root=PARTUUID=073352d9-588e-4045-80da-c9ce39ae3384 **nokaslr**
module2 /boot/vmlinuz-5.10.78-acrn-service-vm Linux_bzImage

}

Sharing requested command output.
sudo lspci -vvvvxxxx  -> [lspci-vvvvxxxx.txt](https://github.com/projectacrn/acrn-hypervisor/files/8552022/lspci-vvvvxxxx.txt)
sudo cat /proc/iomem -> [iomem.txt](https://github.com/projectacrn/acrn-hypervisor/files/8552023/iomem.txt)

fuzhongl, 2022-04-25:
@shrikantuhallur Thanks for sharing the log!
Please help to modify shared.xml as following:
From
```
<PLATFORM_RAM_SIZE>0x400000000</PLATFORM_RAM_SIZE> 
<ENFORCE_TURNOFF_GP>n</ENFORCE_TURNOFF_GP>
```
To
```
<PLATFORM_RAM_SIZE>0x800000000</PLATFORM_RAM_SIZE> 
<ENFORCE_TURNOFF_GP>y</ENFORCE_TURNOFF_GP>
```
```
make clean 
make BOARD=xxx SCENARIO=xxx
```
If system still can't boot up, please share the below file:
`build/hypervisor/configs/boards/board_info.h`

Thanks!

shrikantuhallur, 2022-04-25:
@fuzhongl Thank you for your quick response.
I used modified shared.xml and struck at same point.

I tried, 
1. Modified shared.xml ([shared.xml.txt](https://github.com/projectacrn/acrn-hypervisor/files/8553305/shared.xml.txt)) on development machine
2. https://projectacrn.github.io/latest/getting-started/getting-started.html#build-acrn -> All success and copied required packages from development machine to target machine using USB
4. On target machine, https://projectacrn.github.io/latest/getting-started/getting-started.html#install-acrn

Please let know if i am missing something in building and launching process.

Sharing build/hypervisor/configs/boards/board_info.h ([board_info.h.txt](https://github.com/projectacrn/acrn-hypervisor/files/8553342/board_info.h.txt)) file.

fuzhongl, 2022-04-26:
> @fuzhongl Thank you for your quick response. I used modified shared.xml and struck at same point.
> 
> I tried,
> 
> 1. Modified shared.xml ([shared.xml.txt](https://github.com/projectacrn/acrn-hypervisor/files/8553305/shared.xml.txt)) on development machine
> 2. On development machine, https://projectacrn.github.io/latest/getting-started/getting-started.html#build-acrn -> All success and copied required packages from development machine to target machine using USB
> 3. On target machine, https://projectacrn.github.io/latest/getting-started/getting-started.html#install-acrn
> 
> Please let know if i am missing something in building and launching process.
> 
> Sharing build/hypervisor/configs/boards/board_info.h ([board_info.h.txt](https://github.com/projectacrn/acrn-hypervisor/files/8553342/board_info.h.txt)) file.

@shrikantuhallur 
From the dmesg log; it is pci uart on your board.
`[    0.941217] 0000:00:16.3: ttyS4 at I/O 0x3060 (irq = 19, base_baud = 115200) is a 16550A`
But there is no bdf info in board.xml:
```
  <TTYS_INFO>
	seri:/dev/ttyS4 type:portio base:0x3060 irq:19
	</TTYS_INFO>
```
Please help to try with adding **uart=disabled** in grub menu as following(acrn.bin is build with the modified xml) :
`multiboot2 /boot/acrn.shared.my_board.bin root=PARTUUID=073352d9-588e-4045-80da-c9ce39ae3384 uart=disabled`

Thanks!

shrikantuhallur, 2022-04-26:
@fuzhongl Thank you for all your suggestions.

I tried adding **uart=disabled** and hitting same issue.
"multiboot2 /boot/acrn.shared.my_board.bin root=PARTUUID=073352d9-588e-4045-80da-c9ce39ae3384 uart=disabled" to "/boot/grub/grub.cfg"

Please let me know any specific hardware required to add/remove to board.

gvancuts, 2022-04-26:
> multiboot2 /boot/acrn.shared.my_board.bin root=PARTUUID=073352d9-588e-4045-80da-c9ce39ae3384 nokaslr

Just double-checking here, you did not have the `nokaslr` enclosed in asterisks (or double-asterisks), did you?

fuzhongl, 2022-04-26:
@shrikantuhallur Is there console interface on your device?
Please refer following doc about how to get the console log.
https://projectacrn.github.io/latest/tutorials/using_serial_port.html#option-1-using-the-serial-port-on-tgl-intel-nuc

shrikantuhallur, 2022-04-26:
> > multiboot2 /boot/acrn.shared.my_board.bin root=PARTUUID=073352d9-588e-4045-80da-c9ce39ae3384 nokaslr
> 
> Just double-checking here, you did not have the `nokaslr` enclosed in asterisks (or double-asterisks), did you?

@gvancuts Thank you for your observation.

The "nokaslr" not required as we added 0x800000000 to PLATFORM_RAM_SIZE. Anyway adding nokaslr along with uart=disabled is also not working.

shrikantuhallur, 2022-04-26:
> @shrikantuhallur Is there console interface on your device? Please refer following doc about how to get the console log. https://projectacrn.github.io/latest/tutorials/using_serial_port.html#option-1-using-the-serial-port-on-tgl-intel-nuc

@fuzhongl Thank you for reference.

The console interface not present on this device. I will try to check with other machine.

Please let me know any other workarounds to try in parallel.

dbkinder, 2022-04-27:
After a bit of research, the message you're referring to is actually written by the [GRUB2 Multiboot Boot Loader](https://fossies.org/dox/grub-2.06/multiboot_8c_source.html) and not from the ACRN hypervisor.  I found a discussion stating: 

> "The message "WARNING: no console will be available to OS" appears because UEFI does not support text-mode graphics, which was a graphics mode supported by legacy BIOS. This issue should only prevent messages from being displayed on the screen until the kernel configures the graphics hardware. This issue should not prevent the system from booting."

fuzhongl, 2022-04-27:
> @fuzhongl Thank you for reference.
>
> The console interface not present on this device. I will try to check with other machine.
>
> Please let me know any other workarounds to try in parallel.

@shrikantuhallur Could you please help to re-generate your board xml with the latest master code?
Just to double check: Is security boot disabled in BIOS?
Thanks!

shrikantuhallur, 2022-04-27:
> After a bit of research, the message you're referring to is actually written by the [GRUB2 Multiboot Boot Loader](https://fossies.org/dox/grub-2.06/multiboot_8c_source.html) and not from the ACRN [...]

@dbkinder Thank you for your response.

I came across this article. The issue is preventing the system from booting (waited for more than 1hour with message WARNING: no console will be available to OS) and legacy boot mode is not supported on most of latest machines.

I assume ACRN hypervisor is supporting UEFI boot mode and working with GRUB2 multiboot2.

shrikantuhallur, 2022-04-27:
> @shrikantuhallur Could you please help to re-generate your board xml with the latest master code? Just to double check: Is security boot disabled in BIOS? Thanks!

@fuzhongl I regenerated board.xml ([my_board.xml.txt](https://github.com/projectacrn/acrn-hypervisor/files/8571171/my_board.xml.txt)) with master code with warnings.
Sniff,
WARNING:root:The PCI device at 00:02.0 has SR-IOV capability which is currently disabled.
WARNING:root:If that device is intended to provide virtual functions in your scenario, write (as root) the number of VFs to /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs and rerun the board inspector.
SUCCESS: Board configuration file my_board.xml generated successfully and saved to /home/ubuntu/acrn-work

But, I am unable to launch ACRN Configurator with master code. Hitting error, ImportError: cannot import name 'get_launch_item_values' from 'launch_config.launch_cfg_gen'.

gvancuts, 2022-04-27:
> I came across this article. The issue is preventing the system from booting (waited for more than 1hour with message WARNING: no console will be available to OS) and legacy boot mode is not supported on most of latest machines.

This is not the problem. We see this all the time, it happens to be the last thing to be printed on the monitor but that's because ACRN cannot print any of its messages there, they are sent to the console but we're blind there because we do not have one. I _think_ there were talks to also have a very basic framebuffer driver in ACRN to be able to print such early messages on the monitor but I don't think we ever did it.

> 
> I assume ACRN hypervisor is supporting UEFI boot mode and working with GRUB2 multiboot2.

Yes!

shrikantuhallur, 2022-04-27:
@gvancuts Thank you. It will be helpful if we get print messages with framebuffer support to ACRN.
In my case, its not launching Service OS and we are not sure what is happening with loading ACRN kernel. We see only message "WARNING: no console will be available to OS".

Any help with launching Service OS is appreciated.

fuzhongl, 2022-04-28:
> > > > @shrikantuhallur Is there console interface on your device? Please refer following doc about how to get the console log. https://projectacrn.github.io/latest/tutorials/using_serial_port.html#option-1-using-the-serial-port-on-tgl-intel-nuc
> > > 
> > > 
> > > @fuzhongl Thank you for reference.
> > > The console interface not present on this device. I will try to check with other machine.
> > > Please let me know any other workarounds to try in parallel.
> > 
> > 
> > @shrikantuhallur Could you please help to re-generate your board xml with the latest master code? Just to double check: Is security boot disabled in BIOS? Thanks!
> 
> @fuzhongl Yes, Secure boot is disabled. I regenerated board.xml ([my_board.xml.txt](https://github.com/projectacrn/acrn-hypervisor/files/8571171/my_board.xml.txt)) with master code with warnings. Sniff, WARNING:root:The PCI device at 00:02.0 has SR-IOV capability which is currently disabled. WARNING:root:If that device is intended to provide virtual functions in your scenario, write (as root) the number of VFs to /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs and rerun the board inspector. SUCCESS: Board configuration file my_board.xml generated successfully and saved to /home/ubuntu/acrn-work
> 
> But, I am unable to launch ACRN Configurator with master code. Hitting error, ImportError: cannot import name 'get_launch_item_values' from 'launch_config.launch_cfg_gen'.

@shrikantuhallur I saw hyper-thread is disabled for your regenerated board.xml:
```
 <CPU_PROCESSOR_INFO>
	0, 1, 2, 3
	</CPU_PROCESSOR_INFO>
```
But  hyper-thread is enabled for your original board.xml:
```
 <CPU_PROCESSOR_INFO>
	0, 1, 2, 3, 4, 5, 6, 7
	</CPU_PROCESSOR_INFO>
```
Do you generate the board.xml on the same board with hyper-thread modification on purpose?
Or someone changed the hyper-thread setting after you generate the original board.xml?

ACRN can't boot up if hyper-thread setting is changed.
Please make sure not to change BIOS setting after generate the board.xml; otherwise you need to regenerated board.xml.

shrikantuhallur, 2022-04-28:
@fuzhongl Thank you for pointing the change. I played with BIOS settings after board.xml generated.

I regenerated board.xml ([my_board.xml.txt](https://github.com/projectacrn/acrn-hypervisor/files/8580819/my_board.xml.txt)) with below BIOS settings:
Boot Mode: UEFI only (Legacy Boot mode is not supported on this platform)
Secure Boot -> Disabled
Intel Virtualization Technology (VT) -> Enabled
VT for Direct I/O -> Enabled
Intel Trusted Execution Technology (TXT) -> Disabled
Active Cores -> All Cores
Intel SpeedStep -> Enabled
C-States Control -> Disabled
Intel Turbo Boost Technology -> Enabled
Intel Hyper-Threading Technology -> Disabled
Intel GNA Accelerator -> Disabled
Intel Speed Shift Technology -> Disabled

Generated a Scenario Configuration File and Launch Script with above board.xml:
launch_user_vm_id3.sh ([launch_user_vm_id3.sh.txt](https://github.com/projectacrn/acrn-hypervisor/files/8580815/launch_user_vm_id3.sh.txt))
[shared_launch_6user_vm.xml.txt](https://github.com/projectacrn/acrn-hypervisor/files/8580818/shared_launch_6user_vm.xml.txt)

Also, modified shared.xml ([shared.xml.txt](https://github.com/projectacrn/acrn-hypervisor/files/8581142/shared.xml.txt))
from,
<SERIAL_CONSOLE></SERIAL_CONSOLE>
<PLATFORM_RAM_SIZE>0x400000000</PLATFORM_RAM_SIZE> 
<ENFORCE_TURNOFF_GP>n</ENFORCE_TURNOFF_GP>
to,
<SERIAL_CONSOLE>/dev/ttyS4</SERIAL_CONSOLE>
<PLATFORM_RAM_SIZE>0x800000000</PLATFORM_RAM_SIZE> 
<ENFORCE_TURNOFF_GP>y</ENFORCE_TURNOFF_GP>

ACRN built with above shared.xml and boad.xml and generated acrn-my_board-shared-2.7.deb.

Installation of ACRN acrn-my_board-shared-2.7.deb successful.
Log sniff,
You are in ubuntu, let's pre-check whether fit acrn needs
CPU cores config is ok
vt-x/vt-d config is ok
cpu type is ok
memory is ok, 32605184

Modified grub configuration from "multiboot2 /boot/acrn.shared.my_board.bin root=PARTUUID=073352d9-588e-4045-80da-c9ce39ae3384" to "multiboot2 /boot/acrn.shared.my_board.bin root=PARTUUID=073352d9-588e-4045-80da-c9ce39ae3384 nokaslr uart=disabled"

/boot/grub/grub.cfg ([grub.cfg.txt](https://github.com/projectacrn/acrn-hypervisor/files/8581121/grub.cfg.txt))

Same behavior with all above changes (**Additional observation**, shown "WARNING: no console will be available to OS" after 9sec came to ubuntu login screen and struck with ubuntu login screen even no mouse moment possible). 

Note: Tried with ACRN v2.7 as master giving errors.

fuzhongl, 2022-04-28:
> @fuzhongl Thank you for pointing the change. I played with BIOS settings after board.xml generated.
> 
> I regenerated board.xml ([my_board.xml.txt](https://github.com/projectacrn/acrn-hypervisor/files/8580819/my_board.xml.txt)) on target machine with below BIOS settings: Boot Mode: UEFI only (Legacy Boot mode is not supported on this platform) Secure Boot -> Disabled Intel Virtualization Technology (VT) -> Enabled VT for Direct I/O -> Enabled Intel Trusted Execution Technology (TXT) -> Disabled Active Cores -> All Cores Intel SpeedStep -> Enabled C-States Control -> Disabled Intel Turbo Boost Technology -> Enabled Intel Hyper-Threading Technology -> Disabled Intel GNA Accelerator -> Disabled Intel Speed Shift Technology -> Disabled
> 
> Generated a Scenario Configuration File and Launch Script with above board.xml on development machine: launch_user_vm_id3.sh ([launch_user_vm_id3.sh.txt](https://github.com/projectacrn/acrn-hypervisor/files/8580815/launch_user_vm_id3.sh.txt)) [shared_launch_6user_vm.xml.txt](https://github.com/projectacrn/acrn-hypervisor/files/8580818/shared_launch_6user_vm.xml.txt) Modified shared.xml ([shared.xml.txt](https://github.com/projectacrn/acrn-hypervisor/files/8581142/shared.xml.txt)) from, <SERIAL_CONSOLE></SERIAL_CONSOLE> <PLATFORM_RAM_SIZE>0x400000000</PLATFORM_RAM_SIZE> <ENFORCE_TURNOFF_GP>n</ENFORCE_TURNOFF_GP> to, <SERIAL_CONSOLE>/dev/ttyS4</SERIAL_CONSOLE> <PLATFORM_RAM_SIZE>0x800000000</PLATFORM_RAM_SIZE> <ENFORCE_TURNOFF_GP>y</ENFORCE_TURNOFF_GP>
> 
> ACRN built with above shared.xml and boad.xml on development machine and generated acrn-my_board-shared-2.7.deb.
> 
> Installation of ACRN acrn-my_board-shared-2.7.deb is successful on target machine. Log sniff, You are in ubuntu, let's pre-check whether fit acrn needs CPU cores config is ok vt-x/vt-d config is ok cpu type is ok memory is ok, 32605184
> 
> Modified grub configuration on target machine from "multiboot2 /boot/acrn.shared.my_board.bin root=PARTUUID=073352d9-588e-4045-80da-c9ce39ae3384" to "multiboot2 /boot/acrn.shared.my_board.bin root=PARTUUID=073352d9-588e-4045-80da-c9ce39ae3384 nokaslr uart=disabled"
> 
> /boot/grub/grub.cfg ([grub.cfg.txt](https://github.com/projectacrn/acrn-hypervisor/files/8581121/grub.cfg.txt))
> 
> Same behavior with all above changes (**Additional observation**, shown "WARNING: no console will be available to OS" after 9sec came to ubuntu login screen and struck with ubuntu login screen even no mouse moment possible).
> 
> Note: Tried with ACRN v2.7 as master giving errors.

@shrikantuhallur 
You mean you changed Hyper-Threading to disabled after generate the original board.xml, right?
`Intel Hyper-Threading Technology -> Disabled`

If yes, Please set Hyper-Threading to enabled; and help to try again(with the acrn.bin build with above modification).

Thanks!

fuzhongl, 2022-04-28:
@shrikantuhallur 
Sorry for not aware below info:
`Additional observation, shown "WARNING: no console will be available to OS" after 9sec came to ubuntu login screen and struck with ubuntu login screen even no mouse moment possible`
ACRN boot successfully now, maybe USB module isn't copied to your rootfs.
Please try your native kernel instead of ACRN kernel:
```
menuentry 'ACRN multiboot2 ' --id ACRN_deb_multiboot2 {
    load_video
    insmod gzio
    insmod part_gpt
    insmod ext2
    search --no-floppy --fs-uuid  --set a6c0ff99-ae09-4dda-8370-8e42748375eb
multiboot2 /boot/acrn.shared.my_board.bin root=PARTUUID=073352d9-588e-4045-80da-c9ce39ae3384
module2 /boot/vmlinuz-5.10.78-acrn-service-vm Linux_bzImage
```
change **vmlinuz-5.10.78-acrn-service-vm**  to your native kernel.
From dmesg log; your native kernel is 5.14.0-1033-oem.
`[    0.000000] Linux version 5.14.0-1033-oem`
Thanks!

shrikantuhallur, 2022-04-28:
@fuzhongl Hurrah. Its working. Able to launch Service OS and User OS.

Thank you so much for root causing this issue. I really appreciate your help.

fuzhongl, 2022-04-29:
> @fuzhongl Hurrah. Its working. Able to launch Service OS and User OS.
> 
> Thank you so much for root causing this issue. I really appreciate your help.

@shrikantuhallur Glad to know Service OS and User OS boot up successfully on your side.
Will close this  git issue.
Any more questions, please don't hesitate to raise new git issue!
Thanks!

closed 2022-04-29 by fuzhongl

gvancuts, 2022-04-29:
Great news - thanks very much for your help @fuzhongl !

shrikantuhallur, 2022-05-04:
@fuzhongl Coming back with Yocto boot issue.
Hitting the same problem with yocto.
Reference: https://github.com/intel/meta-acrn/blob/master/docs/getting-started.md
Problem: Booting to yocto Image with ACRN (Yocto) is struck at "ARNING: no console will be available to OS" and able to boot to ('boot' to boot as) normal Linux

Grub version in normal Linux of yocto: grub-install (GRUB) 2.06
Yocto grub: [grub.cfg.txt](https://github.com/projectacrn/acrn-hypervisor/files/8621671/grub.cfg.txt)

fuzhongl, 2022-05-05:
@shrikantuhallur Please help to try with following grub menu:

```
menuentry 'ACRN (Yocto)'{
multiboot2 /acrn.bin root=PARTUUID=981dd5b3-235a-4258-8225-1a84399ba9fe nokaslr uart=disabled
search --set=mpath --file /bzImage 
set modbin=($mpath)/bzImage 
module2 $modbin Linux_bzImage  rootwait console=ttyS0,115200 console=tty0   i915.enable_gvt=1 i915.nuclear_pageflip=1 

}
```

Thanks!

shrikantuhallur, 2022-05-05:
@fuzhongl Thank you for suggestion. This was also tried and same result.

Do we need to check config (board or shared xml) or BIOS settings in Yocto like Linux?

fuzhongl, 2022-05-05:
@shrikantuhallur I suppose there are the same config(board.xml/shared.xml) and BIOS setting for Linux and Yocto.
Do you try Yocto on the different board?
If yes, Could you please try Yocto on the same board first with the same acrn.bin and bzImage used by Linux ?
Thanks!

shrikantuhallur, 2022-05-05:
@fuzhongl Yocto, we are not passing any config (board.xml/shared.xml) manually.
I am trying on the same target machine with same BIOS settings where Linux ACRN working.

I built yocto image (.wic) on the same target machine and burnt that image to USB and booted to USB on the same machine . USB having EFI (grub and bootx64), acrn.bin, bzImage and microcode.cpio.

I also tried replacing acrn.bin and bzImage with Linux /boot/acrn.shared.my_board.bin and /boot/vmlinux-5.10.78-acrn-service-vm respectively.

Result: WARNING: no console will be available to OS (I observe fans are running faster - some processing happening)

fuzhongl, 2022-05-06:
> I also tried replacing acrn.bin and bzImage with Linux /boot/acrn.shared.my_board.bin and /boot/vmlinux-5.10.78-acrn-service-vm respectively.
> 
> Result: WARNING: no console will be available to OS (I observe fans are running faster - some processing happening)
@shrikantuhallur Could you help to replace the acrn.bin/ bzImage/rootfs(verified with Linux) together ?
```
menuentry 'ACRN (Yocto)'{

multiboot2 /acrn.bin root=PARTUUID=073352d9-588e-4045-80da-c9ce39ae3384 nokaslr uart=disabled
search --set=mpath --file /bzImage 
set modbin=($mpath)/bzImage 
module2 $modbin Linux_bzImage  

}
```
With the same acrn.bin/ bzImage/rootfs and kernel cmdline setting , the only difference between Yocto and Linux is grub. 
Base on the test result, we can further narrow down this Yocto related issue.
Thanks!

fuzhongl, 2022-05-10:
> @fuzhongl Same behavior. I copied binaries from /boot folder to USB with grub edit as suggested above. Am i missing something? Please suggested from where I need to copy required binaries.

@shrikantuhallur Could you please boot into Yocto native? And check the `md5sum` of the copied acrn.bin/ bzImage.
Thanks!

shrikantuhallur, 2022-05-11:
@fuzhongl I can boot to yocto native (Linux) with Yocto acrn kernel and Linux acrn kernel but not able to boot to Yocto ACRN.