projectacrn/acrn-hypervisor #7361: config_tool: Add check for RTVM pCPU assignment

yfliuuu, 2022-04-26:
For pre-launched RTVM, config_tool will check its pCPU assignment and make sure they don't conflict with other VMs. We need to add this check for post-launched RTVM too.

yfliuuu, 2022-04-26:
[External_System_ID] ACRN-8833

Pirouf, 2022-05-11:
What is the rational including post-launched RTVM CPU affinity to be fixed at buildtime?
Why wouldn't an system admin use SOS runtime launch-script to define which potential 4 or 6 or 8 RT-capable CPU core the RTVM will be assigned to?

junjiemao1, 2022-05-22:
> What is the rational including post-launched RTVM CPU affinity to be fixed at buildtime? Why wouldn't an system admin use SOS runtime launch-script to define which potential 4 or 6 or 8 RT-capable CPU core the RTVM will be assigned to?

Physical CPU assignment is a pre-requisite of a proper programing to the physical CAT which is statically defined at configuration phase. This is because there are multiple CAT-capable L2 caches on platforms with hybrid cores, and knowing the physical CPU assignment is crucial in determining which L2 cache(s) to use for a specific RTVM.

Pirouf, 2022-05-23:
> Physical CPU assignment is a pre-requisite of a proper programing to the physical CAT which is statically defined at configuration phase. This is because there are multiple CAT-capable L2 caches on [...]

i do acknowledge and agree  _knowing the physical CPU assignment is crucial in determining which L2 cache(s) to use for a specific RTVM_ but i am afraid the explanation is either incomplete or misleading.

Neither [Intel TCC tools](https://www.intel.com/content/www/us/en/developer/articles/technical/cache-allocation-for-real-time-systems.html) nor [Intel RDT](https://www.intel.com/content/www/us/en/architecture-and-technology/resource-director-technology.html) enforces a ACRN buildtime pre-requisite for Post-Launch RTVM `shared` scenario CPU affinity. 

You may desire to fix this as a pre-requisite for ARCN configurator UI usecase simplicity, but I am afraid it defeats the purpose of DX where an admin persona can as effectively determine hybrid-cpus partitioning directly from the Service OS runtime using purposely made CAT L3/L2 Linux UAPI to Get/Set/Reset affinity between physical-CPU and RTVM (ie. without requiring a full acrn-hypervisor debian package rebuild) 

Please do revert this config_tool check that introduces an artificial `Shared` scenario limitation and confusion for user/admin to differentiate RTVM pre-requisite between `Shared`  -vs- `Partitionning`/`Hybrid`.
Thanks

junjiemao1, 2022-05-23:
This is a design decision of ACRN, not something derived from TCC tools or RDT.

Today in ACRN the hypervisor takes a static configuration of CLOS_ID and CLOS_MASK for each vCPU. Such configurations are calculated at build time by the configuration toolset based on the cache way allocation policies a user defines. To allocate certain L2 cache ways to a vCPU it is not sufficient to specify `give # L2 ways to vCPU X` because there can be multiple L2 cache blocks. Instead one should specify `give # L2 ways to vCPU X running on pCPU Y`, where `pCPU Y` can be used to derive the L2 cache block that `vCPU X` will consume.

There is currently no way for a service VM to specify the cache allocation policies of a VM, because RDT is hidden from the service VM and no hypercall for that purposes is yet defined. Making cache allocation dynamic is a topic that should be considered as a new feature to be planned separately for a future release, and before that is implemented, this check must be there to alert ill-formed configurations at an early stage. Removing this check to allow one to dynamically reset CPU affinity could lead to settings contradicting with cache allocation policies and ruin RT guarantees.

To what extent this design decision hurts DX is something to be evaluated by user studies. It did not arise as a top issue in previous ones, but could be in the future.

junjiemao1, 2022-05-24:
The flexibility (esp. CPU affinity and cache allocation) of post-launched RT VMs in a shared scenario is a topic we can reevaluate in future user studies. What we learned from our previous support for users from industry segment is that static configurations of VM resources are preferable due to its determinism, and the service VM is typically used to control the lifecycle of those VMs rather than reallocating resources.

Of course if full control of the service VM over resources of post-launched RT VMs is useful in some other use cases, we are open to understand its priority, evaluate its impact and plan it accordingly.

With regard to the abstraction, ACRN did expose the CLOS_ID and CLOS_MASK for users to config in previous releases, but that turns out not to be easy to get started during our recent DX reviews. In those releases users have to provide the CLOS_MASK for each CLOS_ID for each block of cache and a mapping from vCPUs to those CLOS_IDs. That complexity motivates the reduction of the information users need to provide: a mapping from vCPUs to CLOS_MASK for each block of cache is what users really care about. CLOS_IDs are less important as long as they are sufficient.

Tweaking the CAT programming via the command line options of the hypervisor looks a viable option to avoid a rebuild to me. In order to fully understand its impact on DX, however, further analysis is still needed during DX reviews in which other aspects such as learning curve and complexity will be taken into account as well.

Loop @NanlinXie who can help track this.

NanlinXie, 2022-06-05:
Close with no regression.

closed 2022-06-05 by NanlinXie