projectacrn/acrn-hypervisor #7366: [acrn-configuration tool] make scenario shared file error cp error

zihengL1, 2022-04-26:
Steps:
1. make step：
cd /git/work/acrn-hypervisor
make BOARD=adl-s-crb SCENARIO=shared

2. The error message is as follows:

Build directory: /git/work/acrn-hypervisor/build/hypervisor
This build directory is configured with the settings below.

BOARD = adl-s-crb
SCENARIO = shared
RELEASE = n
cp: cannot stat '/git/work/acrn-hypervisor/build/../build/hypervisor/acpi/ACPI_VM0.bin': No such file or directory
cp: cannot stat '/git/work/acrn-hypervisor/build/../build/hypervisor/acpi/ACPI_VM1.bin': No such file or directory
cp: cannot stat '/git/work/acrn-hypervisor/build/../build/hypervisor/serial.conf': No such file or directory
chmod: cannot access './build/acrn_release_deb/DEBIAN/preinst': No such file or directory
sed: can't read ./build/acrn_release_deb/DEBIAN/preinst: No such file or directory
dpkg-deb: building package 'acrn-hypervisor' in 'acrn-adl-s-crb-shared-3.0-unstable.deb'.
ACRN debian package acrn-adl-s-crb-shared-3.0-unstable.deb was successfully created in the /git/work/acrn-hypervisor/build.
3. This commit causes the above build problems :

commit : "3c4f14ede70134e998e9c583b86c0e8d557a3e11

zihengL1, 2022-04-26:
[External_System_ID] ACRN-8705

NanlinXie, 2022-05-22:
Closed with no regression.

closed 2022-05-22 by NanlinXie