projectacrn/acrn-hypervisor #7367: New target in `gen_acrn_deb.py` missing from error message

gvancuts, 2022-04-26:
There is a new target that's been added to the [gen_acrn_deb.py](https://github.com/projectacrn/acrn-hypervisor/blob/master/misc/packaging/) script but is missing from the error message - see:

```
    if args.deb_mode == 'board_inspector':
        create_acrn_board_inspector_deb(args.version, args.build_dir)
    elif args.deb_mode == 'acrn_all':
        create_acrn_deb(args.board_name, args.scenario, args.version, args.build_dir)
    elif args.deb_mode == 'configurator':
        create_configurator_deb(args.build_dir)
    else:
        print("ERROR: Please check the value of deb_mode: the value shall be acrn_all or board_inspector.")
```

gvancuts, 2022-04-26:
[External_System_ID] ACRN-8843

NanlinXie, 2022-10-31:
Closed with no regression

closed 2022-10-31 by NanlinXie