projectacrn/acrn-hypervisor #7370: [acrn-deb] install_compile_package function is not consistent with gsg

zihengL1, 2022-04-26:
HW/Board

adl-s-crb

 

Build link

hv_repo_commit: c61d41da2717762c4c7b5cb4a683a1865e112d24

 

Expected result

misc/packaging/install_uSoS.py has the following dependencies:

 

vim \
libxml2-utils \
libcjson-dev \
liblz4-tool \
flex \
bison \
xsltproc \
clang-format \
bc \
libpixman-1-dev  \
libsdl2-dev \
libegl-dev \
libgles-dev
 
 

Actual result

This dependency installation does not exist in the misc/packaging/install_uSoS.py file install_compile_package usage:

 

vim \
libxml2-utils \
libcjson-dev \
liblz4-tool \
flex \
bison \
xsltproc \
clang-format \
bc \
libpixman-1-dev  \
libsdl2-dev \
libegl-dev \
libgles-dev
 

Reproduce rate

100%

 

Debugging info

root@acrn-System-Product-Name:/git/work/master_junjie/acrn-hypervisor# vim misc/packaging/install_uSoS.py --# install compile package
def install_compile_package():
    #check compile env
    os.system('sudo apt install gcc \
     git \
     make \
     gnu-efi \
     libssl-dev \
     libpciaccess-dev \
     uuid-dev \
     libsystemd-dev \
     libevent-dev \
     libxml2-dev \
     libusb-1.0-0-dev \
     python3 \
     python3-pip \
     libblkid-dev \
     e2fslibs-dev \
     pkg-config \
     libelf-dev\
     libnuma-dev -y')

zihengL1, 2022-04-26:
[External_System_ID] ACRN-8743

NanlinXie, 2022-05-22:
Closed with no regression.

closed 2022-05-22 by NanlinXie