projectacrn/acrn-hypervisor #7401: Post-RTVM boot failure with SSRAM enabled

izhouwu, 2022-05-02:
Environment:

 ACRN ww18 release_3.0

acrn-hypervisor: [acrn-2022w18.1-180000p](https://github.com/projectacrn/acrn-hypervisor/releases/tag/acrn-2022w18.1-180000p)

acrn-kernel: 5.10.106  [acrn-2022w18.1-180000p](https://github.com/projectacrn/acrn-kernel/releases/tag/acrn-2022w18.1-180000p)

rt-kernel: [5.10.106]

Test step:

1.enable ssram in scenario xml and build acrn

2. boot ACRN and SOS.

3. boot Post RTVM with SSRARM configurations.

 

Expected result: post RTVM boot successfully

Actual Result: rtvm boot failed.

 

Fail Rate: 100%

log:
init_guest_cpu_info, guest_cpu_bitmask: 0x300
vssram_init_buffers, l2_cat_shift:3, l3_cat_shift:7, l3_inclusive:0.
tcc_driver_req_buffer succ: region_id: 2, size: 512 KB, dev_node=1.
ACRN_IOCTL_UNSET_MEMSEG ioctl() returned an error: Invalid argument
ACRN_IOCTL_SET_MEMSEG ioctl() returned an error: Invalid argument
vssram_ept_map_buffers, setup EPT mapping for cache buffer failed.
init_vssram, setup EPT mapping for vssram buffers failed.

izhouwu, 2022-05-02:
[External_System_ID] ACRN-8851

NanlinXie, 2022-05-22:
Closed with no regression.

closed 2022-05-22 by NanlinXie