projectacrn/acrn-hypervisor #7402: [acrn-deb] install board_inspector overwrites grub cmdline

izhouwu, 2022-05-03:
Expected result

append to default grub cmdline:

 

GRUB_TERMINAL="console serial"
GRUB_SERIAL_COMMAND="serial --speed=115200 --unit=0 --word=8 --parity=no --stop=1"
GRUB_CMDLINE_LINUX=""
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash console=tty0 console=ttyS0,115200 idle=nomwait iomem=relaxed intel_idle.max_cstate=0 intel_pstate=disable”
GRUB_TIMEOUT=20
   
 

Actual result

overwrit default grub cmdline:

 

root@acrn-NUC11TNHi5:/tmp# cat  /etc/default/grub|grep -vE "^#|^$"
GRUB_DISTRIBUTOR=`lsb_release -i -s 2> /dev/null || echo Debian`
GRUB_TERMINAL="console serial"
GRUB_SERIAL_COMMAND="serial --speed=115200 --unit=0 --word=8 --parity=no --stop=1"
GRUB_CMDLINE_LINUX=""
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash idle=nomwait iomem=relaxed intel_idle.max_cstate=0 intel_pstate=disable"
GRUB_TIMEOUT=20
 

Reproduce rate

100%

izhouwu, 2022-05-03:
[External_System_ID] ACRN-8888

NanlinXie, 2022-05-22:
Closed with no regression.

closed 2022-05-22 by NanlinXie