projectacrn/acrn-hypervisor #7425: Cache was not locked after post-RTVM power off and restart

yonghuah, 2022-05-06:
HW:ADL-S-CRB

Enable TCC and SRAM in Bios

HV : mainliine-e09968e5e

Kernel:5.10.83 with  2 kernel patches for RTCT: [ACRN-7497](https://jira.devtools.intel.com/browse/ACRN-7497)

Step:

  1. enable sram and disable cdp in shared.xml, then build acrn

  2. replace acrn build on target board and reboot

  3. boot acrn and SOS 

  4. Run PMC(acrn-unit) as post-rt in SOS.
  5. Check if pmc L2 miss/hit is reasonable comparing with no RTCM

 

Expected result:

cache can be locked after Post-RTVM power off and restart

Actual result:

cache not be locked adter Post-RTVM Power off and restart

yonghuah, 2022-05-06:
[External_System_ID] ACRN-7623

NanlinXie, 2022-05-22:
Closed with no regression.

closed 2022-05-22 by NanlinXie