projectacrn/acrn-hypervisor #7445: [Config-Tool][UI]The shared VMs_name for IVSHMEM is not consistent with the VM_name modification

izhouwu, 2022-05-10:
Environment:

acrn-hypervisor:ac9824cd540589300c54808b8f2fce53681996e2

branch: master

Test step:

1: git clone acrn  code

2: build acrn-configurator

3. launch acrn-configurator

4. on UI, create new configuration and import board xml

5. create new scenario and edit HV basic para for IVSHMEM, edit the shared VMs name and shows " VM0,VM1.VM2.."

6. edit the VM0 vm_name to such as ACRN_Service_VM , save the scenario and found the shared vms name in IVSHEMEM still use VM0;

Expect result:  the shared vms name is consistent with VM_name modification

Actual Result:  the shared vms name is not consistent with VM_name modification

Fail Rate: 100%

izhouwu, 2022-05-10:
[External_System_ID] ACRN-8765

NanlinXie, 2022-05-22:
Closed with no regression.

closed 2022-05-22 by NanlinXie