projectacrn/acrn-hypervisor #7453: [Yocto] Build failure with GCC 12.1.0

saininav, 2022-05-11:
ACRN HV 2.7 build failure with GCC 12.1.0.

Error log:
development/acrn-workspace/poky/build-master/master-acrn-sos/work/intel_corei7_64-oe-linux/acrn-hypervisor/2.7-r0/build/hypervisor/lib/crypto/mbedtls/md.o
| lib/crypto/mbedtls/md.c: In function 'mbedtls_md_hmac_starts':
| lib/crypto/mbedtls/md.c:86:67: error: the comparison will always evaluate as 'false' for the address of 'hmac_ctx' will never be NULL [-Werror=address]
|    86 |     if ((ctx == NULL) || (ctx->md_info == NULL) || (ctx->hmac_ctx == NULL) || (temp_key == NULL)) {
|       |                                                                   ^~
| In file included from lib/crypto/mbedtls/md.c:27:
| lib/crypto/mbedtls/md.h:68:13: note: 'hmac_ctx' declared here
|    68 |     uint8_t hmac_ctx[128];
|       |             ^~~~~~~~
| lib/crypto/mbedtls/md.c: In function 'mbedtls_md_hmac_update':
| lib/crypto/mbedtls/md.c:134:67: error: the comparison will always evaluate as 'false' for the address of 'hmac_ctx' will never be NULL [-Werror=address]
|   134 |     if ((ctx == NULL) || (ctx->md_info == NULL) || (ctx->hmac_ctx == NULL)) {
|       |                                                                   ^~
| lib/crypto/mbedtls/md.h:68:13: note: 'hmac_ctx' declared here
|    68 |     uint8_t hmac_ctx[128];
|       |             ^~~~~~~~
| lib/crypto/mbedtls/md.c: In function 'mbedtls_md_hmac_finish':
| lib/crypto/mbedtls/md.c:149:67: error: the comparison will always evaluate as 'false' for the address of 'hmac_ctx' will never be NULL [-Werror=address]
|   149 |     if ((ctx == NULL) || (ctx->md_info == NULL) || (ctx->hmac_ctx == NULL)) {
|       |                                                                   ^~
| lib/crypto/mbedtls/md.h:68:13: note: 'hmac_ctx' declared here
|    68 |     uint8_t hmac_ctx[128];
|       |             ^~~~~~~~
| cc1: all warnings being treated as errors
| make: *** [Makefile:534: /development/acrn-workspace/poky/build-master/master-acrn-sos/work/intel_corei7_64-oe-linux/acrn-hypervisor/2.7-r0/build/hypervisor/lib/crypto/mbedtls/md.o] Error 1

To reproduce :
https://github.com/intel/meta-acrn/blob/master/docs/getting-started.md

poky: latest master branch
meta-acrn: master (ACRN 2.7)
meta-openembedded: master
meta-virtualization: master
meta-intel: master

wenlingz, 2022-05-11:
@lifeix please help check this issue and fix it, thanks

lifeix, 2022-05-11:
@saininav  You could just remove all the "(ctx->hmac_ctx == NULL)" check in hypervisor/lib/crypto/mbedtls/md.c
--- a/hypervisor/lib/crypto/mbedtls/md.c
+++ b/hypervisor/lib/crypto/mbedtls/md.c
@@ -83,7 +83,7 @@ int32_t mbedtls_md_hmac_starts(mbedtls_md_context_t *ctx, const uint8_t *key, si
     const uint8_t *temp_key = key;
     size_t i;

-    if ((ctx == NULL) || (ctx->md_info == NULL) || (ctx->hmac_ctx == NULL) || (temp_key == NULL)) {
+    if ((ctx == NULL) || (ctx->md_info == NULL) || (temp_key == NULL)) {
         ret = MBEDTLS_ERR_MD_BAD_INPUT_DATA;
     }

@@ -131,7 +131,7 @@ int32_t mbedtls_md_hmac_update(mbedtls_md_context_t *ctx, const uint8_t *input,
 {
     int32_t ret;

-    if ((ctx == NULL) || (ctx->md_info == NULL) || (ctx->hmac_ctx == NULL)) {
+    if ((ctx == NULL) || (ctx->md_info == NULL)) {
         ret = MBEDTLS_ERR_MD_BAD_INPUT_DATA;
     } else {
         ret = ctx->md_info->update_func((void *) ctx->md_ctx, input, ilen);
@@ -146,7 +146,7 @@ int32_t mbedtls_md_hmac_finish(mbedtls_md_context_t *ctx, uint8_t *output)
     uint8_t tmp[MBEDTLS_MD_MAX_SIZE];
     uint8_t *opad;

-    if ((ctx == NULL) || (ctx->md_info == NULL) || (ctx->hmac_ctx == NULL)) {
+    if ((ctx == NULL) || (ctx->md_info == NULL)) {
         ret = MBEDTLS_ERR_MD_BAD_INPUT_DATA;
     }

or "CFLAGS += -Werror" in hypervisor/Makefile
--- a/hypervisor/Makefile
+++ b/hypervisor/Makefile
@@ -61,7 +61,6 @@ CFLAGS += -fsigned-char
 CFLAGS += -m64 -mno-mmx -mno-sse -mno-sse2 -mno-80387 -mno-fp-ret-in-387
 CFLAGS += -mno-red-zone -mpopcnt
 CFLAGS += -nostdinc -nostdlib -fno-common
-CFLAGS += -Werror

lifeix, 2022-05-11:
https://github.com/projectacrn/acrn-hypervisor/pull/7454

saininav, 2022-05-12:
Observed while building tools too:

| acrnctl.c: In function 'valid_start_args':
| acrnctl.c:715:27: error: the comparison will always evaluate as 'true' for the pointer operand in 'argv + 8' must not be NULL [-Werror=address]
|   715 |         if (argc != 2 || ((argv + 1) && !strcmp(argv[1], "help"))) {
|       |                           ^
| make[3]: Leaving directory '/development/acrn-workspace/poky/build-master/master-acrn-sos/work/corei7-64-oe-linux/acrn-tools/2.7-r0/git/misc/debug_tools/acrn_crashlog/common'

lifeix, 2022-05-13:
@saininav 
method a:
diff --git a/misc/services/acrn_manager/acrnctl.c b/misc/services/acrn_manager/acrnctl.c
index 5c959ba32..0f0cbe4a3 100644
--- a/misc/services/acrn_manager/acrnctl.c
+++ b/misc/services/acrn_manager/acrnctl.c
@@ -712,7 +712,7 @@ static int valid_start_args(struct acrnctl_cmd *cmd, int argc, char *argv[])
 {
        char df_opt[16] = "VM_NAME";

-       if (argc != 2 || ((argv + 1) && !strcmp(argv[1], "help"))) {
+       if (argc != 2 || !strcmp(argv[1], "help")) {
                printf("acrnctl %s %s\n", cmd->cmd, df_opt);
                return -1;
        }

method b:
diff --git a/misc/services/acrn_manager/Makefile b/misc/services/acrn_manager/Makefile
index af63b81d1..23ee3b8cb 100644
--- a/misc/services/acrn_manager/Makefile
+++ b/misc/services/acrn_manager/Makefile
@@ -22,7 +22,6 @@ MANAGER_CFLAGS += -D_GNU_SOURCE
 MANAGER_CFLAGS += -DNO_OPENSSL
 MANAGER_CFLAGS += -m64
 MANAGER_CFLAGS += -Wall -ffunction-sections
-MANAGER_CFLAGS += -Werror

saininav, 2022-05-13:
Thanks @lifeix , I will ignore warnings for now..

fuzhongl, 2022-05-18:
PR merged!

saininav, 2022-05-19:
There are still gcc-12 warnings coming in devicemodel:

em.o -MMD -MT /home/saininav/work/development/acrn-workspace/poky/build-master/master-acrn-sos/work/corei7-64-oe-linux/acrn-devicemodel/3.0-r0/build/devicemodel/core/mem.o
| hw/pci/gvt.c: In function 'gvt_init_config.constprop':
| hw/pci/gvt.c:263:41: error: array subscript 257 is above array bounds of 'uint8_t[256]' {aka 'unsigned char[256]'} [-Werror=array-bounds]
|   263 |                         gvt->host_config[cap_ptr + 4]);
|       |                         ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
| hw/pci/gvt.c:48:17: note: while referencing 'host_config'
|    48 |         uint8_t host_config[PCI_REGMAX+1];
|       |                 ^~~~~~~~~~~

saininav, 2022-05-19:
acrn life-mngr:

| In file included from uart_channel.c:15:
| In function 'destroy_uart_channel_devs',
|     inlined from 'deinit_uart_channel' at uart_channel.c:354:3:
| uart_channel.c:313:9: error: pointer 'c_dev' used after 'free' [-Werror=use-after-free]
|   313 |         LIST_FOREACH(c_dev, &c->tty_open_head, open_list) {
|       |         ^~~~~~~~~~~~
| uart_channel.c:319:17: note: call to 'free' here
|   319 |                 free(c_dev);
|       |                 ^~~~~~~~~~~
| cc1: all warnings being treated as errors

jiaqingz-intel, 2023-05-30:
Fixed in https://github.com/projectacrn/acrn-hypervisor/commit/632097778865e892daab593d6044ff3823fbb4c7

Similar Issue: https://github.com/projectacrn/acrn-hypervisor/issues/8382

saininav, 2023-06-13:
Still facing following warnings:

 hw/pci/xhci.c: In function 'pci_xhci_cmd_disable_slot':
| hw/pci/xhci.c:1945:32: error: the comparison will always evaluate as 'true' for the pointer operand in 'xdev->portregs + (sizetype)((long unsigned int)i * 16)' must not be NULL [-Werror=address]
|  1945 |         if (i <= XHCI_MAX_DEVS && XHCI_PORTREG_PTR(xdev, i)) {

saininav, 2023-06-13:
I am testing release_3.1 branch with gcc-13. I cherry picked followings from master to release_3.1 branch
https://github.com/projectacrn/acrn-hypervisor/commit/632097778865e892daab593d6044ff3823fbb4c7
https://github.com/projectacrn/acrn-hypervisor/commit/e4e25f076ca374004de9383a89b76c308f71c809

Please let me know, if any other commit also pick from master..!!

saininav, 2023-06-13:
here too.

| In file included from uart_channel.c:15:
| In function 'destroy_uart_channel_devs',
|     inlined from 'deinit_uart_channel' at uart_channel.c:354:3:
| uart_channel.c:313:9: error: pointer 'c_dev' used after 'free' [-Werror=use-after-free]
|   313 |         LIST_FOREACH(c_dev, &c->tty_open_head, open_list) {
|       |         ^~~~~~~~~~~~
| uart_channel.c:319:17: note: call to 'free' here
|   319 |                 free(c_dev);
|       |                 ^~~~~~~~~~~
| cc1: all warnings being treated as errors

jiaqingz-intel, 2023-06-13:
Please try with relese_3.2

saininav, 2023-06-14:
release_3.2 branch builds successfully with gcc-13.

closed 2023-06-14 by saininav