projectacrn/acrn-hypervisor #7494: Copyright year range in code headers is not up to date v2.3

zihengL1, 2022-05-17:
Code files include a copyright comment, something like this:

```

Copyright (c) 2017 Intel Corporation
```

According to legal guidelines, the year should include the first and last year when changes were made, something like this:

```

Copyright (c) 2017-2020 Intel Corporation
```

Of the 232 source files in the hypervisor tree, 186 of them have changed (noted in the GitHub log) but the copyright year has not been updated to reflect this.

 

We should go back and update the copyright comment year to include the year range where edits have been made and we should add a copyright year check to the code review checklist when changes are submitted.

zihengL1, 2022-05-17:
[External_System_ID] ACRN-6921

closed 2022-07-13 by zihengL1

dbkinder, 2022-07-13:
@NanlinXie Why was this issue closed?  The copyright years are still not updated (there are no merged PRs that mention this issue).

zihengL1, 2022-07-13:
> @NanlinXie Why was this issue closed? The copyright years are still not updated (there are no merged PRs that mention this issue).

This is because I created two duplicate issues (7494 and 7559) before, and I will PR at issue 7559.

dbkinder, 2022-07-13:
Ah, OK. Worth a comment when closing that this is a duplicate issue. Thanks for explaining.