projectacrn/acrn-hypervisor #7546: [acrn-configuration-tool]Scenario files generated with acrn-configurator for boards without serial ports ACRN debug build fails

zhangqiang4, 2022-05-23:
Environment:

daily-build/2022ww20.3-1/ubuntu-configurator/acrn-configurator_0.3.0_x64_en-US.msi

Test step:

1:install acrn-configuration

2: start a new configuration-> browser for folder->create new folder(RC3) -> use this folder

3: import board xml and create scenario xml and saved 

4: build acrn: (make clean && make BOARD=nuc11pabi7 SCENARIO=shared RELEASE=y)

Expect result:  

Scenario xml generated with acrn-configurator requires <SERIAL_CONSOLE></SERIAL_CONSOLE> parameter:

<DEBUG_OPTIONS>
      <BUILD_TYPE>debug</BUILD_TYPE>
      <SERIAL_CONSOLE></SERIAL_CONSOLE>
      <MEM_LOGLEVEL>5</MEM_LOGLEVEL>
      <NPK_LOGLEVEL>5</NPK_LOGLEVEL>
      <CONSOLE_LOGLEVEL>3</CONSOLE_LOGLEVEL>
    </DEBUG_OPTIONS>

Actual Result:  

Scenario xml generated with acrn-configurator has no <SERIAL_CONSOLE></SERIAL_CONSOLE> parameter : 

    <DEBUG_OPTIONS>
      <BUILD_TYPE>debug</BUILD_TYPE>
      <MEM_LOGLEVEL>5</MEM_LOGLEVEL>
      <NPK_LOGLEVEL>5</NPK_LOGLEVEL>
      <CONSOLE_LOGLEVEL>3</CONSOLE_LOGLEVEL>
    </DEBUG_OPTIONS>

Fail Rate: 100%

Debugging info{}

root@acrn-System-Product-Name:/work/acrn-hypervisor# make BOARD=nuc11pabi7 SCENARIO=shared RELEASE=y
make[1]: Entering directory '/work/acrn-hypervisor/hypervisor'
Board XML is fetched from /work/acrn-hypervisor/misc/config_tools/data/nuc11pabi7/nuc11pabi7.xml
Scenario XML is being fetched from /work/acrn-hypervisor/misc/config_tools/data/nuc11pabi7/shared.xml
/work/acrn-hypervisor/build/hypervisor/configs/allocation.xml generated
/work/acrn-hypervisor/build/hypervisor/configs/unified.xml generated
/work/acrn-hypervisor/build/hypervisor/configs/config.mk generated
scripts/makefile/config.mk:191: The command line sets RELEASE to be 'y', but an existing build is configured with 'n'
scripts/makefile/config.mk:191: The configuration will be modified for RELEASE=y
/work/acrn-hypervisor/build/hypervisor/configs/allocation.xml generated
/work/acrn-hypervisor/build/hypervisor/configs/unified.xml generated
/work/acrn-hypervisor/build/hypervisor/configs/config.mk generated
CRITICAL:root:/acrn-config/hv/DEBUG_OPTIONS: The content of element 'DEBUG_OPTIONS' is not complete. Tag 'SERIAL_CONSOLE' expected.
ERROR:root:Board nuc11pabi7 and scenario shared are inconsistent: 1 syntax errors, 0 data errors, 0 warnings.
make[1]: Leaving directory '/work/acrn-hypervisor/hypervisor'
make -C /work/acrn-hypervisor/hypervisor BOARD=nuc11pabi7 SCENARIO=shared HV_OBJDIR=/work/acrn-hypervisor/build/hypervisor RELEASE=y
make[1]: Entering directory '/work/acrn-hypervisor/hypervisor'
/work/acrn-hypervisor/build/hypervisor/include/config.h generated
Start pre-build static check ...
make -C ../misc/hv_prebuild BOARD=nuc11pabi7 SCENARIO=shared TARGET_DIR=/work/acrn-hypervisor/build/hypervisor/configs
make[2]: Entering directory '/work/acrn-hypervisor/misc/hv_prebuild'
make[2]: *** No rule to make target '/work/acrn-hypervisor/build/hypervisor/configs/scenarios/shared/vm_configurations.c', needed by 'default'.  Stop.
make[2]: Leaving directory '/work/acrn-hypervisor/misc/hv_prebuild'
make[1]: *** [Makefile:419: pre_build] Error 2
make[1]: Leaving directory '/work/acrn-hypervisor/hypervisor'
make: *** [Makefile:130: hypervisor] Error 2

zhangqiang4, 2022-05-23:
[External_System_ID] ACRN-9029

NanlinXie, 2022-06-05:
Close with no regression.

closed 2022-06-05 by NanlinXie