projectacrn/acrn-hypervisor #7556: [config_tool] VUART is configured to PCI and generate launch script without relevant parameters

Kunhui-Li, 2022-05-24:
**Describe the bug**
After VUART is configured to PCI, the generated launch script without related dm parameters.

**Step**
1. build configurator (Win)
2. install configurator and launch
3. Set workdir and load ad-s-crb board xml
4. Import/Load default hybrid_rt scenario xml
5. Modify HV basic InterVM Virtual UART Connection from legacy to pci
6. Click Save
7. Check scenario xml and launched scipts.

**Expected result**
The parameters of DM in the script should be generated correctly, for example:
`add_virtual_device                       3 uart vuart_idx:1`

**Actual result**
The generated launch script without related dm parameters.

Kunhui-Li, 2022-05-24:
[External_System_ID] ACRN-9044

NanlinXie, 2022-06-05:
Close with no regression.

closed 2022-06-05 by NanlinXie