projectacrn/acrn-hypervisor #7581: [ADL-S][shared]The SOS cmdline parameter shouldn't be added manually and should be changed in Debian package.

zhangqiang4, 2022-05-25:
ACRN version: HV 3.0-unstable-2022-04-28 07:19:54-f8f1689a8 DBG (daily tag: acrn-2022w18.1-180000p) shared@adl-s-crb build by acrn API 1.0
BIOS Version: IOTG_ADL_SR06_C2B1-ADPSXF2_CPSF_SEP5_01580518_2021WW44.1.1_ProdInt2501_TSN25G1G_25MHz_SATA.bin 
SOS kernel: 5.10.106-acrn-service-vm-209086-ga2ffbcda0a64-dirty
WaaG gfx: wingfx driver:Intel(R) UHD Graphics (30.0.101.1631) 

root@virtio-gpu01:~# cat /proc/cmdline | more
root=/dev/nvme0n1p2 idle=halt rw rootwait console=ttyS0 console=tty0 earlyprintk=serial,ttyS0,115200 cons_timer_check consoleblank=0 no_timer_check quiet logleve
l=3 i915.nuclear_pageflip=1 nokaslr i915.force_probe=* i915.enable_guc=0x7 maxcpus=16 hugepagesz=1G hugepages=28  root=PARTUUID=cc230897-ea47-4fdd-a55d-0e38c6d90
d20 *{color:#de350b}udmabuf.list_limit=8192{color}*

Enable zero-copy:
Add udmabuf.list_limit=8192 to cmdline

The SOS cmdline parameter shouldn't be added manually and should be changed in Debian package.

zhangqiang4, 2022-05-25:
[External_System_ID] ACRN-8903

NanlinXie, 2022-06-05:
Close with no regression.

closed 2022-06-05 by NanlinXie