projectacrn/acrn-hypervisor #7609: [Config-Tool][UI]build acrn failed after delete VMs via UI

izhouwu, 2022-05-27:
Environment:

config_tool: latest daily 5/23 build

hv commit: 2ff11c2ef04a2668979b3e363e25f13cf48376ac

offline-patch commit:  e03784297f40c144f4310e2b1d822987302af209

Test step:{}

1:install config_tool and launch

2: select working folder

3: import board xml and load default shared scenario

4: dele some Post VMs on UI and save Scenario and launch scripts

5: build acrn with the save scenario and board xml

Expect result:  build acrn passed with saved scenario which delete VMs via UI

Actual result:  build acrn failed after delete some VMs via UI

python3 misc/packaging/gen_acrn_deb.py clean /root/ld/acrn-hypervisor/build {}version=3.0-unstable"{-}"acrn-2022w21.5-180000p-29-gf5e58e6f3-dirty;
make[1]: Entering directory '/root/ld/acrn-hypervisor/hypervisor'
Board XML is fetched from /root/ld/delete-post-launched/adl-s-crb.board.xml
Scenario XML is being fetched from /root/ld/delete-post-launched/scenario.xml
/root/ld/acrn-hypervisor/build/hypervisor/configs/allocation.xml generated
/root/ld/acrn-hypervisor/build/hypervisor/configs/unified.xml generated
/root/ld/acrn-hypervisor/build/hypervisor/configs/config.mk generated
CRITICAL:root:/acrn-config/hv/CACHE_REGION/CACHE_ALLOCATION[3]/POLICY: The content of element 'POLICY' is not complete. Tag 'VM' expected.
CRITICAL:root:/acrn-config/hv/CACHE_REGION/CACHE_ALLOCATION[3]/POLICY: The content of element 'POLICY' is not complete. Tag 'VCPU' expected.
CRITICAL:root:/acrn-config/hv/CACHE_REGION/CACHE_ALLOCATION[3]/POLICY: The content of element 'POLICY' is not complete. Tag 'TYPE' expected.
CRITICAL:root:/acrn-config/hv/CACHE_REGION/CACHE_ALLOCATION[3]/POLICY: The content of element 'POLICY' is not complete. Tag 'CLOS_MASK' expected.
ERROR:root:Board adl-s-crb and scenario delete-post-launched are inconsistent: 4 syntax errors, 0 data errors, 0 warnings.
make[1]: Leaving directory '/root/ld/acrn-hypervisor/hypervisor'
make -C /root/ld/acrn-hypervisor/hypervisor BOARD=/root/ld/delete-post-launched/adl-s-crb.board.xml SCENARIO=/root/ld/delete-post-launched/scenario.xml HV_OBJDIR=/root/ld/acrn-hypervisor/build/hypervisor RELEASE=
make[1]: Entering directory '/root/ld/acrn-hypervisor/hypervisor'

izhouwu, 2022-05-27:
[External_System_ID] ACRN-9069

NanlinXie, 2022-06-05:
Close with no regression.

closed 2022-06-05 by NanlinXie