projectacrn/acrn-hypervisor #7618: v2.7 ServiceVM is not booting on QEMU

archersmind, 2022-05-29:
**Describe the bug**
A clear and concise description of what the bug is.
ServiceVM is not booting on QEMU, stuck at:

```
[    0.703573] smpboot: CPU0: Intel Atom Processor (Denverton) (family: 0x6, model: 0x5f, stepping: 0x1)
[    0.704531] Performance Events: unsupported p6 CPU model 95 no PMU driver, software events only.
[    0.705490] rcu: Hierarchical SRCU implementation.
[    0.706708] NMI watchdog: Perf NMI watchdog permanently disabled
[    0.707493] smp: Bringing up secondary CPUs ...
[    0.708703] x86: Booting SMP configuration:
[    0.709444] .... node  #0, CPUs:      #1
[    0.385763] smpboot: CPU 1 Converting physical 0 to logical die 1
[    0.776716]  #2
[    0.385763] smpboot: CPU 2 Converting physical 0 to logical die 2
[    0.842689]  #3
[    0.385763] smpboot: CPU 3 Converting physical 0 to logical die 3
```

**Platform**
Host Platform NUC11THNi5
Following Guide: https://projectacrn.github.io/2.7/tutorials/acrn_on_qemu.html

**Codebase**
v2.7 for both acrn-hypervisor and acrn-kernel(ServiceVM)

**Scenario**
Describe the scenario: QEMU

**To Reproduce**
Steps to reproduce the behavior:
Following guide:
https://projectacrn.github.io/2.7/tutorials/acrn_on_qemu.html
https://projectacrn.github.io/2.7/getting-started/getting-started.html#gsg

**Expected behavior**
QEMU should booting ServiceVM

**Additional context**
While acrn-hypervisor launching the ServiceVM, use `vm_console 0` to see the VM log, I see the booting procedure stuck at following line:

```
ACRN:\>vm_console 0

----- Entering VM 0 Shell -----
000001 BXPC 00000001)
[    0.057238] ACPI: HPET 0x000000007FFE2164 000038 (v01 BOCHS  BXPCHPET 00000001 BXPC 00000001)
[    0.060743] ACPI: MCFG 0x000000007FFE219C 00003C (v01 BOCHS  BXPCMCFG 00000001 BXPC 00000001)
[    0.064243] ACPI: DMAR 0x000000007FFE21D8 000048 (v01 BOCHS  BXPCDMAR 00000001 BXPC 00000001)
[    0.067748] ACPI: Reserving FACP table memory at [mem 0x7ffe1fe0-0x7ffe20d3]
[    0.070626] ACPI: Reserving DSDT table memory at [mem 0x7ffe0040-0x7ffe1fdf]
[    0.073523] ACPI: Reserving FACS table memory at [mem 0x7ffe0000-0x7ffe003f]
[    0.076429] ACPI: Reserving APIC table memory at [mem 0x7ffe20d4-0x7ffe2163]
[    0.079329] ACPI: Reserving HPET table memory at [mem 0x7ffe2164-0x7ffe219b]
[    0.082225] ACPI: Reserving MCFG table memory at [mem 0x7ffe219c-0x7ffe21d7]
[    0.085122] ACPI: Reserving DMAR table memory at [mem 0x7ffe21d8-0x7ffe221f]
[    0.088062] ACPI: Local APIC address 0xfee00000
[    0.089986] Zone ranges:
[    0.091024]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.093563]   DMA32    [mem 0x0000000001000000-0x00000000ffffffff]
[    0.096109]   Normal   [mem 0x0000000100000000-0x00000001759fffff]
[    0.098634] Movable zone start for each node
[    0.100393] Early memory node ranges
[    0.101870]   node   0: [mem 0x0000000000001000-0x000000000008efff]
[    0.104447]   node   0: [mem 0x0000000000100000-0x0000000010ffffff]
[    0.107027]   node   0: [mem 0x0000000018e00000-0x000000007ffd7fff]
[    0.109599]   node   0: [mem 0x0000000100000000-0x00000001759fffff]
[    0.112179] Initmem setup node 0 [mem 0x0000000000001000-0x00000001759fffff]
[    0.115061] On node 0 totalpages: 973670
[    0.116699]   DMA zone: 64 pages used for memmap
[... 68 lines cut ...]
[    0.689437] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.690430] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
Poking KASLR using RDRAND RDTSC...
[    0.694930] Last level iTLB entries: 4KB 0, 2MB 0, 4MB 0
[    0.695425] Last level dTLB entries: 4KB 0, 2MB 0, 4MB 0, 1GB 0
[    0.696428] Spectre V1 : Mitigation: usercopy/swapgs barriers and __user pointer sanitization
[    0.697426] Spectre V2 : Mitigation: Full generic retpoline
[    0.698425] Spectre V2 : Spectre v2 / SpectreRSB mitigation: Filling RSB on context switch
[    0.699425] Spectre V2 : Enabling Restricted Speculation for firmware calls
[    0.700426] Spectre V2 : mitigation: Enabling conditional Indirect Branch Prediction Barrier
[    0.701426] Speculative Store Bypass: Mitigation: Speculative Store Bypass disabled via prctl and seccomp
[    0.702665] Freeing SMP alternatives memory: 52K
[    0.703573] smpboot: CPU0: Intel Atom Processor (Denverton) (family: 0x6, model: 0x5f, stepping: 0x1)
[    0.704531] Performance Events: unsupported p6 CPU model 95 no PMU driver, software events only.
[    0.705490] rcu: Hierarchical SRCU implementation.
[    0.706708] NMI watchdog: Perf NMI watchdog permanently disabled
[    0.707493] smp: Bringing up secondary CPUs ...
[    0.708703] x86: Booting SMP configuration:
[    0.709444] .... node  #0, CPUs:      #1
[    0.385763] smpboot: CPU 1 Converting physical 0 to logical die 1
[    0.776716]  #2
[    0.385763] smpboot: CPU 2 Converting physical 0 to logical die 2
[    0.842689]  #3
[    0.385763] smpboot: CPU 3 Converting physical 0 to logical die 3
```

gvancuts, 2022-05-30:
It does not look like you are booting that kernel under ACRN. Can you confirm you did select the "ACRN hypervisor" from the Grub boot menu (just after you ran `virsh start ServiceVM --console`)?

archersmind, 2022-05-30:
@gvancuts I can confirm that, after a few trials it stucks at another pos while booting ServiceVM kernel:
```

[    0.605255] Last level iTLB entries: 4KB 0, 2MB 0, 4MB 0
[    0.606089] Last level dTLB entries: 4KB 0, 2MB 0, 4MB 0, 1GB 0
[    0.607093] Spectre V1 : Mitigation: usercopy/swapgs barriers and __user pointer sanitization
[    0.608090] Spectre V2 : Mitigation: Full generic retpoline
[    0.609088] Spectre V2 : Spectre v2 / SpectreRSB mitigation: Filling RSB on context switch
[    0.610088] Spectre V2 : Enabling Restricted Speculation for firmware calls
[    0.611089] Spectre V2 : mitigation: Enabling conditional Indirect Branch Prediction Barrier
[    0.612089] Speculative Store Bypass: Mitigation: Speculative Store Bypass disabled via prctl and seccomp
[    0.614228] Freeing SMP alternatives memory: 52K
[    0.615225] smpboot: CPU0: Intel Atom Processor (Denverton) (family: 0x6, model: 0x5f, stepping: 0x1)
[    0.616193] Performance Events: unsupported p6 CPU model 95 no PMU driver, software events only.
[    0.617152] rcu: Hierarchical SRCU implementation.
[    0.618366] NMI watchdog: Perf NMI watchdog permanently disabled
[    0.619169] smp: Bringing up secondary CPUs ...
[    0.620333] x86: Booting SMP configuration:
[    0.621104] .... node  #0, CPUs:      #1
[    0.323627] smpboot: CPU 1 Converting physical 0 to logical die 1
[    0.687342]  #2
[    0.323627] smpboot: CPU 2 Converting physical 0 to logical die 2
[    0.752345]  #3
[    0.323627] smpboot: CPU 3 Converting physical 0 to logical die 3
[    0.817188] smp: Brought up 1 node, 4 CPUs
[    0.818107] smpboot: Max logical packages: 4
[    0.819104] smpboot: Total of 4 processors activated (19597.50 BogoMIPS)
[    0.821287] devtmpfs: initialized
[    0.822577] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275000 ns
[    0.823108] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.824150] pinctrl core: initialized pinctrl subsystem
[    0.825504] PM: RTC time: 11:58:03, date: 2022-05-30
```

archersmind, 2022-05-30:
ServiceVM Log from the beginning:

-```
---- Entering VM 0 Shell -----
Wrong EFI loader signature.
early console in extract_kernel
input_data: 0x000000000f9a73b1
input_len: 0x0000000000e4d386
output: 0x000000000e000000
output_len: 0x00000000025e7f58
kernel_total_size: 0x000000000282c000
needed_size: 0x0000000002a00000
trampoline_32bit: 0x000000000008d000
Physical KASLR using RDRAND RDTSC...
Virtual KASLR using RDRAND RDTSC...

Decompressing Linux... Parsing ELF... Performing relocations... done.
Booting the kernel.
[    0.000000] Linux version 5.10.78-acrn-service-vm (alan@ubuntu) (gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0, GNU ld (GNU Binutils for Ubuntu) 2.30) #1 SMP PREEMPT Mon May 30 03:15:28 CST 2022
[    0.000000] Command line: root=/dev/vda1 console=ttyS0 idle=halt earlyprintk=serial,ttyS0,115200n8 rw rootwait console=tty0 consoleblank=0 no_timer_check ignore_loglevel ignore_loglevel no_timer_check intel_iommu=off tsc=reliable maxcpus=4 hugepagesz=1G hugepages=1  
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000008efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000008f000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000010ffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000018e00000-0x000000007ffd7fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffd8000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b0000000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000001759fffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000175a00000-0x000000017fffffff] reserved
[    0.000000] printk: bootconsole [earlyser0] enabled
[    0.000000] printk: debug: ignoring loglevel setting.
[    0.000000] printk: debug: ignoring loglevel setting.
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 1.13.0-1ubuntu1.1 04/01/2014
[    0.000000] Hypervisor detected: ACRN
[    0.000000] tsc: Detected 2422.000 MHz processor
[    0.000271] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.003027] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.005318] last_pfn = 0x175a00 max_arch_pfn = 0x400000000
[    0.007859] MTRR default type: uncachable
[    0.009522] MTRR fixed ranges enabled:
[    0.011094]   00000-9FFFF write-back
[    0.012724]   A0000-BFFFF uncachable
[    0.014330]   C0000-FFFFF write-protect
[    0.023083] MTRR variable ranges enabled:
[    0.024963] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.027907] CPU MTRRs all blank - virtualized system.
Memory KASLR using RDRAND RDTSC...
[    0.031286] last_pfn = 0x7ffd8 max_arch_pfn = 0x400000000
[    0.037414] found SMP MP-table at [mem 0x000f5c40-0x000f5c4f]
[    0.039842] check: Scanning 1 areas for low memory corruption
[    0.042267] Using GB pages for direct mapping
[    0.044423] ACPI: Early table checksum verification disabled
[    0.047313] ACPI: RSDP 0x00000000000F59A0 000014 (v00 BOCHS )
[    0.049932] ACPI: RSDT 0x000000007FFE2220 000038 (v01 BOCHS  BXPCRSDT 00000001 BXPC 00000001)
[    0.053464] ACPI: FACP 0x000000007FFE1FE0 0000F4 (v03 BOCHS  BXPCFACP 00000001 BXPC 00000001)
[    0.063437] ACPI: DSDT 0x000000007FFE0040 001FA0 (v01 BOCHS  BXPCDSDT 00000001 BXPC 00000001)
[    0.067003] ACPI: FACS 0x000000007FFE0000 000040
[    0.068903] ACPI: APIC 0x000000007FFE20D4 000090 (v01 BOCHS  BXPCAPIC 00000001 BXPC 00000001)
[    0.072401] ACPI: HPET 0x000000007FFE2164 000038 (v01 BOCHS  BXPCHPET 00000001 BXPC 00000001)
[    0.076248] ACPI: MCFG 0x000000007FFE219C 00003C (v01 BOCHS  BXPCMCFG 00000001 BXPC 00000001)
[    0.080614] ACPI: DMAR 0x000000007FFE21D8 000048 (v01 BOCHS  BXPCDMAR 00000001 BXPC 00000001)
[    0.084130] ACPI: Reserving FACP table memory at [mem 0x7ffe1fe0-0x7ffe20d3]
[    0.087029] ACPI: Reserving DSDT table memory at [mem 0x7ffe0040-0x7ffe1fdf]
[... 215 lines cut ...]
Poking KASLR using RDRAND RDTSC...
[    0.922637] Last level iTLB entries: 4KB 0, 2MB 0, 4MB 0
[    0.923186] Last level dTLB entries: 4KB 0, 2MB 0, 4MB 0, 1GB 0
[    0.924201] Spectre V1 : Mitigation: usercopy/swapgs barriers and __user pointer sanitization
[    0.925186] Spectre V2 : Mitigation: Full generic retpoline
[    0.926185] Spectre V2 : Spectre v2 / SpectreRSB mitigation: Filling RSB on context switch
[    0.927185] Spectre V2 : Enabling Restricted Speculation for firmware calls
[    0.928186] Spectre V2 : mitigation: Enabling conditional Indirect Branch Prediction Barrier
[    0.929186] Speculative Store Bypass: Mitigation: Speculative Store Bypass disabled via prctl and seccomp
[    0.931192] Freeing SMP alternatives memory: 52K
[    0.932413] smpboot: CPU0: Intel Atom Processor (Denverton) (family: 0x6, model: 0x5f, stepping: 0x1)
[    0.933440] Performance Events: unsupported p6 CPU model 95 no PMU driver, so[845952058us][cpu=0][vm0:vcpu0][sev=3][seq=36]:vlapic: Start Secondary VCPU1 for VM[0]...
ftware events only.
[    0.934275] rcu: Hierarchical SRCU implementation.
[... cut ...] og permanently disabled
[    0.936326] smp: Bringing up secondary CPUs ...
[    0.937843] x86: Booting SMP configuration:
[    0.938204] .... node  #0, CPUs:      #1
[    0.478540] smpboot: CPU 1 Converting physical 0 to logical die 1
[846026138us][cpu=0][vm0:vcpu0][sev=3][seq=40]:vlapic: Start Secondary VCPU2 for VM[0]...
[    0.478540] calibrate_delay_direct() dropping max bogoMips estimate 1 = 2976025
[    1.008570]  #2
[    0.478540] smpboot: CPU 2 Converting physical 0 to logical die 2

```

fuzhongl, 2022-06-01:
@archersmind 
This issue is fixed by add following parameter in --qemu-commandline:
`-vmx-apicv-vid`

`--qemu-commandline="-machine kernel-irqchip=split -cpu Denverton,+invtsc,+lm,+nx,+smep,+smap,+mtrr,+clflushopt,+vmx,+x2apic,+popcnt,-xsave,+sse,+rdrand,-vmx-apicv-vid,+vmx-apicv-xapic,+vmx-apicv-x2apic,+vmx-flexpriority,+tsc-deadline,+pdpe1gb -device intel-iommu,intremap=on,caching-mode=on,aw-bits=48" \`

About the modification, please  refer: https://projectacrn.github.io/latest/tutorials/acrn_on_qemu.html

archersmind, 2022-06-01:
@fuzhongl 
Thank you for the info. I don't have the chance to re-verify the fix currently. But I think the ACRN team should update the tutorials for v2.7 https://github.com/projectacrn/acrn-hypervisor/blob/v2.7/doc/tutorials/acrn_on_qemu.rst.

I can submit a PR but not now.

fuzhongl, 2022-06-02:
@archersmind 
Thanks! Any issue during the verification, please let us know.

BTW: **TGL** isn't the verified platform of v2.7; it is verified with master branch and update the doc also.
V2.7:
`Platforms tested: Kaby Lake, Skylake`

Master：
`Platforms tested: Kaby Lake, Skylake, Whiskey Lake, Tiger Lake`

archersmind, 2022-06-16:
@fuzhongl Thanks for the info above, I can get ACRN running over QEMU. This ticket can be closed.

closed 2022-06-16 by archersmind