projectacrn/acrn-hypervisor #7619: [Question] How to see acrn log and ServiceVM console log simultaneously, when Enable acrn on QEMU.

archersmind, 2022-05-29:
When Enable acrn over QEMU, how to see logs on both side simultaneously? Appreciate for your answers.

archersmind, 2022-05-30:
This question is regarding another issue raised by me [7618](https://github.com/projectacrn/acrn-hypervisor/issues/7618). 

I mean once  ACRN hypervisor launched, I can see a ACRN prompt  `ACRN:\>`,  by given the command `vm_console 0`, console switched to ServiceVM kernel log.

Then how can I switch back to see the ACRN log? Can I see ACRN log and ServiceVM log at the same time? I can't ssh login to the Service domain, as the ServiceVM is not boot yet(as per my issue [7618](https://github.com/projectacrn/acrn-hypervisor/issues/7618)).

```
ACRN:\>vm_list

VM_ID VM_NAME                          VM_STATE
===== ================================ ========
  0   ACRN_Service_VM                  Running 
ACRN:\>vm_console 0

----- Entering VM 0 Shell -----
e memory at [mem 0x7ffe0000-0x7ffe003f]
[    0.065438] ACPI: Reserving APIC table memory at [mem 0x7ffe20d4-0x7ffe2163]
[    0.067882] ACPI: Reserving HPET table memory at [mem 0x7ffe2164-0x7ffe219b]
[    0.070346] ACPI: Reserving MCFG table memory at [mem 0x7ffe219c-0x7ffe21d7]
[    0.072778] ACPI: Reserving DMAR table memory at [mem 0x7ffe21d8-0x7ffe221f]
[    0.075228] ACPI: Local APIC address 0xfee00000
[    0.076830] Zone ranges:
[    0.077697]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.079815]   DMA32    [mem 0x0000000001000000-0x00000000ffffffff]
[    0.081935]   Normal   [mem 0x0000000100000000-0x00000001759fffff]
[    0.084051] Movable zone start for each node
[    0.085545] Early memory node ranges

```

gvancuts, 2022-05-30:
As an FYI, one patch that could be useful in your case too is from Helmut (see https://github.com/tttech-industrial-buchsbaum/acrn-hypervisor/blob/debian/master/debian/patches/0008-hv-add-vm_console-command-line-parameter.patch). It would allow to add the `vm_console=0` to the hypervisor command-line and make it switch immediately to the Service VM console as soon as it starts, this would allow us to capture **all** the kernel messages as opposed to the one only after you've manually typed `vm_console 0` in the ACRN shell (which inevitably makes us loose some).

/cc @tttech-industrial-buchsbaum

closed 2022-05-30 by archersmind