projectacrn/acrn-hypervisor #7632: Clean top-level Makefile of old, deprecated boards

gvancuts, 2022-05-30:
The top-level Makefile contains a number of targets that are no longer supported, and for which we no longer have the relevant files such as `$(BOARD).xml` in the repository - see what should be removed: https://github.com/projectacrn/acrn-hypervisor/blob/833137517202c7c69a73f6910bbf6d59d5a4e281/Makefile#L178-L200

```
kbl-nuc-i7-industry:
	$(call build_acrn,nuc7i7dnb,industry)
apl-up2-hybrid:
	$(call build_acrn,apl-up2,hybrid)

sbl-hypervisor: kbl-nuc-i7-industry \
                apl-up2-hybrid

kbl-nuc-i7-industry-install:
	$(call install_acrn,nuc7i7dnb,industry)
apl-up2-hybrid-install:
	$(call install_acrn,apl-up2,hybrid)

sbl-hypervisor-install: kbl-nuc-i7-industry-install \
                        apl-up2-hybrid-install

kbl-nuc-i7-industry-install-debug:
	$(call install_acrn_debug,nuc7i7dnb,industry)
apl-up2-hybrid-install-debug:
	$(call install_acrn_debug,apl-up2,hybrid)

sbl-hypervisor-install-debug: kbl-nuc-i7-industry-install-debug \
			      apl-up2-hybrid-install-debug
```

gvancuts, 2022-05-30:
@NanlinXie , can we go ahead and remove those?

NanlinXie, 2022-05-31:
Yes, Those are legacy targets for clearlinux, and could be removed.

gvancuts, 2022-06-06:
[External_System_ID] ACRN-9160

closed 2022-07-05 by fuzhongl