projectacrn/acrn-hypervisor #7634: [DX][TeamFooding][ADL-S] Failed to build acrn when disable multiboot2 in configurator

zhangqiang4, 2022-05-31:
*How Found*

    RC4 DX test

*Environment*

    Windows10 acrn-configurator

    Version acrn-2022w22.3-180000p-11-g8d878c32a-dirty

*HW/Board*

    adl-s-crb

*Build link*

    Daily build:

    Date: 2022-5-26

*Image info*

     Na

*Steps*
 #  Install acrn-configurator and Launch
 # Set work directory
 # Import adl-s-crb board xml
 # Load default shared scenario xml
 # Uncheck option Multiboot2 in UI and then click save
 # copy the board and scenario xml to acrn-hypervisor src code directory
 # rename baord and scenario xml
 # make clean && make BOARD=adl-s-crb SCNEARIO=shared

*Expected result*

     build suceessfully

*Actual result*

    build failed.

*Debugging info*
Check scenario xml file：
<MULTIBOOT2_ENABLED>n</MULTIBOOT2_ENABLED>
Build Partial fail log：

boot/guest/bzimage_loader.c:188:17: error: ‘create_service_vm_efi_mmap_desc’ defined but not used [-Werror=unused-function]
  188 | static uint16_t create_service_vm_efi_mmap_desc(struct acrn_vm *vm, struct efi_memory_desc *efi_mmap_desc)
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors

zhangqiang4, 2022-05-31:
[External_System_ID] ACRN-9125

NanlinXie, 2022-06-05:
Close with no regression.

closed 2022-06-05 by NanlinXie