projectacrn/acrn-hypervisor #7657: [acrn-configuration-tool] Failed to build acrn with make BOARD=xx SCENARIO=shared RELEASE=y

zhangqiang4, 2022-06-01:
acrn-hypervisor: [1b7e8e393|https://github.com/projectacrn/acrn-hypervisor/commit/1b7e8e393fa87beb27e6eb6f8ebd2634fa00224e]   (branch: release_3.0)

*HW/Board*

     whl-ipc-i5

*Build link*

     Na

*Image info*

      Na

*Steps*
 #  branch checkout to release_3.0 and update to latest
 # make clean && make BOARD=whl-ipc-i5 SCENARIO=shared RELEASE=y 

*Expected result*

    build successfully

*Actual result*

    failed to build， Cycle to print warning information：

WARNING:root:The sum 7.0 of memory configured in post launch VMs should not be larger than the calculated total hugepages 1.71875 of service VMs. Please update the configuration in post launch VMs
.....

*Reproduce rate*

     5/5

*Debugging info*

     Cannot to build acrn with release=y for all boards

Partial failure log fyi：

Board XML is fetched from /home/abc/workspace/mainline/acrn-hypervisor/misc/config_tools/data/whl-ipc-i5/whl-ipc-i5.xml
Scenario XML is being fetched from /home/abc/workspace/mainline/acrn-hypervisor/misc/config_tools/data/whl-ipc-i5/shared.xml
WARNING:root:The sum 7.0 of memory configured in post launch VMs should not be larger than the calculated total hugepages 1.71875 of service VMs. Please update the configuration in post launch VMs
/home/abc/workspace/mainline/acrn-hypervisor/build/hypervisor/configs/allocation.xml generated
/home/abc/workspace/mainline/acrn-hypervisor/build/hypervisor/configs/unified.xml generated
/home/abc/workspace/mainline/acrn-hypervisor/build/hypervisor/configs/config.mk generated
scripts/makefile/config.mk:191: The command line sets RELEASE to be 'y', but an existing build is configured with 'n'
scripts/makefile/config.mk:191: The configuration will be modified for RELEASE=y
WARNING:root:The sum 7.0 of memory configured in post launch VMs should not be larger than the calculated total hugepages 1.71875 of service VMs. Please update the configuration in post launch VMs
/home/abc/workspace/mainline/acrn-hypervisor/build/hypervisor/configs/allocation.xml generated
/home/abc/workspace/mainline/acrn-hypervisor/build/hypervisor/configs/unified.xml generated
/home/abc/workspace/mainline/acrn-hypervisor/build/hypervisor/configs/config.mk generated
scripts/makefile/config.mk:191: The command line sets RELEASE to be 'y', but an existing build is configured with 'n'
scripts/makefile/config.mk:191: The configuration will be modified for RELEASE=y
WARNING:root:The sum 7.0 of memory configured in post launch VMs should not be larger than the calculated total hugepages 1.71875 of service VMs. Please update the configuration in post launch VMs

zhangqiang4, 2022-06-01:
[External_System_ID] ACRN-9132

NanlinXie, 2022-06-05:
Close with no regression.

closed 2022-06-05 by NanlinXie