projectacrn/acrn-hypervisor #7683: Passthrough Wifi module init error

hidrol, 2022-06-02:
**Describe the bug**
The wifi device doesn't appear with `ip a` command.
Failed to run INIT ucode: -110

`dmesg |grep iwlwifi` gives:

```
[    4.150315] iwlwifi 0000:00:0b.0: can't derive routing for PCI INT A
[    4.150318] iwlwifi 0000:00:0b.0: PCI INT A: no GSI
[    4.261704] iwlwifi 0000:00:0b.0: loaded firmware version 46.6b541b68.0 op_mode iwlmvm
[    4.346059] iwlwifi 0000:00:0b.0: Detected Intel(R) Wireless-AC 9260 160MHz, REV=0x324
[    9.399867] iwlwifi 0000:00:0b.0: Failed to load firmware chunk!
[    9.399871] iwlwifi 0000:00:0b.0: iwlwifi transaction failed, dumping registers
[    9.399873] iwlwifi 0000:00:0b.0: iwlwifi device config registers:
[    9.400318] iwlwifi 0000:00:0b.0: 00000000: 25268086 00100406 02800029 00000000 80020004 00000000 00
000000 00000000
[    9.400320] iwlwifi 0000:00:0b.0: 00000020: 00000000 00000000 00000000 e0148086 00000000 000000c8 00
000000 00000100
[    9.400323] iwlwifi 0000:00:0b.0: 00000040: 00028010 10008ec0 00100c10 0045e812 10120000 00000000 00
000000 00000000
[    9.400325] iwlwifi 0000:00:0b.0: 00000060: 00000000 00080812 00000005 00000000 00000002 00000000 00
000000 00000000
[    9.400327] iwlwifi 0000:00:0b.0: 00000080: 800f0011 00002000 00003000 00000000 00000000 00000000 00000000 00000000
[    9.400328] iwlwifi 0000:00:0b.0: 000000a0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    9.400330] iwlwifi 0000:00:0b.0: 000000c0: 00000000 00000000 c823d001 0d000008 00804005 00000000 00000000 00000000
[    9.400332] iwlwifi 0000:00:0b.0: 000000e0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    9.400334] iwlwifi 0000:00:0b.0: 00000100: 14c10001 00000000 00000000 00462031 00000000 00002000 00000000 00000000
[    9.400335] iwlwifi 0000:00:0b.0: 00000120: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    9.400337] iwlwifi 0000:00:0b.0: 00000140: 14c00000 ff000000 000000ff 15410018 00000000 0001001e 00481e1f 00000000
[    9.400339] iwlwifi 0000:00:0b.0: iwlwifi device memory mapped registers:
[    9.400372] iwlwifi 0000:00:0b.0: 00000000: 00c89004 00000040 00000000 00000000 00000000 00000000 00027e1f 00000000
[    9.400374] iwlwifi 0000:00:0b.0: 00000020: 00000010 0c040005 00000321 d55555d5 d55555d5 d55555d5 80008040 001f0044
[    9.400383] iwlwifi 0000:00:0b.0: iwlwifi device AER capability structure:
[    9.400449] iwlwifi 0000:00:0b.0: 00000000: 14c10001 00000000 00000000 00462031 00000000 00002000 00000000 00000000
[    9.400451] iwlwifi 0000:00:0b.0: 00000020: 00000000 00000000 00000000
[    9.400453] iwlwifi 0000:00:0b.0: Could not load the [0] uCode section
[    9.400465] iwlwifi 0000:00:0b.0: Failed to start INIT ucode: -110
[    9.400470] iwlwifi 0000:00:0b.0: Collecting data: trigger 16 fired.
[    9.400499] iwlwifi 0000:00:0b.0: Not valid error log pointer 0x00000000 for Init uCode
[    9.400509] iwlwifi 0000:00:0b.0: Fseq Registers:
[    9.400513] iwlwifi 0000:00:0b.0: 0x3E85250B | FSEQ_ERROR_CODE
[    9.400517] iwlwifi 0000:00:0b.0: 0x00000000 | FSEQ_TOP_INIT_VERSION
[    9.400521] iwlwifi 0000:00:0b.0: 0x29FE32E9 | FSEQ_CNVIO_INIT_VERSION
[    9.400525] iwlwifi 0000:00:0b.0: 0x0000A374 | FSEQ_OTP_VERSION
[    9.400529] iwlwifi 0000:00:0b.0: 0x8BECF6EC | FSEQ_TOP_CONTENT_VERSION
[    9.400543] iwlwifi 0000:00:0b.0: 0x0E0A10A2 | FSEQ_ALIVE_TOKEN
[    9.400548] iwlwifi 0000:00:0b.0: 0x2C61A12C | FSEQ_CNVI_ID
[    9.400552] iwlwifi 0000:00:0b.0: 0x5627C2D9 | FSEQ_CNVR_ID
[    9.400556] iwlwifi 0000:00:0b.0: 0x01000200 | CNVI_AUX_MISC_CHIP
[    9.400562] iwlwifi 0000:00:0b.0: 0x01300202 | CNVR_AUX_MISC_CHIP
[    9.400568] iwlwifi 0000:00:0b.0: 0x00004861 | CNVR_SCU_SD_REGS_SD_REG_DIG_DCDC_VTRIM
[    9.400624] iwlwifi 0000:00:0b.0: 0x0BADCAFE | CNVR_SCU_SD_REGS_SD_REG_ACTIVE_VDIG_MIRROR
[    9.400631] iwlwifi 0000:00:0b.0: Firmware not running - cannot dump error
[    9.418918] iwlwifi 0000:00:0b.0: Failed to run INIT ucode: -110
```

**Platform**

Elkhart Lake, Intel Atom(R) x6427FE Processor
Intel® Wireless-AC 9260 

**Scenario**
Partitioning Scenario with Yocto Linux and Device Passthrough
```
Linux intel-corei7-64 5.4.123-rt59-linux-intel-preempt-rt-acrn-uos #1 SMP PREEMPT_RT Wed Jun 1 08:55:53 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

**Expected behavior**
device should appear with `ip a` command under e.g. `wlp0s11` name.

**Additional context**
The Same Setup worked with an older wifi module, Intel Dual Band Wireless-N 7265.
Running the same Yocto Linux natively on the EHL also worked with the Intel 9260.

fuzhongl, 2022-06-06:
@hidrol Thanks for raise this issue; but ACRN doesn't support passthrough WIFI and BT device so far.

hidrol, 2022-06-07:
@fuzhongl Thanks for the reply. In the native setup I use the same kernel as in the partitioning scenario which is `5.4.123-rt59-linux-intel-preempt-rt-acrn-uos`. In the current setup we don't use a service vm (partitioning scenario) and therefore no user vm. I will try to setup the hybrid scenario with user VM available and give you feedback. Do you know when or in which release WIFI and BT devices will be supported for ACRN device passthrough?

fuzhongl, 2022-06-09:
@hidrol If the same kernel is used in native and partitioning scenario; it seems like the firmware of wifi isn't built into kernel, but in rootfs.
Do you use the same rootfs for native and partitioning scenario?
Could you please share the kernel config for native and partitioning scenario?

hidrol, 2022-06-09:
@fuzhongl  Yes I use the same rootfs for native and partitioning scenario.
These are my kernel configs:

### native
`rw rootwait root=/dev/sda2 console=tty0  noxsave nohpet no_timer_check ignore_loglevel log_buf_len=16M consoleblank=0 tsc=reliable swiotlb=131072 modprobe.blacklist=pinctrl_elkhartlake,intel_ehl_gpio,stmmac_pci`

### virtualized
`rw rootwait root=/dev/sda2 console=ttyS0,115200 earlyprintk=serial,ttyS0,115200 rdinit=/sbin/init swiotlb=131072 no_ipi_broadcast=1 noxsave nohpet no_timer_check ignore_loglevel clocksource=tsc x2apic_phys processor.max_cstate=0 intel_idle.max_cstate=0 intel_pstate=disable mce=ignore_ce audit=0 isolcpus=nohz,domain,1 nohz_full=1 rcu_nocbs=1 nosoftlockup idle=poll irqaffinity=0 consoleblank=0 tsc=reliable reboot=acpi`

hidrol, 2022-06-09:
@fuzhongl Yes I use the same rootfs for native and partitioning scenario.
I used the same kernel configs for native and partitioning scenario. But  I used different boot parameters.

### native

rw rootwait root=/dev/sda2 console=tty0 noxsave nohpet no_timer_check ignore_loglevel log_buf_len=16M consoleblank=0 tsc=reliable swiotlb=131072 modprobe.blacklist=pinctrl_elkhartlake,intel_ehl_gpio,stmmac_pci

### virtualized

rw rootwait root=/dev/sda2 console=ttyS0,115200 earlyprintk=serial,ttyS0,115200 rdinit=/sbin/init swiotlb=131072 no_ipi_broadcast=1 noxsave nohpet no_timer_check ignore_loglevel clocksource=tsc x2apic_phys processor.max_cstate=0 intel_idle.max_cstate=0 intel_pstate=disable mce=ignore_ce audit=0 isolcpus=nohz,domain,1 nohz_full=1 rcu_nocbs=1 nosoftlockup idle=poll irqaffinity=0 consoleblank=0 tsc=reliable reboot=acpi

hidrol, 2022-06-10:
@fuzhongl  this is our kernel config:
```
#
# Automatically generated file; DO NOT EDIT.
# Linux/x86 5.4.123 Kernel Configuration
#

#
# Compiler: x86_64-oe-linux-gcc (GCC) 11.2.0
#
CONFIG_CC_IS_GCC=y
CONFIG_GCC_VERSION=110200
CONFIG_CLANG_VERSION=0
CONFIG_CC_HAS_ASM_GOTO=y
CONFIG_CC_HAS_ASM_INLINE=y
CONFIG_IRQ_WORK=y
CONFIG_BUILDTIME_EXTABLE_SORT=y
CONFIG_THREAD_INFO_IN_TASK=y

#
# General setup
#
CONFIG_INIT_ENV_ARG_LIMIT=32
# CONFIG_COMPILE_TEST is not set
CONFIG_LOCALVERSION="-linux-intel-preempt-rt-acrn-uos"
# CONFIG_LOCALVERSION_AUTO is not set
CONFIG_BUILD_SALT=""
CONFIG_HAVE_KERNEL_GZIP=y
CONFIG_HAVE_KERNEL_BZIP2=y
CONFIG_HAVE_KERNEL_LZMA=y
CONFIG_HAVE_KERNEL_XZ=y
CONFIG_HAVE_KERNEL_LZO=y
CONFIG_HAVE_KERNEL_LZ4=y
CONFIG_KERNEL_GZIP=y
# CONFIG_KERNEL_BZIP2 is not set
# CONFIG_KERNEL_LZMA is not set
# CONFIG_KERNEL_XZ is not set
# CONFIG_KERNEL_LZO is not set
# CONFIG_KERNEL_LZ4 is not set
CONFIG_DEFAULT_HOSTNAME="(none)"
CONFIG_SWAP=y
CONFIG_SYSVIPC=y
CONFIG_SYSVIPC_SYSCTL=y
CONFIG_POSIX_MQUEUE=y
CONFIG_POSIX_MQUEUE_SYSCTL=y
CONFIG_CROSS_MEMORY_ATTACH=y
CONFIG_USELIB=y
# CONFIG_AUDIT is not set
CONFIG_HAVE_ARCH_AUDITSYSCALL=y

#
# IRQ subsystem
#
CONFIG_GENERIC_IRQ_PROBE=y
CONFIG_GENERIC_IRQ_SHOW=y
CONFIG_GENERIC_IRQ_EFFECTIVE_AFF_MASK=y
CONFIG_GENERIC_PENDING_IRQ=y
CONFIG_GENERIC_IRQ_MIGRATION=y
CONFIG_IRQ_DOMAIN=y
CONFIG_IRQ_DOMAIN_HIERARCHY=y
CONFIG_GENERIC_MSI_IRQ=y
CONFIG_GENERIC_MSI_IRQ_DOMAIN=y
CONFIG_GENERIC_IRQ_MATRIX_ALLOCATOR=y
CONFIG_GENERIC_IRQ_RESERVATION_MODE=y
CONFIG_IRQ_FORCED_THREADING=y
CONFIG_SPARSE_IRQ=y
# CONFIG_GENERIC_IRQ_DEBUGFS is not set
# end of IRQ subsystem

CONFIG_CLOCKSOURCE_WATCHDOG=y
[... 7279 lines cut ...]
CONFIG_EARLY_PRINTK=y
# CONFIG_EARLY_PRINTK_DBGP is not set
# CONFIG_EARLY_PRINTK_USB_XDBC is not set
# CONFIG_X86_PTDUMP is not set
# CONFIG_EFI_PGT_DUMP is not set
# CONFIG_DEBUG_WX is not set
CONFIG_DOUBLEFAULT=y
# CONFIG_DEBUG_TLBFLUSH is not set
CONFIG_HAVE_MMIOTRACE_SUPPORT=y
# CONFIG_X86_DECODER_SELFTEST is not set
CONFIG_IO_DELAY_0X80=y
# CONFIG_IO_DELAY_0XED is not set
# CONFIG_IO_DELAY_UDELAY is not set
# CONFIG_IO_DELAY_NONE is not set
# CONFIG_DEBUG_BOOT_PARAMS is not set
# CONFIG_CPA_DEBUG is not set
# CONFIG_DEBUG_ENTRY is not set
# CONFIG_DEBUG_NMI_SELFTEST is not set
CONFIG_X86_DEBUG_FPU=y
# CONFIG_PUNIT_ATOM_DEBUG is not set
# CONFIG_UNWINDER_ORC is not set
CONFIG_UNWINDER_FRAME_POINTER=y
# CONFIG_UNWINDER_GUESS is not set
# end of Kernel hacking
```

fuzhongl, 2022-06-17:
@hidrol 
Checked you shared kernel config; the firmware isn't built into kernel:
`CONFIG_EXTRA_FIRMWARE=""`
https://www.kernel.org/doc/Documentation/driver-api/firmware/built-in-fw.rst
Please double check if the firmware which matches with your kernel is included in your rootfs.

hidrol, 2022-06-21:
@fuzhongl I enabled `CONFIG_EXTRA_FIRMAWARE` but still got the same error message

gvancuts, 2022-06-21:
Can you share the new kernel config and dmesg logs?

hidrol, 2022-06-21:
```
#
# Automatically generated file; DO NOT EDIT.
# Linux/x86 5.4.123 Kernel Configuration
#

#
# Compiler: x86_64-oe-linux-gcc (GCC) 11.2.0
#
CONFIG_CC_IS_GCC=y
CONFIG_GCC_VERSION=110200
CONFIG_CLANG_VERSION=0
CONFIG_CC_HAS_ASM_GOTO=y
CONFIG_CC_HAS_ASM_INLINE=y
CONFIG_IRQ_WORK=y
CONFIG_BUILDTIME_EXTABLE_SORT=y
CONFIG_THREAD_INFO_IN_TASK=y

#
# General setup
#
CONFIG_INIT_ENV_ARG_LIMIT=32
# CONFIG_COMPILE_TEST is not set
CONFIG_LOCALVERSION="-linux-intel-preempt-rt-acrn-uos"
# CONFIG_LOCALVERSION_AUTO is not set
CONFIG_BUILD_SALT=""
CONFIG_HAVE_KERNEL_GZIP=y
CONFIG_HAVE_KERNEL_BZIP2=y
CONFIG_HAVE_KERNEL_LZMA=y
CONFIG_HAVE_KERNEL_XZ=y
CONFIG_HAVE_KERNEL_LZO=y
CONFIG_HAVE_KERNEL_LZ4=y
CONFIG_KERNEL_GZIP=y
# CONFIG_KERNEL_BZIP2 is not set
# CONFIG_KERNEL_LZMA is not set
# CONFIG_KERNEL_XZ is not set
# CONFIG_KERNEL_LZO is not set
# CONFIG_KERNEL_LZ4 is not set
CONFIG_DEFAULT_HOSTNAME="(none)"
CONFIG_SWAP=y
CONFIG_SYSVIPC=y
CONFIG_SYSVIPC_SYSCTL=y
CONFIG_POSIX_MQUEUE=y
CONFIG_POSIX_MQUEUE_SYSCTL=y
CONFIG_CROSS_MEMORY_ATTACH=y
CONFIG_USELIB=y
# CONFIG_AUDIT is not set
CONFIG_HAVE_ARCH_AUDITSYSCALL=y

#
# IRQ subsystem
#
CONFIG_GENERIC_IRQ_PROBE=y
CONFIG_GENERIC_IRQ_SHOW=y
CONFIG_GENERIC_IRQ_EFFECTIVE_AFF_MASK=y
CONFIG_GENERIC_PENDING_IRQ=y
CONFIG_GENERIC_IRQ_MIGRATION=y
CONFIG_IRQ_DOMAIN=y
CONFIG_IRQ_DOMAIN_HIERARCHY=y
CONFIG_GENERIC_MSI_IRQ=y
CONFIG_GENERIC_MSI_IRQ_DOMAIN=y
CONFIG_GENERIC_IRQ_MATRIX_ALLOCATOR=y
CONFIG_GENERIC_IRQ_RESERVATION_MODE=y
CONFIG_IRQ_FORCED_THREADING=y
CONFIG_SPARSE_IRQ=y
# CONFIG_GENERIC_IRQ_DEBUGFS is not set
# end of IRQ subsystem

CONFIG_CLOCKSOURCE_WATCHDOG=y
CONFIG_ARCH_CLOCKSOURCE_DATA=y
[... 7279 lines cut ...]
CONFIG_EARLY_PRINTK=y
# CONFIG_EARLY_PRINTK_DBGP is not set
# CONFIG_EARLY_PRINTK_USB_XDBC is not set
# CONFIG_X86_PTDUMP is not set
# CONFIG_EFI_PGT_DUMP is not set
# CONFIG_DEBUG_WX is not set
CONFIG_DOUBLEFAULT=y
# CONFIG_DEBUG_TLBFLUSH is not set
CONFIG_HAVE_MMIOTRACE_SUPPORT=y
# CONFIG_X86_DECODER_SELFTEST is not set
CONFIG_IO_DELAY_0X80=y
# CONFIG_IO_DELAY_0XED is not set
# CONFIG_IO_DELAY_UDELAY is not set
# CONFIG_IO_DELAY_NONE is not set
# CONFIG_DEBUG_BOOT_PARAMS is not set
# CONFIG_CPA_DEBUG is not set
# CONFIG_DEBUG_ENTRY is not set
# CONFIG_DEBUG_NMI_SELFTEST is not set
CONFIG_X86_DEBUG_FPU=y
# CONFIG_PUNIT_ATOM_DEBUG is not set
# CONFIG_UNWINDER_ORC is not set
CONFIG_UNWINDER_FRAME_POINTER=y
# CONFIG_UNWINDER_GUESS is not set
# end of Kernel hacking
```

gvancuts, 2022-06-22:
Thanks @hidrol , I think you mentionned this but can you double-confirm that this **exact same** version of the firmware works in a native environment (Yocto or other)?

hidrol, 2022-06-22:
@fuzhongl @gvancuts yes the exact **same firmware** works in native yocto.
```
dmesg |grep iwlwifi
[    6.431787] iwlwifi 0000:01:00.0: loaded firmware version 46.9d0122c0.0 op_mode iwlmvm
[    6.494278] iwlwifi 0000:01:00.0: Detected Intel(R) Wireless-AC 9260 160MHz, REV=0x324
[    6.547294] iwlwifi 0000:01:00.0: base HW address: 8c:8d:28:1e:f0:e1
[    6.857533] iwlwifi 0000:01:00.0 wlp1s0: renamed from wlan0
```

gvancuts, 2022-06-22:
Thanks @hidrol . I'm out of my depth here quite frankly, but I have seen reports of similar issues that were related to the use of `intel_iommu=off` or not. Can you try add that to your Service VM kernel command-line parameter and see if it makes any difference? (Disclaimer: I do not know what other effects this could have on the system, let's just treat that as a test for now ;-))

gvancuts, 2022-06-22:
> @gvancuts unfortunately I have the same error with using `intel_iommu=off`.

Thanks for checking this @hidrol . I'm not sure what else to trym @fuzhongl , do you have any idea how we can further debug this?

gvancuts, 2022-06-22:
One thing we could do to have more debug info is this:
Prints

The simplest way to provide minimal output is to dump your kernel log: dmesg. Sometimes we will ask for logs from the supplicant too - they typically land in syslog. iwlwifi can print more data to the kernel log if asked to: this is controlled by the debug module parameter. This is a [bitmap](https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/tree/drivers/net/wireless/intel/iwlwifi/iwl-debug.h#n129). To see more debug prints, [CONFIG_IWLWIFI_DEBUG](https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/tree/drivers/net/wireless/intel/iwlwifi/Kconfig#n105) must be enabled. Please don't add debug level unless instructed to do so. Adding more prints typically adds useless information.

Not sure we can interpret all the debug messages but this may shed some new lights on the problem.

fuzhongl, 2022-06-23:
> > @gvancuts unfortunately I have the same error with using `intel_iommu=off`.
> 
> Thanks for checking this @hidrol . I'm not sure what else to trym @fuzhongl , do you have any idea how we can further debug this?

@hidrol Could you help to try 

> @fuzhongl @gvancuts yes the exact **same firmware** works in native yocto.
>
> ```
> dmesg |grep iwlwifi
> [    6.431787] iwlwifi 0000:01:00.0: loaded firmware version 46.9d0122c0.0 op_mode iwlmvm
> [ [...]

@hidrol With the same kernel/rootfs, the bdf is **0000:01:00.0** in native, but it is **0000:00:0b.0** in pre_launched VM.
Do you generate the board.xml and scenario.xml with the same codebase?
Could you please share the board.xml and scenario.xml?
Thanks!

hidrol, 2022-06-23:
I wonder why the bdf is 0000:00:0b.0. I would have expected that the bdf would be 0000:01:00.0 also in virtualized mode.
I got the bdf from the pci command in the slim boot loader shell. And that's also the bdf I put in the pci_devs section of the scenario xml.

```
        <pci_devs>
            <pci_dev desc="pci device">00:14.0 USB controller: Intel Corporation Device 4b7d</pci_dev>
            <pci_dev desc="pci device">00:1d.1 Ethernet controller: Intel Corporation Device 4ba0</pci_dev>
            <pci_dev desc="pci device">00:1d.2 Ethernet controller: Intel Corporation Device 4bb0</pci_dev>
            <pci_dev desc="pci device">00:1e.4 Ethernet controller: Intel Corporation Device 4b32</pci_dev>
            <pci_dev desc="pci device">01:00.0 Network controller: Intel Corporation Device 2526</pci_dev>
        </pci_devs>
```
I got the board.xml from Intel and never touched it.

fuzhongl, 2022-06-24:
@hidrol It seems like the the wifi device doesn't has MSI/MSI-X capability.

```
[    4.150315] iwlwifi 0000:00:0b.0: can't derive routing for PCI INT A
[    4.150318] iwlwifi 0000:00:0b.0: PCI INT A: no GSI
```

Please help to add following into the scenario.xml

```
    <pt_intx>
            (X, 0)
        </pt_intx>
```

X is the interrupt number used in native; 0 is the interrupt number used in pre_launched VM.

hidrol, 2022-06-27:
How should I assign the interrupt numbers. I tried a few different mappings, but nothing worked.
I tried the following mappings, but ACRN didn't boot.
```
            <pci_dev desc="pci device">01:00.0 Network controller: Intel Corporation Device 2526</pci_dev>
              <pt_intx>
                (127, 5)
                (128, 6)
                (129, 7)
                (130, 8)
                (131, 9)
                (132, 10)
              </pt_intx>
```

### native interrupts:
```
           CPU0       CPU1       CPU2       CPU3       
   4:         12          0          0          0  IR-IO-APIC    4-edge      ttyS0
   8:          0          0          0          0  IR-IO-APIC    8-edge      rtc0
  16:          0          0         97          0  IR-IO-APIC   16-fasteoi   i2c_designware.0, idma64.6, i801_smbus, mmc0
  17:          0          0          0          0  IR-IO-APIC   17-fasteoi   i2c_designware.1, idma64.7
  27:          0          0          0          0  IR-IO-APIC   27-fasteoi   idma64.2, i2c_designware.2
  29:          0          0          0          0  IR-IO-APIC   29-fasteoi   idma64.3, i2c_designware.3
  30:          0          0          0          0  IR-IO-APIC   30-fasteoi   idma64.4, i2c_designware.4
  31:          0          0          0          0  IR-IO-APIC   31-fasteoi   idma64.5, i2c_designware.5
 120:          0          0          0          0  DMAR-MSI    0-edge      dmar0
 121:          0          0          0          0  DMAR-MSI    1-edge      dmar1
 122:          0          0          0          0  IR-PCI-MSI 458752-edge      PCIe PME, aerdrv
 123:          0      12239          0          0  IR-PCI-MSI 327680-edge      xhci_hcd
 124:          0          0          0        539  IR-PCI-MSI 32768-edge      i915
 125:          0          0          0          0  IR-PCI-MSI 393216-edge      0000:00:18.0
 126:          0         40          0          0  IR-PCI-MSI 360448-edge      mei_me
 127:          0          0         30          0  IR-PCI-MSI 524288-edge      iwlwifi: default queue
 128:          0          0          0          0  IR-PCI-MSI 524289-edge      iwlwifi: queue 1
 129:          0          0          0          0  IR-PCI-MSI 524290-edge      iwlwifi: queue 2
 130:          0          0          0          0  IR-PCI-MSI 524291-edge      iwlwifi: queue 3
 131:          0          0          0          0  IR-PCI-MSI 524292-edge      iwlwifi: queue 4
 132:          0          0          0         11  IR-PCI-MSI 524293-edge      iwlwifi: exception
 NMI:          0          0          0          0   Non-maskable interrupts
 LOC:      21331      22073      21384      21647   Local timer interrupts
 SPU:          0          0          0          0   Spurious interrupts
 PMI:          0          0          0          0   Performance monitoring interrupts
 IWI:          0          0          0          0   IRQ work interrupts
 RTR:          0          0          0          0   APIC ICR read retries
 RES:       8177       6756       4232       4510   Rescheduling interrupts
 CAL:        421        488        568        475   Function call interrupts
 TLB:          9          3          1          4   TLB shootdowns
 TRM:          0          0          0          0   Thermal event interrupts
 THR:          0          0          0          0   Threshold APIC interrupts
 DFR:          0          0          0          0   Deferred Error APIC interrupts
 MCE:          0          0          0          0   Machine check exceptions
 MCP:          1          2          2          2   Machine check polls
 ERR:          0
 MIS:          0
 PIN:          0          0          0          0   Posted-interrupt notification event
 NPI:          0          0          0          0   Nested posted-interrupt event
 PIW:          0          0          0          0   Posted-interrupt wakeup event
```
### virtualized interrupts
```
          CPU0       CPU1
  1:       3089          0   IO-APIC   4-edge      ttyS0
  3:          0          0   IO-APIC   8-edge      rtc0
  4:      12296          0   PCI-MSI 114688-edge      xhci_hcd
  5:          0          0   PCI-MSI 180224-edge      iwlwifi: default queue
  6:          0          0   PCI-MSI 180225-edge      iwlwifi: queue 1
  7:          0          0   PCI-MSI 180226-edge      iwlwifi: queue 2
  8:          0          0   PCI-MSI 180227-edge      iwlwifi: exception
  9:          0          0   PCI-MSI 131072-edge      enp0s8f0:rx-0
 10:          1          0   PCI-MSI 131073-edge      enp0s8f0:tx-0
[... 54 lines cut ...]
 99:          0          0   PCI-MSI 163866-edge      enp0s10:safety-ue
100:          0          0   PCI-MSI 163867-edge      enp0s10:safety-ce
101:          0          0   PCI-MSI 163868-edge      enp0s10:lpi
102:          0          0   PCI-MSI 163869-edge      enp0s10:mac
NMI:          0          0   Non-maskable interrupts
LOC:      50365      34756   Local timer interrupts
SPU:          0          0   Spurious interrupts
PMI:          0          0   Performance monitoring interrupts
IWI:          0          6   IRQ work interrupts
RTR:          0          0   APIC ICR read retries
RES:          5          1   Rescheduling interrupts
CAL:          0        463   Function call interrupts
TLB:          0          0   TLB shootdowns
TRM:          0          0   Thermal event interrupts
THR:          0          0   Threshold APIC interrupts
DFR:          0          0   Deferred Error APIC interrupts
MCE:          0          0   Machine check exceptions
MCP:          1          1   Machine check polls
HYP:          0          0   Hypervisor callback interrupts
ERR:          0
MIS:          0
PIN:          0          0   Posted-interrupt notification event
NPI:          0          0   Nested posted-interrupt event
PIW:          0          0   Posted-interrupt wakeup event
```

fuzhongl, 2022-06-28:
@hidrol Could you please help to double check if this wifi device have MSI/MSIX capability?
Thanks!

fuzhongl, 2022-06-30:
@hidrol What' the value in your scenario.xml?
`<MAX_EMULATED_MMIO>16</MAX_EMULATED_MMIO>`
Please help to change 16 to 32; and have a try.
If still fail, please share your board.xml/scenario.xml/dmesg log in native/dmesg log in pre_launched VM/lspci -s B:D.F -vvvvxxxx in pre_launched vm.
Thanks!

hidrol, 2022-06-30:
@fuzhongl I tried changing `<MAX_EMULATED_MMIO>16</MAX_EMULATED_MMIO>` to `<MAX_EMULATED_MMIO>32</MAX_EMULATED_MMIO>`. Unfortunately it still fails.

### dmesg in virtual environment
```
root@intel-corei7-64:~# dmesg |grep iwlwifi
[    3.622293] iwlwifi 0000:00:0b.0: can't derive routing for PCI INT A
[    3.622295] iwlwifi 0000:00:0b.0: PCI INT A: no GSI
[    3.709482] iwlwifi 0000:00:0b.0: loaded firmware version 46.fae53a8b.0 op_mode iwlmvm
[    3.807461] iwlwifi 0000:00:0b.0: Detected Intel(R) Wireless-AC 9260 160MHz, REV=0x324
[   10.940098] iwlwifi 0000:00:0b.0: Failed to load firmware chunk!
[   10.940105] iwlwifi 0000:00:0b.0: iwlwifi transaction failed, dumping registers
[   10.940107] iwlwifi 0000:00:0b.0: iwlwifi device config registers:
[   10.940589] iwlwifi 0000:00:0b.0: 00000000: 25268086 00100406 02800029 00000000 80020004 00000000 00000000 00000000
[   10.940593] iwlwifi 0000:00:0b.0: 00000020: 00000000 00000000 00000000 e0148086 00000000 000000c8 00000000 00000100
[   10.940595] iwlwifi 0000:00:0b.0: 00000040: 00028010 10008ec0 00100c10 0045e812 10120000 00000000 00000000 00000000
[   10.940597] iwlwifi 0000:00:0b.0: 00000060: 00000000 00080812 00000005 00000000 00000002 00000000 00000000 00000000
[   10.940599] iwlwifi 0000:00:0b.0: 00000080: 800f0011 00002000 00003000 00000000 00000000 00000000 00000000 00000000
[   10.940601] iwlwifi 0000:00:0b.0: 000000a0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   10.940602] iwlwifi 0000:00:0b.0: 000000c0: 00000000 00000000 c823d001 0d000008 00804005 00000000 00000000 00000000
[   10.940604] iwlwifi 0000:00:0b.0: 000000e0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   10.940606] iwlwifi 0000:00:0b.0: 00000100: 14c10001 00000000 00000000 00462031 00000000 00002000 00000000 00000000
[   10.940608] iwlwifi 0000:00:0b.0: 00000120: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   10.940610] iwlwifi 0000:00:0b.0: 00000140: 14c00000 ff000000 000000ff 15410018 00000000 0001001e 00481e1f 00000000
[   10.940612] iwlwifi 0000:00:0b.0: iwlwifi device memory mapped registers:
[   10.940644] iwlwifi 0000:00:0b.0: 00000000: 18c89004 00000040 00000000 00000000 00000000 00000000 00027e1f 00000000
[   10.940646] iwlwifi 0000:00:0b.0: 00000020: 00000010 0c040005 00000321 d55555d5 d55555d5 d55555d5 80008040 001f0044
[   10.940655] iwlwifi 0000:00:0b.0: iwlwifi device AER capability structure:
[   10.940724] iwlwifi 0000:00:0b.0: 00000000: 14c10001 00000000 00000000 00462031 00000000 00002000 00000000 00000000
[   10.940726] iwlwifi 0000:00:0b.0: 00000020: 00000000 00000000 00000000
[   10.940729] iwlwifi 0000:00:0b.0: Could not load the [0] uCode section
[   10.940741] iwlwifi 0000:00:0b.0: Failed to start INIT ucode: -110
[   10.940747] iwlwifi 0000:00:0b.0: Collecting data: trigger 16 fired.
[   11.190384] iwlwifi 0000:00:0b.0: Not valid error log pointer 0x00000000 for Init uCode
[   11.190401] iwlwifi 0000:00:0b.0: Fseq Registers:
[   11.190406] iwlwifi 0000:00:0b.0: 0x3A85250B | FSEQ_ERROR_CODE
[   11.190410] iwlwifi 0000:00:0b.0: 0x00000000 | FSEQ_TOP_INIT_VERSION
[   11.190414] iwlwifi 0000:00:0b.0: 0x29FE32ED | FSEQ_CNVIO_INIT_VERSION
[   11.190418] iwlwifi 0000:00:0b.0: 0x0000A374 | FSEQ_OTP_VERSION
[   11.190423] iwlwifi 0000:00:0b.0: 0x8BECE6EC | FSEQ_TOP_CONTENT_VERSION
[   11.190426] iwlwifi 0000:00:0b.0: 0x0E0A50E2 | FSEQ_ALIVE_TOKEN
[   11.190431] iwlwifi 0000:00:0b.0: 0x7C41A524 | FSEQ_CNVI_ID
[   11.190435] iwlwifi 0000:00:0b.0: 0x5663C2D9 | FSEQ_CNVR_ID
[   11.190439] iwlwifi 0000:00:0b.0: 0x01000200 | CNVI_AUX_MISC_CHIP
[   11.190445] iwlwifi 0000:00:0b.0: 0x01300202 | CNVR_AUX_MISC_CHIP
[   11.190451] iwlwifi 0000:00:0b.0: 0x00004861 | CNVR_SCU_SD_REGS_SD_REG_DIG_DCDC_VTRIM
[   11.190518] iwlwifi 0000:00:0b.0: 0x0BADCAFE | CNVR_SCU_SD_REGS_SD_REG_ACTIVE_VDIG_MIRROR
[   11.190545] iwlwifi 0000:00:0b.0: Firmware not running - cannot dump error
[   11.548176] iwlwifi 0000:00:0b.0: Failed to run INIT ucode: -110
```

### lspci in virtual environment
```
root@intel-corei7-64:~# lspci -s 00:0b.0 -vvvvxxxx
00:0b.0 Network controller: Intel Corporation Wireless-AC 9260 (rev 29)
        Subsystem: Intel Corporation Wireless-AC 9260
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 0
        Region 0: Memory at 80020000 (64-bit, non-prefetchable) [size=16K]
        Capabilities: [c8] Power Management version 3
                Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [d0] MSI: Enable- Count=1/1 Maskable- 64bit+
                Address: 0000000000000000  Data: 0000
        Capabilities: [40] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0, Latency L0s <512ns, L1 unlimited
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0.000W
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+ FLReset-
[... 2944 lines cut ...]
      </device>
      <device id="PNP0C0B">
        <acpi_object>\_TZ_.FAN1</acpi_object>
        <acpi_uid>1</acpi_uid>
[... cut ...]
      </device>
      <device id="PNP0C0B">
        <acpi_object>\_TZ_.FAN3</acpi_object>
        <acpi_uid>3</acpi_uid>
        <aml_template>5b821c5c2e5f545a5f46414e33085f4849440c41d00c0b085f5549440a03</aml_template>
      </device>
      <device id="PNP0C0B">
        <acpi_object>\_TZ_.FAN4</acpi_object>
        <acpi_uid>4</acpi_uid>
        <aml_template>5b821c5c2e5f545a5f46414e34085f4849440c41d00c0b085f5549440a04</aml_template>
      </device>
    </thermalzone>
  </devices>
</acrn-config>
```

hidrol, 2022-08-10:
Hallo @fuzhongl. I got a new Wifi Module from Intel which is the Intel W-Fi 6E AX210.
This time the firmware gets loaded but still got error messages (both with acrn and native).

```
root@intel-corei7-64:~# dmesg |grep iwlwifi
[    6.737645] iwlwifi 0000:00:0b.0: can't derive routing for PCI INT A
[    6.737648] iwlwifi 0000:00:0b.0: PCI INT A: no GSI
[    7.023533] iwlwifi 0000:00:0b.0: api flags index 2 larger than supported by driver
[    7.023585] iwlwifi 0000:00:0b.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.63.2.2
[    7.026264] iwlwifi 0000:00:0b.0: loaded firmware version 66.f1c864e0.0 ty-a0-gf-a0-66.ucode op_mode iwlmvm
[    7.476496] iwlwifi 0000:00:0b.0: Detected Intel(R) Wi-Fi 6 AX210 160MHz, REV=0x420
[    8.509865] iwlwifi 0000:00:0b.0: SecBoot CPU1 Status: 0x7eba, CPU2 Status: 0xb03
[    8.509880] iwlwifi 0000:00:0b.0: UMAC PC: 0xc0080b80
[    8.509894] iwlwifi 0000:00:0b.0: LMAC PC: 0x4d0cce
[    8.509899] iwlwifi 0000:00:0b.0: WRT: Collecting data: ini trigger 13 fired (delay=0ms).
[    8.510029] iwlwifi 0000:00:0b.0: Start IWL Error Log Dump:
[    8.510031] iwlwifi 0000:00:0b.0: Transport status: 0x00000042, valid: 6
[    8.510034] iwlwifi 0000:00:0b.0: Loaded firmware version: 66.f1c864e0.0 ty-a0-gf-a0-66.ucode
[    8.510036] iwlwifi 0000:00:0b.0: 0x00000084 | NMI_INTERRUPT_UNKNOWN
[    8.510039] iwlwifi 0000:00:0b.0: 0x002002F0 | trm_hw_status0
[    8.510040] iwlwifi 0000:00:0b.0: 0x00000000 | trm_hw_status1
[    8.510042] iwlwifi 0000:00:0b.0: 0x004DA722 | branchlink2
[    8.510043] iwlwifi 0000:00:0b.0: 0x004D0CCE | interruptlink1
[    8.510045] iwlwifi 0000:00:0b.0: 0x004D0CCE | interruptlink2
[    8.510046] iwlwifi 0000:00:0b.0: 0x004D94DA | data1
[    8.510047] iwlwifi 0000:00:0b.0: 0x01000000 | data2
[    8.510049] iwlwifi 0000:00:0b.0: 0x00000000 | data3
[    8.510050] iwlwifi 0000:00:0b.0: 0x00000000 | beacon time
[    8.510052] iwlwifi 0000:00:0b.0: 0x000E271C | tsf low
[    8.510053] iwlwifi 0000:00:0b.0: 0x00000000 | tsf hi
[    8.510055] iwlwifi 0000:00:0b.0: 0x00000000 | time gp1
[    8.510056] iwlwifi 0000:00:0b.0: 0x000F67B1 | time gp2
[    8.510057] iwlwifi 0000:00:0b.0: 0x00000001 | uCode revision type
[    8.510059] iwlwifi 0000:00:0b.0: 0x00000042 | uCode version major
[    8.510060] iwlwifi 0000:00:0b.0: 0xF1C864E0 | uCode version minor
[    8.510062] iwlwifi 0000:00:0b.0: 0x00000420 | hw version
[    8.510063] iwlwifi 0000:00:0b.0: 0x00C89002 | board version
[    8.510065] iwlwifi 0000:00:0b.0: 0x8004FC12 | hcmd
[    8.510066] iwlwifi 0000:00:0b.0: 0x00020000 | isr0
[    8.510067] iwlwifi 0000:00:0b.0: 0x00000000 | isr1
[    8.510069] iwlwifi 0000:00:0b.0: 0x00F00002 | isr2
[    8.510070] iwlwifi 0000:00:0b.0: 0x00C0000C | isr3
[    8.510071] iwlwifi 0000:00:0b.0: 0x00000000 | isr4
[    8.510072] iwlwifi 0000:00:0b.0: 0x00000000 | last cmd Id
[    8.510074] iwlwifi 0000:00:0b.0: 0x004D94DA | wait_event
[    8.510075] iwlwifi 0000:00:0b.0: 0x00000000 | l2p_control
[    8.510076] iwlwifi 0000:00:0b.0: 0x00000000 | l2p_duration
[    8.510078] iwlwifi 0000:00:0b.0: 0x00000000 | l2p_mhvalid
[    8.510079] iwlwifi 0000:00:0b.0: 0x00000000 | l2p_addr_match
[    8.510081] iwlwifi 0000:00:0b.0: 0x00000009 | lmpm_pmg_sel
[    8.510082] iwlwifi 0000:00:0b.0: 0x00000000 | timestamp
[    8.510084] iwlwifi 0000:00:0b.0: 0x001C0000 | flow_handler
[    8.510123] iwlwifi 0000:00:0b.0: 0x00000000 | ADVANCED_SYSASSERT
[    8.510124] iwlwifi 0000:00:0b.0: 0x00000000 | umac branchlink1
[    8.510126] iwlwifi 0000:00:0b.0: 0x00000000 | umac branchlink2
[    8.510127] iwlwifi 0000:00:0b.0: 0x00000000 | umac interruptlink1
[    8.510129] iwlwifi 0000:00:0b.0: 0x00000000 | umac interruptlink2
[    8.510130] iwlwifi 0000:00:0b.0: 0x00000000 | umac data1
[    8.510131] iwlwifi 0000:00:0b.0: 0x00000000 | umac data2
[    8.510133] iwlwifi 0000:00:0b.0: 0x00000000 | umac data3
[    8.510134] iwlwifi 0000:00:0b.0: 0x00000000 | umac major
[    8.510135] iwlwifi 0000:00:0b.0: 0x00000000 | umac minor
[    8.510136] iwlwifi 0000:00:0b.0: 0x00000000 | frame pointer
[    8.510138] iwlwifi 0000:00:0b.0: 0x00000000 | stack pointer
[    8.510139] iwlwifi 0000:00:0b.0: 0x00000000 | last host cmd
[    8.510140] iwlwifi 0000:00:0b.0: 0x00000000 | isr status reg
[    8.510153] iwlwifi 0000:00:0b.0: IML/ROM dump:
[    8.510154] iwlwifi 0000:00:0b.0: 0x00000B03 | IML/ROM error/state
[    8.510166] iwlwifi 0000:00:0b.0: 0x00007EBA | IML/ROM data1
[    8.510178] iwlwifi 0000:00:0b.0: 0x00000080 | IML/ROM WFPM_AUTH_KEY_0
[    8.510186] iwlwifi 0000:00:0b.0: Fseq Registers:
[    8.510190] iwlwifi 0000:00:0b.0: 0x60000000 | FSEQ_ERROR_CODE
[    8.510194] iwlwifi 0000:00:0b.0: 0x00440003 | FSEQ_TOP_INIT_VERSION
[    8.510199] iwlwifi 0000:00:0b.0: 0x00080009 | FSEQ_CNVIO_INIT_VERSION
[    8.510203] iwlwifi 0000:00:0b.0: 0x0000A652 | FSEQ_OTP_VERSION
[    8.510207] iwlwifi 0000:00:0b.0: 0x00000002 | FSEQ_TOP_CONTENT_VERSION
[    8.510212] iwlwifi 0000:00:0b.0: 0x4552414E | FSEQ_ALIVE_TOKEN
[    8.510216] iwlwifi 0000:00:0b.0: 0x00400410 | FSEQ_CNVI_ID
[    8.510220] iwlwifi 0000:00:0b.0: 0x00400410 | FSEQ_CNVR_ID
[    8.510224] iwlwifi 0000:00:0b.0: 0x00400410 | CNVI_AUX_MISC_CHIP
[    8.510230] iwlwifi 0000:00:0b.0: 0x00400410 | CNVR_AUX_MISC_CHIP
[    8.510236] iwlwifi 0000:00:0b.0: 0x00009061 | CNVR_SCU_SD_REGS_SD_REG_DIG_DCDC_VTRIM
[    8.510243] iwlwifi 0000:00:0b.0: 0x00000061 | CNVR_SCU_SD_REGS_SD_REG_ACTIVE_VDIG_MIRROR
[    8.510247] iwlwifi 0000:00:0b.0: Failed to start RT ucode: -110
[    8.510251] iwlwifi 0000:00:0b.0: WRT: Collecting data: ini trigger 13 fired (delay=0ms).
[    9.875522] iwlwifi 0000:00:0b.0: Failed to run INIT ucode: -110
[    9.893379] iwlwifi 0000:00:0b.0: retry init count 0
```

fuzhongl, 2022-08-10:
@hidrol Please help to fix this issue for native first. Seems like WIFI firmware related.