projectacrn/acrn-hypervisor #7759: ERROR: LeakSanitizer: detected memory leaks found when run

liudlong, 2022-06-14:
==1619==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 4392 byte(s) in 9 object(s) allocated from:
    #0 0x55bafdf84eb6 in calloc (/usr/bin/acrn-dm+0x328eb6)
    #1 0x55bafe1d4bc7 in virtio_vhost_vsock_init hw/pci/virtio/vhost_vsock.c:266
    #2 0x55bafe0e25f1 in pci_emul_init hw/pci/core.c:1042
    #3 0x55bafe0f6c1f in init_pci hw/pci/core.c:1523
    #4 0x55bafe21805e in vm_init_vdevs core/main.c:507
    #5 0x55bafe21ba56 in main core/main.c:1085
    #6 0x7ff3f40c40b2 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x240b2)

Direct leak of 1064 byte(s) in 7 object(s) allocated from:
    #0 0x55bafdf84eb6 in calloc (/usr/bin/acrn-dm+0x328eb6)
    #1 0x7ff3f4fa60e9  (/lib/x86_64-linux-gnu/libusb-1.0.so.0+0x50e9)

Direct leak of 792 byte(s) in 9 object(s) allocated from:
    #0 0x55bafdf850ee in realloc (/usr/bin/acrn-dm+0x3290ee)
    #1 0x7ff3f4e0d4e5  (/lib/x86_64-linux-gnu/libSDL2-2.0.so.0+0x744e5)

Direct leak of 392 byte(s) in 7 object(s) allocated from:
    #0 0x55bafdf84eb6 in calloc (/usr/bin/acrn-dm+0x328eb6)
    #1 0x7ff3f4faf3a1  (/lib/x86_64-linux-gnu/libusb-1.0.so.0+0xe3a1)

Indirect leak of 14336 byte(s) in 14 object(s) allocated from:
    #0 0x55bafdf850ee in realloc (/usr/bin/acrn-dm+0x3290ee)
    #1 0x7ff3f4fae979  (/lib/x86_64-linux-gnu/libusb-1.0.so.0+0xd979)

Indirect leak of 6984 byte(s) in 9 object(s) allocated from:
    #0 0x55bafdf84cb8 in malloc (/usr/bin/acrn-dm+0x328cb8)
    #1 0x7ff3f4e0d447  (/lib/x86_64-linux-gnu/libSDL2-2.0.so.0+0x74447)

Indirect leak of 1064 byte(s) in 7 object(s) allocated from:
    #0 0x55bafdf84eb6 in calloc (/usr/bin/acrn-dm+0x328eb6)
    #1 0x7ff3f4fa60e9  (/lib/x86_64-linux-gnu/libusb-1.0.so.0+0x50e9)

Indirect leak of 70 byte(s) in 14 object(s) allocated from:
    #0 0x55bafdf0d8cd in strdup (/usr/bin/acrn-dm+0x2b18cd)
    #1 0x7ff3f4fae8b3  (/lib/x86_64-linux-gnu/libusb-1.0.so.0+0xd8b3)

SUMMARY: AddressSanitizer: 29094 byte(s) leaked in 76 allocation(s).

liudlong, 2022-06-14:
[External_System_ID] ACRN-9163

fuzhongl, 2022-07-05:
Fixed in v3.0.

closed 2022-07-05 by fuzhongl