projectacrn/acrn-hypervisor #7778: Grub configuration not modified correctly when using a custom $(SCENARIO) name

gvancuts, 2022-06-16:
The Debian package generated when compiling ACRN also includes a `arcn-hypervisor.postinst` (e.g. [this file](https://github.com/projectacrn/acrn-hypervisor/blob/master/misc/packaging/acrn-hypervisor.postinst)) that gets tweaked to use the `$(BOARD)` and `$(SCENARIO)` names used at build time.

The problem is that as soon as you use a `$(SCENARIO)` name that is **not** one of `[shared | hybrid | hybrid_rt | partitioned]`, then the Grub configuration (that is done via `/etc/grub.d/40_custom`) is not done at all, the post-installation script just does not do anything with that file. It does however change the default Grub boot option to `ACRN_deb_multiboot2` but which is now a non-existing or wrong entry. If you had a previous installation of ACRN on your system, it will complain that the ACRN hypervisor (aka `acrn.${SCENARIO}.${BOARD}.bin`) does not exist.

This issue will be met when people rename or use custom names for their scenario files. The most straightforward way to reproduce this is also very simply to import your board XML and create a new scenario from scratch. That scenario will be called `scenario.xml` and that name not being on of the four names recognized by the script, the Grub config will not be generated correctly leading to a system that will not boot.

gvancuts, 2022-06-16:
/cc @tttech-industrial-buchsbaum

gvancuts, 2022-06-16:
[External_System_ID] ACRN-9079

junjiemao1, 2022-06-17:
The scripts under `misc/packaging` are the legacy of previous releases which I assume should be replaced with the solution from TTTech.

Are we still using those legacy ones in GSG?

junjiemao1, 2022-06-17:
Considering the timeline for release 3.0, I was wondering if we have time to switch to the dpkg-buildpackage based approach for the GSG. Today both implementations have issues:

1. The scripts under misc/packaging have hardcoded scenario names and tags for OS kernel images. They only work for the scenarios defined under misc/config_tools/data, but probably won't for any user-customized ones.
2. The dpkg-buildpackage scripts under debian/ have handled the scenario names and kernel image tags properly. Unfortunately, adding pre-launched VM kernel images and virtual ACPI tables are yet to be done.

The PR I just submitted is a workaround to make the current GSG happy. Meanwhile, I'll take a look at that second item. If that workaround looks too ugly from your point of view, we need to evaluate the impact of switching to dpkg-buildpackage.

gvancuts, 2022-06-17:
> The PR I just submitted is a workaround to make the current GSG happy. Meanwhile, I'll take a look at that second item. If that workaround looks too ugly from your point of view, we need to evaluate the impact of switching to dpkg-buildpackage.

It is ugly but still think this is the best we can do given the time we have.

NanlinXie, 2022-09-07:
@gvancuts , could we close this issue?

gvancuts, 2022-09-07:
> @gvancuts , could we close this issue?

I don't believe we have done anything better than the workaround we put in place for the 3.0 release, or have we? If not, then we should not close this yet.