projectacrn/acrn-hypervisor #7780: The SCENARIO name is changed to the folder name containing it

gvancuts, 2022-06-16:
Let's consider the following set-up (folder and its content have been produced by the ACRN Configurator):
```
$ ls -1 ~/work/My-ACRN-configs/
launch_user_vm_id1.sh
launch_user_vm_id2.sh
My-Vecow.board.xml
scenario.xml
```
Let's now build ACRN using these files:
```
make BOARD=~/work/My-ACRN-configs/My-Vecow.board.xml SCENARIO=~/work/My-ACRN-configs/scenario.xml
```

This produces the following result:
```
ACRN Configuration Summary:
Build directory: /home/gvancuts/work/acrn-hypervisor/build/hypervisor
This build directory is configured with the settings below.
- BOARD = My-Vecow
- SCENARIO = My-ACRN-configs
- RELEASE = y
dpkg-deb: building package 'acrn-hypervisor' in 'acrn-My-Vecow-My-ACRN-configs-3.0-acrn-2022w25.3-180000p-7-g4a7d6e411-dirty.deb'.
ACRN debian package acrn-My-Vecow-My-ACRN-configs-3.0-acrn-2022w25.3-180000p-7-g4a7d6e411-dirty.deb was successfully created in the /home/gvancuts/work/acrn-hypervisor/build.
```

We can see in the output above that the name of the scenario (i.e. `scenario`) has somehow been changed to use the name of the folder containing all config files, here `My-ACRN-configs`.

dbkinder, 2022-06-16:
I think what's happening here is the scenario is unnamed.  It's called "scenario.xml" with no prefix, while the board file is called "My-Vecow-board.xml" with the board name prefix.

This might just be a documentation issue that we don't describe the naming rules for files.  That debian package is named acrn-<board name>-<working folder name>-<git tag name>.deb

gvancuts, 2022-06-16:
This is not a documentation issue, the scenario is unnamed (so to speak) because the ACRN Configurator creates it that way. And there is no way to change that value from within the Configurator. If documenting this means instructing the user to go fetch the file to rename it first before it can be used to build ACRN, this can only really be a poor workaround because we can fix this in any better way before the release.

cjbooth, 2022-06-17:
For v3.0 we’re changing the configuration file structure to help organize files around configurations (in the Configurator it’s called the working folder and see the image below for a visual representation).  This will hopefully help users keep track of different configurations they try and the different files associated with those configurations.  The design intention is that users will remember the name of their working folder as that’s their way of naming the configuration.  With that, having the folder name as a part of the hypervisor binary makes sense as it’s what users will, hopefully, remember.  David, Amy and I talked about documenting some of this design rationale and how it connects to the folder and file structure to make it more clear to users.

This is the initial version of a substantial change to ACRN configuration…certainly not perfect but a good first step. I think we need to see if this new model is understandable or confusing to our users, learn from that, and keep iterating.

amyreye, 2022-06-28:
I added a sentence to the GSG: "The Debian file name contains the board name (my_board) and the working folder name (MyConfiguration)."

The GSG doesn't show the build output, so I didn't explain the `SCENARIO = My-ACRN-configs`. Suggest we wait to see if we get any other feedback.