projectacrn/acrn-hypervisor #7790: User VM fail to get IP with vhost net

liudlong, 2022-06-20:
How Found: Function Test

Environment: –

HW/Board: ADL-S

Build link: http://sit-image.sh.intel.com/ubuntu/daily-build/2022ww25.3-2/

Image info: 3.0-acrn-2022w25.3-180000p-2022-06-15

Steps:
1. Flash RC6 and boot ACRN shared
2. Generate launch script for rt with vhost virtio network:

(venv) root@robot-WL10:[virtualization.hypervisors.acrn.acrn-dev.acrn-offline-patch]# git diff
diff --git a/boards/adl-s-crb/shared_launch_6user_vm.xml b/boards/adl-s-crb/shared_launch_6user_vm.xml
index ad0bf230..883cd2d8 100644
--- a/boards/adl-s-crb/shared_launch_6user_vm.xml
+++ b/boards/adl-s-crb/shared_launch_6user_vm.xml
@@ -328,7 +328,7 @@
         <backend_type>stdio</backend_type>
       </console>
       <network>
-        <virtio_framework/>
+             <virtio_framework>Kernel based (Virtual Host)</virtio_framework>
         <interface_name>RT</interface_name>
       </network>
       <input/>

3. enable vhost in SOS:

modprobe vhost-net experimental_zcopytx=1
4. launch rt with the new launch script:

acrn-dm --cpu_affinity 64,66 -m 1024M --ovmf /usr/share/acrn/bios/OVMF.fd -s 0:0,hostbridge --virtio_poll 1000000 -s 3,virtio-console,@stdio:stdio_por
t -s 4,virtio-net,tap=RT,vhost,mac_seed=90:12:10:b5:7d:f7-POST_RT_VM1 -s 5,virtio-blk,/root/preempt_rt/core-image-weston-intel-corei7-64.wic --rtvm --la
pic_pt --logger_setting 'console,level=5;kmsg,level=5;disk,level=5' POST_RT_VM1
Expected result
Rt get IP with vhost net

Actual result:
RT fail to get IP

Reproduce rate: 100%

liudlong, 2022-06-20:
[External_System_ID] ACRN-9208

fuzhongl, 2022-07-05:
Fixed in v3.0.

closed 2022-07-05 by fuzhongl