projectacrn/acrn-hypervisor #7836: Cannot build ACRN for Vecow

gvancuts, 2022-06-28:
It is not possible to build ACRN for the Vecow platform on today's code, i.e.: `acrn-2022w22.2-180000p-67-gcd60df638` (master on June 28th 2022)

Here is the error message:
```
$ make BOARD=tgl-vecow-spc-7100-Corei7 SCENARIO=partitioned
make[1]: Entering directory '/home/gvancuts/work/acrn-hypervisor/hypervisor'
Board XML is fetched from /home/gvancuts/work/acrn-hypervisor/misc/config_tools/data/tgl-vecow-spc-7100-Corei7/tgl-vecow-spc-7100-Corei7.xml
Scenario XML is being fetched from /home/gvancuts/work/acrn-hypervisor/misc/config_tools/data/tgl-vecow-spc-7100-Corei7/partitioned.xml
Traceback (most recent call last):
  File "/home/gvancuts/work/acrn-hypervisor/misc/config_tools/static_allocators/main.py", line 41, in <module>
    main(args)
  File "/home/gvancuts/work/acrn-hypervisor/misc/config_tools/static_allocators/main.py", line 19, in main
    common.get_vm_num(args.scenario)
AttributeError: module 'common' has no attribute 'get_vm_num'
make[1]: *** No rule to make target '/home/gvancuts/work/acrn-hypervisor/build/hypervisor/configs/allocation.xml', needed by '/home/gvancuts/work/acrn-hypervisor/build/hypervisor/configs/unified.xml'.  Stop.
make[1]: Leaving directory '/home/gvancuts/work/acrn-hypervisor/hypervisor'
make: *** [Makefile:123: hvdefconfig] Error 2

```

fuzhongl, 2022-06-30:
@gvancuts There is NOT this issue for v3.0:
```
ACRN Configuration Summary:
Build directory: /home/acrn/acrn-hypervisor/build/hypervisor
This build directory is configured with the settings below.
- BOARD = tgl-vecow-spc-7100-Corei7
- SCENARIO = partitioned
- RELEASE = n
dpkg-deb: building package 'acrn-hypervisor' in 'acrn-tgl-vecow-spc-7100-Corei7-partitioned-3.0-v3.0-1-ge1e2ed898.deb'.
ACRN debian package acrn-tgl-vecow-spc-7100-Corei7-partitioned-3.0-v3.0-1-ge1e2ed898.deb was successfully created in the /home/acrn/acrn-hypervisor/build.
```
Could you please help to double check on your side with v3.0?
Thanks!

gvancuts, 2022-06-30:
I'm still facing the same issue:
```
$ make BOARD=tgl-vecow-spc-7100-Corei7 SCENARIO=partitioned
make[1]: Entering directory '/home/gvancuts/work/acrn-hypervisor/hypervisor'
Board XML is fetched from /home/gvancuts/work/acrn-hypervisor/misc/config_tools/data/tgl-vecow-spc-7100-Corei7/tgl-vecow-spc-7100-Corei7.xml
Scenario XML is being fetched from /home/gvancuts/work/acrn-hypervisor/misc/config_tools/data/tgl-vecow-spc-7100-Corei7/partitioned.xml
Traceback (most recent call last):
  File "/home/gvancuts/work/acrn-hypervisor/misc/config_tools/static_allocators/main.py", line 41, in <module>
    main(args)
  File "/home/gvancuts/work/acrn-hypervisor/misc/config_tools/static_allocators/main.py", line 19, in main
    common.get_vm_num(args.scenario)
AttributeError: module 'common' has no attribute 'get_vm_num'
make[1]: *** No rule to make target '/home/gvancuts/work/acrn-hypervisor/build/hypervisor/configs/allocation.xml', needed by '/home/gvancuts/work/acrn-hypervisor/build/hypervisor/configs/unified.xml'.  Stop.
make[1]: Leaving directory '/home/gvancuts/work/acrn-hypervisor/hypervisor'
make: *** [Makefile:137: hvdefconfig] Error 2
```
```
$ git describe
v3.0
```

gvancuts, 2022-06-30:
I figured out what was wrong in my environment. I had a `common` module available in my user environment that was coming from `sphinx-notfound-page`. This is the `common` module that was being loaded instead of the one we provide in `../library/` (https://github.com/projectacrn/acrn-hypervisor/tree/master/misc/config_tools/library).

I have removed that module and in fact, all other Python modules that were installed in my user virtual env. I can now build ACRN successfully.

I'm no Python expert but is there a way we can be more prescriptive about where to load `common` from?

gvancuts, 2022-07-05:
> I'm no Python expert but is there a way we can be more prescriptive about where to load `common` from?

@junjiemao1 , do you know who could look into this?