projectacrn/acrn-hypervisor #7867: [debian] apt build-dep acrn-hypervisor failed due to control file typo

Pirouf, 2022-07-03:
**Describe the bug**
fail to debian build-deps format 
```
Reading package lists...
mk-build-deps: warning: can't parse dependency python3-elementpath (>= 2.5.0)   python3-requests
Use of uninitialized value $positive in concatenation (.) or string at /usr/bin/mk-build-deps line 576.
```
due to missing `python3-elementpath (>= 2.5.0) ,` comma-separator : 
```
   python3-elementpath (>= 2.5.0)
   python3-requests,
   python3-tqdm
```
**Platform**
All 

**Codebase**
Release 3.0-tag

**Scenario**
Any 

**To Reproduce**
Steps to reproduce the behavior:
2. 'sudo apt-get source acrn-hypervisor'
2. 'sudo apt build-dep acrn-hypervisor'
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Attach the following logs if available:

- native Linux boot kernel message log
- ACRN serial port log (when issues happens)
- Service VM kernel message when issue happens
- acrn-dm log: all files in /var/log/acrn-tm
- console log when running script to launch guest VM

Pirouf, 2022-07-03:
Fix available under https://github.com/projectacrn/acrn-hypervisor/pull/7807 but incorrectly merged onto [v3.0-tag](https://github.com/projectacrn/acrn-hypervisor/commits/v3.0)

dbkinder, 2022-07-03:
@NanlinXie @junjiemao1 please verify the correct elementpath version to use. This is different than what's in the GSG.

junjiemao1, 2022-07-04:
Let's use elementpath 2.5.0 as our overlay is not compatible with the recent 2.5.3 version for now.

Pirouf, 2022-07-04:
Not a version imcompatility. i think you got it wrong. 
the typo issue debian control file is missing `,` comma  at the end 
   python3-elementpath (>= 2.5.0) **`,`**
   python3-requests,
   python3-tqdm

junjiemao1, 2022-07-04:
> Not a version imcompatility. i think you got it wrong. the debian control file typo issue is a missing `,` comma on `Build-Depends:` python3-elementpath (>= 2.5.0) **`,`** python3-requests, python3-tqdm

Yes, I have noticed that.

I'm responding to the comment from @dbkinder who also raised another question about the exact version we should use, considering the incompatibility issue (when using elementpath 2.5.3) we are aware of. That is a different issue and could be tracked separately, of course.

closed 2022-09-28 by fuzhongl