projectacrn/acrn-hypervisor #7883: ACRN-3.0  not able to parse all the PCI devs for refence boards.

pchand20, 2022-07-08:
**Describe the bug**
latest acrn code is not able to identify all the PCI devices , this issue we saw on TGL-U RVP board. 

**Platform**
TGL-U RVP board.

**Codebase**
Observations for ACRN 3.0:

**Scenario**
hybrid 
**To Reproduce**
try to add PCI based vUART devices ( 5 were supported in previous releasae) and than generate xml from configurator. this will show that  PCI devs are not get identify correctly and SOS boot will have issues. 

**Expected behavior**
adding PCI based vUART shall be working fine for vm to vm connections. 

**Additional context**
Observations for ACRN 3.0:
Each board required to have its own board.xml generated.
Experiments performed:
We have two similar configuration tgl board, I generated the board.xml file on one of the board.
The image is booting successfully with both pre-launch and sos-vm only on the board where I generated the board.xml, but need the below workaround to boot successfully on another.
*******************************************
+                 if (pbdf.bits.b > 0x2) {
+                       continue;
+                }
*********************************************   

- native Linux boot kernel message log
- ACRN serial port log (when issues happens)
- Service VM kernel message when issue happens
- acrn-dm log: all files in /var/log/acrn-tm
- console log when running script to launch guest VM

fuzhongl, 2022-07-11:
@pchand20 Rajput, Anamika help to confirmed this issue comes from board.xml NOT match with the board:

```
[2022/7/8 19:02] Rajput, Anamika

i have two boards, so i tried on other board..it booted without patch
```
Make sure to re-generate the board.xml if you changed the board or BIOS setting.

Thanks!

pchand20, 2022-07-11:
It does not make sense that for same version of RVP board I have to generate board config for every board. what we have two same RVP boards with similar bios and firmware and same SKU.  please make sure that you test it properly before copy pasting a portion of chat which is not relevant without full message.

NanlinXie, 2022-07-11:
ACRN has the configuration item "CONFIG_IOMMU_BUS_NUM", which is auto calculated from board config information via #lspci command. After check on Anamika's board, we found in the board.xml, there's no "ab:00.0" in <PCI_DEVICE> node, but when run #lspci command on her board, we found "ab:00:0" device. The wrong board.xml will cause wrong value of "CONFIG_IOMMU_BUS_NUM" be calculated, which finally caused panic when ACRN boot. That's why ACRN's development model is requiring user to generate board.xml whenever they change BIOS setting or PCI ports, please see the Important notes from our overview document: https://projectacrn.github.io/3.0/getting-started/overview_dev.html#use-the-board-inspector-to-generate-a-board-configuration-file.

pchand20, 2022-07-11:
seems this is regression from previous version of the ACRN to this version earlier same board and scenario do work fine on all RVP boards of same  generation, if we have to generate board file for every board it wont be easy to deploy this in field. please check this regression why this is happening in latest code rather than using developers setup directly to debug. could you list boards revision you have default support here with PCI uart you tested on ?

NanlinXie, 2022-07-11:
In previous release, we depend on user to proper config "CONFIG_IOMMU_BUS_NUM" value in scenario.xml file, but it is hard for user to calcuate, thus in v3.0, we automatically generate this value from board.xml file in the assumption that board.xml is accurate.  The possible solution maybe you could use the board.xml generated on more PCI device board as your unique board.xml if you don't want to maintain multiple board.xml, this should also works for the board with less PCI device since this configuration is the max value of the PCI device.  ACRN doesn't dynamically detect this value from board at runtime due to static configuration characteristic and LOC control, thus we have to provide a static way to feed the max value to hypervisor, either via user input in scenario.xml as previous release, or via auto calculated from board.xml.

closed 2022-07-26 by fuzhongl