projectacrn/acrn-hypervisor #7913: config-tools: build acrn successfully even set the same memory address between the pre vm 0 and hv

zihengL1, 2022-07-21:
How Found: Function Test

Environment:
#HW/Board: ADL-S
#Build link: http://sit-image.sh.intel.com/ubuntu/daily-build/2022ww20.3-1/

Steps:
1. Apply the patch below and build acrn hypervisor:

diff --git a/boards/adl-s-crb/hybrid.xml b/boards/adl-s-crb/hybrid.xml
index 53321e00..cdba995d 100644
--- a/boards/adl-s-crb/hybrid.xml
+++ b/boards/adl-s-crb/hybrid.xml
@@ -85,7 +85,7 @@
     </epc_section>
     <memory>
       <hpa_region>
-        <start_hpa>0x100000000</start_hpa>
+        <start_hpa>0x00400000</start_hpa>
         <size_hpa>512</size_hpa>
       </hpa_region>
     </memory>
Expected result:
build acrn failed due to the same memory address configured in xml.
 
Actual result:
Able to build acrn

Debug info:
–

zihengL1, 2022-07-21:
[External_System_ID] ACRN-9039

chuangxke, 2022-07-22:
[External_System_ID] ACRN-9077

closed 2022-09-07 by NanlinXie