projectacrn/acrn-hypervisor #7975: GSG python package xmlschema/elementpath incompatibility

zihengL1, 2022-08-11:
On a clean Ubuntu 20.04.2 installation, I followed the GSG, and while installing the ACRN pre-requisites I got this error:

david@X220-20:~$ sudo pip3 install "elementpath==2.5.0" lxml xmlschema defusedxml tqdm
Collecting elementpath==2.5.0
  Downloading elementpath-2.5.0-py3-none-any.whl (181 kB)
     |████████████████████████████████| 181 kB 1.2 MB/s 
Collecting lxml
  Downloading lxml-4.9.1-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.manylinux_2_24_x86_64.whl (6.9 MB)
     |████████████████████████████████| 6.9 MB 9.7 MB/s 
Collecting xmlschema
  Downloading xmlschema-2.0.1-py3-none-any.whl (354 kB)
     |████████████████████████████████| 354 kB 28.2 MB/s 
Collecting defusedxml
  Downloading defusedxml-0.7.1-py2.py3-none-any.whl (25 kB)
Collecting tqdm
  Downloading tqdm-4.64.0-py2.py3-none-any.whl (78 kB)
     |████████████████████████████████| 78 kB 6.5 MB/s 
ERROR: xmlschema 2.0.1 has requirement elementpath<4.0.0,>=3.0.0, but you'll have elementpath 2.5.0 which is incompatible.
Installing collected packages: elementpath, lxml, xmlschema, defusedxml, tqdm
Successfully installed defusedxml-0.7.1 elementpath-2.5.0 lxml-4.9.1 tqdm-4.64.0 xmlschema-2.0.1

 

On my other Ubuntu 20.04 system the xmlschema version is 1.9.2, leading me to think we have another version requirement missing in the GSG.  Do we need to require xmlschema==1.9.2?

zihengL1, 2022-08-11:
[External_System_ID] ACRN-9283

dbkinder, 2022-08-11:
In general we've not specified specific compatible versions of apt or python packages. The "latest" GSG was recently updated to say:

sudo pip3 install "elementpath==2.5.0" lxml "xmlschema==1.10.0" defusedxml tqdm

while a more complete solution is underway.

closed 2022-09-07 by NanlinXie