projectacrn/acrn-hypervisor #7980: How to use irtio-gpu with SR-IOV Virtual Function

davidlamblin, 2022-08-12:
How to use irtio-gpu with SR-IOV Virtual Function?
Please offer some guide.
Thanks

fuzhongl, 2022-08-12:
@davidlamblin Could you please share the hardware that has GPU SRIOV capability?
And make sure GPU VF can be created successfully in native.
If yes, you can use virtio-gpu + GPU  SRIOV in acrn-dm as following:

```
 `add_virtual_device                       5 virtio-gpu geometry=1280x720+100+100`
 `add_passthrough_device             2 0000:00:02.1`
```
00:02.1 is the  BDF of GPU VF; which is passthrough to User VM.

davidlamblin, 2022-08-16:
I have followed  the above operation, and in the user VM, there are two VGA controller.
one is using kernel driver i915, one is using pci_stub. as shown below.

00:02.0 VGA compatible controller: Intel Corporation Device 9a49 (rev 01) (prog-if 00 [VGA controller])
	Subsystem: Intel Corporation Device 3002
	Flags: fast devsel
	Memory at 80000000 (64-bit, non-prefetchable) [size=16M]
	Memory at a0000000 (64-bit, prefetchable) [size=512M]
	Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
	Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
	Capabilities: [ac] MSI: Enable- Count=1/1 Maskable+ 64bit-
	Kernel modules: i915

00:09.0 VGA compatible controller: Red Hat, Inc. Virtio GPU (rev 01) (prog-if 00 [VGA controller])
	Subsystem: Red Hat, Inc. Virtio GPU
	Flags: bus master, fast devsel, latency 64, IRQ 20
	Memory at c1000000 (32-bit, non-prefetchable) [size=16M]
	Memory at 4000004000 (64-bit, prefetchable) [size=16K]
	Memory at c2000000 (32-bit, non-prefetchable) [size=8K]
	I/O ports at 5000 [size=4]
	Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
	Capabilities: [40] Vendor Specific Information: VirtIO: CommonCfg
	Capabilities: [50] Vendor Specific Information: VirtIO: ISR
	Capabilities: [60] Vendor Specific Information: VirtIO: DeviceCfg
	Capabilities: [70] Vendor Specific Information: VirtIO: Notify
	Capabilities: [84] Vendor Specific Information: VirtIO: <unknown>
	Capabilities: [98] MSI-X: Enable+ Count=3 Masked-
	Capabilities: [a4] MSI: Enable- Count=1/1 Maskable+ 64bit+
	Capabilities: [bc] Vendor Specific Information: VirtIO: Notify
	Kernel driver in use: virtio-pci

But when move window in user VM, the window drawing picture slowly. so it wonder if the VF+ virtio-gpu mechanism work.
How to verify it.

By the way,  the Service VM will show the following error sometime:
[   97.304416] i915 0000:00:02.0: [drm] *ERROR* Atomic update failure on pipe A (start=6624 end=6625) time 4450 us, min 1431, max 1439, scanline start 1118, end 32
[  102.108404] i915 0000:00:02.0: [drm] *ERROR* Atomic update failure on pipe A (start=6912 end=6913) time 4544 us, min 1431, max 1439, scanline start 1270, end 193
[  119.582405] i915 0000:00:02.0: [drm] *ERROR* Atomic update failure on pipe A (start=7960 end=7961) time 4704 us, min 1431, max 1439, scanline start 1198, end 135

Hardware: NUC11TNKi5000, model name	: 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz
i915 firmware: tgl_dmc_ver2_12.bin  tgl_guc_70.1.1.bin

 i915.enable_guc=5

fuzhongl, 2022-08-17:
@yakuizyouhao Could you please help to check this GPU SRIOV + virtio-gpu related question?
Thanks!

davidlamblin, 2022-08-22:
@fuzhongl   @yakuizhao   if any suggestion?

fuzhongl, 2022-08-22:
@davidlamblin Could you please share the hardware info which you run ACRN and Service/User VM kernel info?
Thanks!

davidlamblin, 2022-08-23:
Hardware: NUC11TNKi5000, model name : 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz
i915 firmware: tgl_dmc_ver2_12.bin tgl_guc_70.1.1.bin
i915.enable_guc=5

service vm 的内核启动参数是：
root=/dev/nvme0n1p2 idle=halt rw rootwait console=tty0 console=ttyS0 consoleblank=0 no_timer_check quiet loglevel=3 i915.nuclear_pageflip=1 swiotlb=131072 i915.modeset=1 i915.max_vfs=4 i915.enable_guc=5 maxcpus=8 hugepagesz=1G hugepages=8 udmabuf.list_limit=8192  root=PARTUUID=b0dc4d9d-8cc4-4747-93c7-e7dde371a661 

 User VM 是  ubuntu 20.04， 内核启动参数为:
$ cat /proc/cmdline 
BOOT_IMAGE=/boot/vmlinuz-5.15.0-46-generic root=UUID=25548a85-339c-4b21-9b34-ba75c424f1f0 ro quiet splash vt.handoff=7

fuzhongl, 2022-08-23:
@davidlamblin GPU SRIOV isn't verified on TGL platform.
Could you please help to try ACRN master branch kernel for Service VM and User VM, which is 5.15.44 version.
https://github.com/projectacrn/acrn-kernel/tree/5.15/linux

Thanks!

davidlamblin, 2022-08-23:
@fuzhongl 
1. using https://github.com/projectacrn/acrn-kernel/tree/5.15/linux as service vm kernel?     user vm use the original version of ubuntu kernel?
2. what is the proper way for gpu virtualization in TGL?

fuzhongl, 2022-08-23:
> @fuzhongl
> 
> 1. using https://github.com/projectacrn/acrn-kernel/tree/5.15/linux as service vm kernel?     user vm use the original version of ubuntu kernel?
User VM also use https://github.com/projectacrn/acrn-kernel/tree/5.15/linux; but this kernel config:
https://github.com/projectacrn/acrn-kernel/blob/5.15/linux/kernel_config_user_vm
> 2. what is the proper way for gpu virtualization in TGL?
GVT-d is verified on TGL.

davidlamblin, 2022-08-23:
How to enable SRIOV funtion is 5.15  with i915? I found there are something  different with 5.10.

In 5.10,  pass i915.enable_guc=5 in bootargs, but in 5.15, i915.enable_guc=5 is not supported.
In 5.10,  i915_pci_driver  has sriov_configure callback, but in 5.15 has not.

davidlamblin, 2022-08-24:
@fuzhongl

fuzhongl, 2022-08-29:
> How to enable SRIOV funtion is 5.15 with i915? I found there are something different with 5.10.
> 
> In 5.10, pass i915.enable_guc=5 in bootargs, but in 5.15, i915.enable_guc=5 is not supported. In 5.10, i915_pci_driver has sriov_configure callback, but in 5.15 has not.

@davidlamblin Please add following kernel cmdline, it works well with kernel 5.15
`i915.force_probe=* i915.enable_guc=0x7`

Thanks!

davidlamblin, 2022-08-30:
1. I guess the branch should be   **5.15/adl-linux** , not  **5.15/linux**
2. I will check cmdline: i915.force_probe=* i915.enable_guc=0x7
3. in Uservm there are two VGA controller, How the virtio-gpu frontend driver use the VF related VGA controller?

fuzhongl, 2022-08-30:
> 1. I guess the branch should be   **5.15/adl-linux** , not  **5.15/linux**
Please use `remotes/origin/release_3.1` which is verified 

> 2. I will check cmdline: i915.force_probe=* i915.enable_guc=0x7
Thanks!
> 3. in Uservm there are two VGA controller, How the virtio-gpu frontend driver use the VF related VGA controller?
Could you please share the launch script of User VM?

davidlamblin, 2022-08-30:
# Invoking ACRN device model
dm_params=(
    `add_cpus                                 0 1 2 3 4 5 6 7`
    -m 4096M
    -k  /boot/vmlinuz-5.15.44-acrn-user-vm-272197-g692361d31657-dirty
    --ovmf /usr/share/acrn/bios/OVMF.fd
    `add_virtual_device                       1:0 lpc`
    `add_virtual_device                       0:0 hostbridge`
    `add_virtual_device                       3 xhci 3-3`
    `add_virtual_device                       6 virtio-input Device name: USB OPTICAL MOUSE, Device physical path: usb-0000:00:14.0-7/input0`
    `add_virtual_device                       7 virtio-console @stdio:stdio_port`
    `add_virtual_device                       10 virtio-blk /home/ubuntu_images/ubuntu_uservm3.img`
    `add_virtual_device                       9 virtio-gpu geometry=5360x1440+0+0`
    `add_passthrough_device                   2 0000:00:02.1`
    `add_logger_settings                      console=4 kmsg=3 disk=5`
    VM1
)

fuzhongl, 2022-08-30:
> 

Please share` lspci` and `uname -a` in User VM.
Thanks!

davidlamblin, 2022-08-30:
uservm@uservm-Standard-PC-i440FX-PIIX-1996:~$ uname -r
5.15.44-acrn-user-vm-272197-g692361d31657-dirty

root@uservm-Standard-PC-i440FX-PIIX-1996:~# lspci -v
00:00.0 Host bridge: Network Appliance Corporation Device 1275
	Flags: bus master, fast devsel, latency 0
	Capabilities: [40] Express Root Port (Slot-), MSI 00

00:01.0 ISA bridge: Intel Corporation 82371SB PIIX3 ISA [Natoma/Triton II]
	Flags: bus master, fast devsel, latency 0

00:02.0 VGA compatible controller: Intel Corporation Device 9a49 (rev 01) (prog-if 00 [VGA controller])
	Subsystem: Intel Corporation Device 3002
	Flags: fast devsel
	Memory at 80000000 (64-bit, non-prefetchable) [size=16M]
	Memory at a0000000 (64-bit, prefetchable) [size=512M]
	Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
	Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
	Capabilities: [ac] MSI: Enable- Count=1/1 Maskable+ 64bit-

00:03.0 USB controller: Device 1234:1234 (prog-if 30 [XHCI])
	Flags: bus master, fast devsel, latency 64, IRQ 90
	Memory at c0000000 (32-bit, non-prefetchable) [size=4K]
	Capabilities: [40] MSI: Enable+ Count=1/1 Maskable+ 64bit+
	Kernel driver in use: xhci_hcd

00:06.0 Input device controller: Red Hat, Inc. Virtio input (rev 01)
	Subsystem: Red Hat, Inc. Virtio input
	Flags: bus master, fast devsel, latency 64, IRQ 17
	Memory at c0002000 (32-bit, non-prefetchable) [size=8K]
	I/O ports at 2000 [size=4]
	Memory at 4000000000 (64-bit, prefetchable) [size=16K]
	Capabilities: [40] MSI-X: Enable+ Count=3 Masked-
	Capabilities: [4c] MSI: Enable- Count=1/1 Maskable+ 64bit+
	Capabilities: [64] Vendor Specific Information: VirtIO: Notify
	Capabilities: [78] Vendor Specific Information: VirtIO: CommonCfg
	Capabilities: [88] Vendor Specific Information: VirtIO: ISR
	Capabilities: [98] Vendor Specific Information: VirtIO: DeviceCfg
	Capabilities: [a8] Vendor Specific Information: VirtIO: Notify
	Capabilities: [bc] Vendor Specific Information: VirtIO: <unknown>
	Kernel driver in use: virtio-pci

00:07.0 Communication controller: Red Hat, Inc. Virtio console
	Subsystem: Red Hat, Inc. Virtio console
	Flags: bus master, fast devsel, latency 64, IRQ 18
	I/O ports at 3000 [size=64]
	Memory at c0004000 (32-bit, non-prefetchable) [size=8K]
	Capabilities: [40] MSI-X: Enable+ Count=35 Masked-
	Capabilities: [4c] MSI: Enable- Count=1/1 Maskable+ 64bit+
	Kernel driver in use: virtio-pci

00:09.0 VGA compatible controller: Red Hat, Inc. Virtio GPU (rev 01) (prog-if 00 [VGA controller])
	Subsystem: Red Hat, Inc. Virtio GPU
	Flags: bus master, fast devsel, latency 64, IRQ 19
	Memory at c1000000 (32-bit, non-prefetchable) [size=16M]
	Memory at 4000004000 (64-bit, prefetchable) [size=16K]
	Memory at c2000000 (32-bit, non-prefetchable) [size=8K]
	I/O ports at 4000 [size=4]
	Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
	Capabilities: [40] Vendor Specific Information: VirtIO: CommonCfg
	Capabilities: [50] Vendor Specific Information: VirtIO: ISR
	Capabilities: [60] Vendor Specific Information: VirtIO: DeviceCfg
	Capabilities: [70] Vendor Specific Information: VirtIO: Notify
	Capabilities: [84] Vendor Specific Information: VirtIO: <unknown>
	Capabilities: [98] MSI-X: Enable+ Count=3 Masked-
	Capabilities: [a4] MSI: Enable- Count=1/1 Maskable+ 64bit+
	Capabilities: [bc] Vendor Specific Information: VirtIO: Notify
	Kernel driver in use: virtio-pci

00:0a.0 SCSI storage controller: Red Hat, Inc. Virtio block device
	Subsystem: Red Hat, Inc. Virtio block device
	Flags: bus master, fast devsel, latency 64, IRQ 20
	I/O ports at 5000 [size=128]
	Memory at c2002000 (32-bit, non-prefetchable) [size=8K]
	Capabilities: [40] MSI-X: Enable+ Count=2 Masked-
	Capabilities: [4c] MSI: Enable- Count=1/1 Maskable+ 64bit+
	Kernel driver in use: virtio-pci

fuzhongl, 2022-08-30:
> 3. in Uservm there are two VGA controller, How the virtio-gpu frontend driver use the VF related VGA controller?
virtio-gpu frontend driver is only used for display, the VF related VGA controller will be used for 2D/3D performance  GPU HW accelerate.

@yakuizhao Please help to share more info about GPU VF.
Thanks!

davidlamblin, 2022-08-30:
so how can I verify that  VF is working?
I use Unigine_Valley-1.0  and hardinfo to test the performance, the result is shown below.

<b style="font-weight:normal;" id="docs-internal-guid-c317cb27-7fff-2e00-3473-1c47723ba172"><div dir="ltr" style="margin-left:9pt;" align="left">

Test Point | service VM | user VM（virtioGPU） | user VM（virtioGPU + SRIOV） | user VM（GPU passthrough）
-- | -- | -- | -- | --
FPS | 11.7 | 1.0 | 1.0 | 11.9
Score | 491 | 42 | 43 | 496
Min FPS | 7.3 | 1.0 | 1.0 | 7.9
Max FPS | 21.4 | 1.3 | 1.3 | 21.4
Hardinfo → GPU drawing | 7676.74 | 16817.65 | 23156.11 | 11087.09

</div></b>

davidlamblin, 2022-08-30:
@yakuizhao  @fuzhongl

fuzhongl, 2022-08-30:
@davidlamblin 
Checked with GPU expert; please use `opencl` to test the performance of User VM.
Thanks!

davidlamblin, 2022-08-30:
could you please say something detailed?
how can i verify that vf is working and how about the score of Unigine_Valley-1.0?

davidlamblin, 2022-08-30:
i found the vga hardware not doing well,  so  fall back to use llvmpipe as opengl renderer. 
input  glxinfo | grep OpenGL in user vm, showing:

OpenGL vendor string: Mesa/X.org
OpenGL renderer string: llvmpipe (LLVM 12.0.0, 256 bits)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 21.2.6
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.1 Mesa 21.2.6
OpenGL shading language version string: 1.40
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 21.2.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:

And in user VM， input dmesg | grep i915， showing that：
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.15.44-acrn-user-vm-dirtyy
 root=UUID=acc29708-3ba1-461f-a9d9-694204c85a0d ro quiet splash vt.handoff=7 i911
5.enable_guc=0x7
[    0.017936] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.15.44-acrn-user-vv
m-dirty root=UUID=acc29708-3ba1-461f-a9d9-694204c85a0d ro quiet splash vt.handoff
f=7 i915.enable_guc=0x7
[    0.018042] Setting dangerous option i915.enable_guc - tainting kernel
[    1.015225] i915 0000:00:02.0: drm_WARN_ON((root->device & 0xFF00) != 0x9A00)
[    1.015234] WARNING: CPU: 4 PID: 1 at drivers/gpu/drm/i915/intel_device_info..
c:231 intel_device_info_subplatform_init+0x1e1/0x200
[    1.015261]  i915_driver_probe+0x134/0xdd0
[    1.015264]  i915_pci_probe+0xd1/0x1c0
[    1.015286]  i915_init+0x1d/0x8b
[    1.015337] i915 0000:00:02.0: Running in SR-IOV VF mode
[    1.016058] i915 0000:00:02.0: GuC interface version 0.1.0.0
[    1.022841] i915 0000:00:02.0: GuC interface version 0.1.0.0
[    1.023269] i915 0000:00:02.0: GuC firmware PRELOADED version 1.0 submission::
SR-IOV VF
[    1.023271] i915 0000:00:02.0: HuC firmware PRELOADED
[    1.027264] i915 0000:00:02.0: [drm] PMU not supported for this GPU.
[    1.027342] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0

davidlamblin, 2022-08-30:
@fuzhongl

davidlamblin, 2022-08-31:
so how to use the VF GPU？

davidlamblin, 2022-10-12:
is a bug or not ??  @yakuizhao @fuzhongl

davidlamblin, 2022-10-28:
@fuzhongl  uservm need to  support two GPU? one is VF, the another is virtio GPU? 
is the code open source?

fuzhongl, 2022-10-28:
> @fuzhongl uservm need to support two GPU? one is VF, the another is virtio GPU? is the code open source?

@davidlamblin yes, the code is open sourced for function; but need more Mesa patch for performance.
Unfortunately, these Mesa patch aren't  open source yet; which is out of ACRN scope.

davidlamblin, 2022-10-28:
where 

> > @fuzhongl uservm need to support two GPU? one is VF, the another is virtio GPU? is the code open source?
> 
> @davidlamblin yes, the code is open sourced for function; but need more Mesa patch for performance. Unfortunately, these Mesa patch aren't open source yet; which is out of ACRN scope.

could you please share the open source link ?

fuzhongl, 2022-10-28:
> where
>
>
> could you please share the open source link ?

Already in ACRN master branch for GPU VF and virtio GPU。

davidlamblin, 2022-10-28:
is  origin/release_3.1  support ? @fuzhongl

fuzhongl, 2022-10-28:
yes, this feature is enabled at v3.0.