projectacrn/acrn-hypervisor #7984: Can not find virtio drivers when install Windows10 user VM

zihengL1, 2022-08-12:


zihengL1, 2022-08-12:
We followed the "Run Windows as the User VM OS" guide to install win10. After we run "install_win.sh" scripts we saw Win10 installation screen. In the fourth step of win10 installation, we cannot find the virtio drivers (There only a Boot(X:) disk that we can choose, and there is no driver in it). Therefore, we are stuck at this stage and want to seek any help.

zihengL1, 2022-08-12:
Our platform: TGL
Product Name: NUC11TNBi5

Our install_win.sh scripts:

#!/bin/bash
#
# Copyright (C) 2022 Intel Corporation.
#
# SPDX-License-Identifier: BSD-3-Clause
#

# Launch script for VM name: VM2

#

# Helper functions

function probe_modules() {
    modprobe pci_stub
}

function offline_cpus() {
    # Each parameter of this function is considered the APIC ID (as is reported in /proc/cpuinfo, in decimal) of a CPU
    # assigned to a post-launched RTVM.
    for i in $*; do
        processor_id=$(grep -B 15 "apicid.*: ${i}$" /proc/cpuinfo | grep "^processor" | head -n 1 | cut -d ' ' -f 2)
        if [ -z ${processor_id} ]; then
            continue
        fi
        cpu_path="/sys/devices/system/cpu/cpu${processor_id}"
        if [ -f ${cpu_path}/online ]; then
            online=`cat ${cpu_path}/online`
            echo cpu${processor_id} online=${online} >> /dev/stderr
            if [ "${online}" = "1" ] && [ "${processor_id}" != "0" ]; then
                echo 0 > ${cpu_path}/online
                online=`cat ${cpu_path}/online`
                # during boot time, cpu hotplug may be disabled by pci_device_probe during a pci module insmod
                while [ "${online}" = "1" ]; do
                    sleep 1
                    echo 0 > ${cpu_path}/online
                    online=`cat ${cpu_path}/online`
                done
                echo ${processor_id} > /sys/devices/virtual/misc/acrn_hsm/remove_cpu
            fi
        fi
    done
}

function unbind_device() {
    physical_bdf=$1

    vendor_id=$(cat /sys/bus/pci/devices/${physical_bdf}/vendor)
    device_id=$(cat /sys/bus/pci/devices/${physical_bdf}/device)

    echo $(printf "%04x %04x" ${vendor_id} ${device_id}) > /sys/bus/pci/drivers/pci-stub/new_id
    echo ${physical_bdf} > /sys/bus/pci/devices/${physical_bdf}/driver/unbind
    echo ${physical_bdf} > /sys/bus/pci/drivers/pci-stub/bind
}

function create_tap() {
    # create a unique tap device for each VM
    tap=$1
    tap_exist=$(ip a show dev $tap)
    if [ "$tap_exist"x != "x" ]; then
        echo "$tap TAP device already available, reusing it."
    else
        ip tuntap add dev $tap mode tap
    fi

[... 149 lines cut ...]

# Initializing
probe_modules
mac=$(cat /sys/class/net/e*/address)

# Invoking ACRN device model
dm_params=(
    --windows
    `add_cpus                                 4 6`
    -m 4096M
    --ovmf /usr/share/acrn/bios/OVMF.fd
    `add_virtual_device                       1:0 lpc`
    `add_virtual_device                       0:0 hostbridge`
    `add_virtual_device                       3 virtio-net tap=WaaG,mac_seed=${mac:0:17}-VM2`
    `add_virtual_device                       4 virtio-gpu geometry=1024x768+100+100`
    `add_passthrough_device                   5 0000:00:14.0 d3hot_reset`
    `add_interrupt_storm_monitor              10000 10 1 100`
    `add_logger_settings                      console=4 kmsg=3 disk=5`
    VM2
)

echo "Launch device model with parameters: ${dm_params[@]}"
acrn-dm "${dm_params[@]}"

# Deinitializing

zihengL1, 2022-08-12:
Our install_win.sh scripts:

#!/bin/bash
#
# Copyright (C) 2022 Intel Corporation.
#
# SPDX-License-Identifier: BSD-3-Clause
#

# Launch script for VM name: VM2

#

# Helper functions

function probe_modules() {
    modprobe pci_stub
}

function offline_cpus() {
    # Each parameter of this function is considered the APIC ID (as is reported in /proc/cpuinfo, in decimal) of a CPU
    # assigned to a post-launched RTVM.
    for i in $*; do
        processor_id=$(grep -B 15 "apicid.*: ${i}$" /proc/cpuinfo | grep "^processor" | head -n 1 | cut -d ' ' -f 2)
        if [ -z ${processor_id} ]; then
            continue
        fi
        cpu_path="/sys/devices/system/cpu/cpu${processor_id}"
        if [ -f ${cpu_path}/online ]; then
            online=`cat ${cpu_path}/online`
            echo cpu${processor_id} online=${online} >> /dev/stderr
            if [ "${online}" = "1" ] && [ "${processor_id}" != "0" ]; then
                echo 0 > ${cpu_path}/online
                online=`cat ${cpu_path}/online`
                # during boot time, cpu hotplug may be disabled by pci_device_probe during a pci module insmod
                while [ "${online}" = "1" ]; do
                    sleep 1
                    echo 0 > ${cpu_path}/online
                    online=`cat ${cpu_path}/online`
                done
                echo ${processor_id} > /sys/devices/virtual/misc/acrn_hsm/remove_cpu
            fi
        fi
    done
}

function unbind_device() {
    physical_bdf=$1

    vendor_id=$(cat /sys/bus/pci/devices/${physical_bdf}/vendor)
    device_id=$(cat /sys/bus/pci/devices/${physical_bdf}/device)

    echo $(printf "%04x %04x" ${vendor_id} ${device_id}) > /sys/bus/pci/drivers/pci-stub/new_id
    echo ${physical_bdf} > /sys/bus/pci/devices/${physical_bdf}/driver/unbind
    echo ${physical_bdf} > /sys/bus/pci/drivers/pci-stub/bind
}

function create_tap() {
    # create a unique tap device for each VM
    tap=$1
    tap_exist=$(ip a show dev $tap)
    if [ "$tap_exist"x != "x" ]; then
        echo "$tap TAP device already available, reusing it."
    else
        ip tuntap add dev $tap mode tap
    fi

    # if acrn-br0 exists, add VM's unique tap device under it
    br_exist=$(ip a | grep acrn-br0 | awk '{print $1}')
    if [ "$br_exist"x != "x" -a "$tap_exist"x = "x" ]; then
[... 146 lines cut ...]

# Initializing
probe_modules
mac=$(cat /sys/class/net/e*/address)

# Invoking ACRN device model
dm_params=(
    --windows
    `add_cpus                                 4 6`
    -m 4096M
    --ovmf /usr/share/acrn/bios/OVMF.fd
    `add_virtual_device                       1:0 lpc`
    `add_virtual_device                       0:0 hostbridge`
    `add_virtual_device                       3 virtio-net tap=WaaG,mac_seed=${mac:0:17}-VM2`
    `add_virtual_device                       4 virtio-gpu geometry=1024x768+100+100`
    `add_passthrough_device                   5 0000:00:14.0 d3hot_reset`
    `add_interrupt_storm_monitor              10000 10 1 100`
    `add_logger_settings                      console=4 kmsg=3 disk=5`
    VM2
)

echo "Launch device model with parameters: ${dm_params[@]}"
acrn-dm "${dm_params[@]}"

# Deinitializing

zihengL1, 2022-08-12:
Our install_win.sh scprit:

`#!/bin/bash
#
# Copyright (C) 2022 Intel Corporation.
#
# SPDX-License-Identifier: BSD-3-Clause
#

# Launch script for VM name: VM2

#

# Helper functions

function probe_modules() {
    modprobe pci_stub
}

function offline_cpus() {
    # Each parameter of this function is considered the APIC ID (as is reported in /proc/cpuinfo, in decimal) of a CPU
    # assigned to a post-launched RTVM.
    for i in $*; do
        processor_id=$(grep -B 15 "apicid.*: ${i}$" /proc/cpuinfo | grep "^processor" | head -n 1 | cut -d ' ' -f 2)
        if [ -z ${processor_id} ]; then
            continue
        fi
        cpu_path="/sys/devices/system/cpu/cpu${processor_id}"
        if [ -f ${cpu_path}/online ]; then
            online=`cat ${cpu_path}/online`
            echo cpu${processor_id} online=${online} >> /dev/stderr
            if [ "${online}" = "1" ] && [ "${processor_id}" != "0" ]; then
                echo 0 > ${cpu_path}/online
                online=`cat ${cpu_path}/online`
                # during boot time, cpu hotplug may be disabled by pci_device_probe during a pci module insmod
                while [ "${online}" = "1" ]; do
                    sleep 1
                    echo 0 > ${cpu_path}/online
                    online=`cat ${cpu_path}/online`
                done
                echo ${processor_id} > /sys/devices/virtual/misc/acrn_hsm/remove_cpu
            fi
        fi
    done
}

function unbind_device() {
    physical_bdf=$1

    vendor_id=$(cat /sys/bus/pci/devices/${physical_bdf}/vendor)
    device_id=$(cat /sys/bus/pci/devices/${physical_bdf}/device)

    echo $(printf "%04x %04x" ${vendor_id} ${device_id}) > /sys/bus/pci/drivers/pci-stub/new_id
    echo ${physical_bdf} > /sys/bus/pci/devices/${physical_bdf}/driver/unbind
    echo ${physical_bdf} > /sys/bus/pci/drivers/pci-stub/bind
}

function create_tap() {
    # create a unique tap device for each VM
    tap=$1
    tap_exist=$(ip a show dev $tap)
    if [ "$tap_exist"x != "x" ]; then
        echo "$tap TAP device already available, reusing it."
    else
        ip tuntap add dev $tap mode tap
    fi

    # if acrn-br0 exists, add VM's unique tap device under it
    br_exist=$(ip a | grep acrn-br0 | awk '{print $1}')
    if [ "$br_exist"x != "x" -a "$tap_exist"x = "x" ]; then
[... 147 lines cut ...]
# Initializing
probe_modules
mac=$(cat /sys/class/net/e*/address)

# Invoking ACRN device model
dm_params=(
    --windows
    `add_cpus                                 4 6`
    -m 4096M
    --ovmf /usr/share/acrn/bios/OVMF.fd
    `add_virtual_device                       1:0 lpc`
    `add_virtual_device                       0:0 hostbridge`
    `add_virtual_device                       3 virtio-net tap=WaaG,mac_seed=${mac:0:17}-VM2`
    `add_virtual_device                       4 virtio-gpu geometry=1024x768+100+100`
    `add_passthrough_device                   5 0000:00:14.0 d3hot_reset`
    `add_interrupt_storm_monitor              10000 10 1 100`
    `add_logger_settings                      console=4 kmsg=3 disk=5`
    VM2
)

echo "Launch device model with parameters: ${dm_params[@]}"
acrn-dm "${dm_params[@]}"

# Deinitializing
`

zihengL1, 2022-08-12:
`function probe_modules() {
    modprobe pci_stub
}

function offline_cpus() {
    # Each parameter of this function is considered the APIC ID (as is reported in /proc/cpuinfo, in decimal) of a CPU
    # assigned to a post-launched RTVM.
    for i in $*; do
        processor_id=$(grep -B 15 "apicid.*: ${i}$" /proc/cpuinfo | grep "^processor" | head -n 1 | cut -d ' ' -f 2)
        if [ -z ${processor_id} ]; then
            continue
        fi
        cpu_path="/sys/devices/system/cpu/cpu${processor_id}"
        if [ -f ${cpu_path}/online ]; then
            online=`cat ${cpu_path}/online`
            echo cpu${processor_id} online=${online} >> /dev/stderr
            if [ "${online}" = "1" ] && [ "${processor_id}" != "0" ]; then
                echo 0 > ${cpu_path}/online
                online=`cat ${cpu_path}/online`
                # during boot time, cpu hotplug may be disabled by pci_device_probe during a pci module insmod
                while [ "${online}" = "1" ]; do
                    sleep 1
                    echo 0 > ${cpu_path}/online
                    online=`cat ${cpu_path}/online`
                done
                echo ${processor_id} > /sys/devices/virtual/misc/acrn_hsm/remove_cpu
            fi
        fi
    done
}

function unbind_device() {
    physical_bdf=$1

    vendor_id=$(cat /sys/bus/pci/devices/${physical_bdf}/vendor)
    device_id=$(cat /sys/bus/pci/devices/${physical_bdf}/device)

    echo $(printf "%04x %04x" ${vendor_id} ${device_id}) > /sys/bus/pci/drivers/pci-stub/new_id
    echo ${physical_bdf} > /sys/bus/pci/devices/${physical_bdf}/driver/unbind
    echo ${physical_bdf} > /sys/bus/pci/drivers/pci-stub/bind
}

function create_tap() {
    # create a unique tap device for each VM
    tap=$1
    tap_exist=$(ip a show dev $tap)
    if [ "$tap_exist"x != "x" ]; then
        echo "$tap TAP device already available, reusing it."
    else
        ip tuntap add dev $tap mode tap
    fi

    # if acrn-br0 exists, add VM's unique tap device under it
    br_exist=$(ip a | grep acrn-br0 | awk '{print $1}')
    if [ "$br_exist"x != "x" -a "$tap_exist"x = "x" ]; then
        echo "acrn-br0 bridge already exists, adding new $tap TAP device to it..."
        ip link set "$tap" master acrn-br0
        ip link set dev "$tap" down
        ip link set dev "$tap" up
    fi
}

function mount_partition() {
    partition=$1

    tmpdir=`mktemp -d`
    mount ${partition} ${tmpdir}
    echo ${tmpdir}
}

[... 131 lines cut ...]

# Initializing
probe_modules
mac=$(cat /sys/class/net/e*/address)

# Invoking ACRN device model
dm_params=(
    --windows
    `add_cpus                                 4 6`
    -m 4096M
    --ovmf /usr/share/acrn/bios/OVMF.fd
    `add_virtual_device                       1:0 lpc`
    `add_virtual_device                       0:0 hostbridge`
    `add_virtual_device                       3 virtio-net tap=WaaG,mac_seed=${mac:0:17}-VM2`
    `add_virtual_device                       4 virtio-gpu geometry=1024x768+100+100`
    `add_passthrough_device                   5 0000:00:14.0 d3hot_reset`
    `add_interrupt_storm_monitor              10000 10 1 100`
    `add_logger_settings                      console=4 kmsg=3 disk=5`
    VM2
)

echo "Launch device model with parameters: ${dm_params[@]}"
acrn-dm "${dm_params[@]}"

# Deinitializing`

zihengL1, 2022-08-12:
Our install_win.sh scripts:

`function probe_modules() {
    modprobe pci_stub
}

function offline_cpus() {

    for i in $*; do
        processor_id=$(grep -B 15 "apicid.*: ${i}$" /proc/cpuinfo | grep "^processor" | head -n 1 | cut -d ' ' -f 2)
        if [ -z ${processor_id} ]; then
            continue
        fi
        cpu_path="/sys/devices/system/cpu/cpu${processor_id}"
        if [ -f ${cpu_path}/online ]; then
            online=`cat ${cpu_path}/online`
            echo cpu${processor_id} online=${online} >> /dev/stderr
            if [ "${online}" = "1" ] && [ "${processor_id}" != "0" ]; then
                echo 0 > ${cpu_path}/online
                online=`cat ${cpu_path}/online`

                while [ "${online}" = "1" ]; do
                    sleep 1
                    echo 0 > ${cpu_path}/online
                    online=`cat ${cpu_path}/online`
                done
                echo ${processor_id} > /sys/devices/virtual/misc/acrn_hsm/remove_cpu
            fi
        fi
    done
}

function unbind_device() {
    physical_bdf=$1

    vendor_id=$(cat /sys/bus/pci/devices/${physical_bdf}/vendor)
    device_id=$(cat /sys/bus/pci/devices/${physical_bdf}/device)

    echo $(printf "%04x %04x" ${vendor_id} ${device_id}) > /sys/bus/pci/drivers/pci-stub/new_id
    echo ${physical_bdf} > /sys/bus/pci/devices/${physical_bdf}/driver/unbind
    echo ${physical_bdf} > /sys/bus/pci/drivers/pci-stub/bind
}

function create_tap() {

    tap=$1
    tap_exist=$(ip a show dev $tap)
    if [ "$tap_exist"x != "x" ]; then
        echo "$tap TAP device already available, reusing it."
    else
        ip tuntap add dev $tap mode tap
    fi

    br_exist=$(ip a | grep acrn-br0 | awk '{print $1}')
    if [ "$br_exist"x != "x" -a "$tap_exist"x = "x" ]; then
        echo "acrn-br0 bridge already exists, adding new $tap TAP device to it..."
        ip link set "$tap" master acrn-br0
        ip link set dev "$tap" down
        ip link set dev "$tap" up
    fi
}

function mount_partition() {
    partition=$1

    tmpdir=`mktemp -d`
    mount ${partition} ${tmpdir}
    echo ${tmpdir}
}

[... 117 lines cut ...]
}

vm_type='STANDARD_VM'
scheduler='SCHED_BVT'

probe_modules
mac=$(cat /sys/class/net/e*/address)

dm_params=(
    --windows
    `add_cpus                                 4 6`
    -m 4096M
    --ovmf /usr/share/acrn/bios/OVMF.fd
    `add_virtual_device                       1:0 lpc`
    `add_virtual_device                       0:0 hostbridge`
    `add_virtual_device                       3 virtio-net tap=WaaG,mac_seed=${mac:0:17}-VM2`
    `add_virtual_device                       4 virtio-gpu geometry=1024x768+100+100`
    `add_passthrough_device                   5 0000:00:14.0 d3hot_reset`
    `add_interrupt_storm_monitor              10000 10 1 100`
    `add_logger_settings                      console=4 kmsg=3 disk=5`
    VM2
)

echo "Launch device model with parameters: ${dm_params[@]}"
acrn-dm "${dm_params[@]}"`

zihengL1, 2022-08-12:
Our install_win.sh scripts:

**
function probe_modules() {
    modprobe pci_stub
}

function offline_cpus() {

    for i in $*; do
        processor_id=$(grep -B 15 "apicid.*: ${i}$" /proc/cpuinfo | grep "^processor" | head -n 1 | cut -d ' ' -f 2)
        if [ -z ${processor_id} ]; then
            continue
        fi
        cpu_path="/sys/devices/system/cpu/cpu${processor_id}"
        if [ -f ${cpu_path}/online ]; then
            online=`cat ${cpu_path}/online`
            echo cpu${processor_id} online=${online} >> /dev/stderr
            if [ "${online}" = "1" ] && [ "${processor_id}" != "0" ]; then
                echo 0 > ${cpu_path}/online
                online=`cat ${cpu_path}/online`

                while [ "${online}" = "1" ]; do
                    sleep 1
                    echo 0 > ${cpu_path}/online
                    online=`cat ${cpu_path}/online`
                done
                echo ${processor_id} > /sys/devices/virtual/misc/acrn_hsm/remove_cpu
            fi
        fi
    done
}

function unbind_device() {
    physical_bdf=$1

    vendor_id=$(cat /sys/bus/pci/devices/${physical_bdf}/vendor)
    device_id=$(cat /sys/bus/pci/devices/${physical_bdf}/device)

    echo $(printf "%04x %04x" ${vendor_id} ${device_id}) > /sys/bus/pci/drivers/pci-stub/new_id
    echo ${physical_bdf} > /sys/bus/pci/devices/${physical_bdf}/driver/unbind
    echo ${physical_bdf} > /sys/bus/pci/drivers/pci-stub/bind
}

function create_tap() {

    tap=$1
    tap_exist=$(ip a show dev $tap)
    if [ "$tap_exist"x != "x" ]; then
        echo "$tap TAP device already available, reusing it."
    else
        ip tuntap add dev $tap mode tap
    fi

    br_exist=$(ip a | grep acrn-br0 | awk '{print $1}')
    if [ "$br_exist"x != "x" -a "$tap_exist"x = "x" ]; then
        echo "acrn-br0 bridge already exists, adding new $tap TAP device to it..."
        ip link set "$tap" master acrn-br0
        ip link set dev "$tap" down
        ip link set dev "$tap" up
    fi
}

function mount_partition() {
    partition=$1

    tmpdir=`mktemp -d`
    mount ${partition} ${tmpdir}
    echo ${tmpdir}
}
[... 119 lines cut ...]

vm_type='STANDARD_VM'
scheduler='SCHED_BVT'

probe_modules
mac=$(cat /sys/class/net/e*/address)

dm_params=(
    --windows
    `add_cpus                                 4 6`
    -m 4096M
    --ovmf /usr/share/acrn/bios/OVMF.fd
    `add_virtual_device                       1:0 lpc`
    `add_virtual_device                       0:0 hostbridge`
    `add_virtual_device                       3 virtio-net tap=WaaG,mac_seed=${mac:0:17}-VM2`
    `add_virtual_device                       4 virtio-gpu geometry=1024x768+100+100`
    `add_passthrough_device                   5 0000:00:14.0 d3hot_reset`
    `add_interrupt_storm_monitor              10000 10 1 100`
    `add_logger_settings                      console=4 kmsg=3 disk=5`
    VM2
)

echo "Launch device model with parameters: ${dm_params[@]}"
acrn-dm "${dm_params[@]}"
**

zihengL1, 2022-08-12:
``` function probe_modules() {
    modprobe pci_stub
}

function offline_cpus() {
    # Each parameter of this function is considered the APIC ID (as is reported in /proc/cpuinfo, in decimal) of a CPU
    # assigned to a post-launched RTVM.
    for i in $*; do
        processor_id=$(grep -B 15 "apicid.*: ${i}$" /proc/cpuinfo | grep "^processor" | head -n 1 | cut -d ' ' -f 2)
        if [ -z ${processor_id} ]; then
            continue
        fi
        cpu_path="/sys/devices/system/cpu/cpu${processor_id}"
        if [ -f ${cpu_path}/online ]; then
            online=`cat ${cpu_path}/online`
            echo cpu${processor_id} online=${online} >> /dev/stderr
            if [ "${online}" = "1" ] && [ "${processor_id}" != "0" ]; then
                echo 0 > ${cpu_path}/online
                online=`cat ${cpu_path}/online`
                # during boot time, cpu hotplug may be disabled by pci_device_probe during a pci module insmod
                while [ "${online}" = "1" ]; do
                    sleep 1
                    echo 0 > ${cpu_path}/online
                    online=`cat ${cpu_path}/online`
                done
                echo ${processor_id} > /sys/devices/virtual/misc/acrn_hsm/remove_cpu
            fi
        fi
    done
}

function unbind_device() {
    physical_bdf=$1

    vendor_id=$(cat /sys/bus/pci/devices/${physical_bdf}/vendor)
    device_id=$(cat /sys/bus/pci/devices/${physical_bdf}/device)

    echo $(printf "%04x %04x" ${vendor_id} ${device_id}) > /sys/bus/pci/drivers/pci-stub/new_id
    echo ${physical_bdf} > /sys/bus/pci/devices/${physical_bdf}/driver/unbind
    echo ${physical_bdf} > /sys/bus/pci/drivers/pci-stub/bind
}

function create_tap() {
    # create a unique tap device for each VM
    tap=$1
    tap_exist=$(ip a show dev $tap)
    if [ "$tap_exist"x != "x" ]; then
        echo "$tap TAP device already available, reusing it."
    else
        ip tuntap add dev $tap mode tap
    fi

    # if acrn-br0 exists, add VM's unique tap device under it
    br_exist=$(ip a | grep acrn-br0 | awk '{print $1}')
    if [ "$br_exist"x != "x" -a "$tap_exist"x = "x" ]; then
        echo "acrn-br0 bridge already exists, adding new $tap TAP device to it..."
        ip link set "$tap" master acrn-br0
        ip link set dev "$tap" down
        ip link set dev "$tap" up
    fi
}

function mount_partition() {
    partition=$1

    tmpdir=`mktemp -d`
    mount ${partition} ${tmpdir}
    echo ${tmpdir}
}

[... 129 lines cut ...]
vm_type='STANDARD_VM'
scheduler='SCHED_BVT'

# Initializing
probe_modules
mac=$(cat /sys/class/net/e*/address)

# Invoking ACRN device model
dm_params=(
    --windows
    `add_cpus                                 4 6`
    -m 4096M
    --ovmf /usr/share/acrn/bios/OVMF.fd
    `add_virtual_device                       1:0 lpc`
    `add_virtual_device                       0:0 hostbridge`
    `add_virtual_device                       3 virtio-net tap=WaaG,mac_seed=${mac:0:17}-VM2`
    `add_virtual_device                       4 virtio-gpu geometry=1024x768+100+100`
    `add_passthrough_device                   5 0000:00:14.0 d3hot_reset`
    `add_interrupt_storm_monitor              10000 10 1 100`
    `add_logger_settings                      console=4 kmsg=3 disk=5`
    VM2
)

echo "Launch device model with parameters: ${dm_params[@]}"
acrn-dm "${dm_params[@]}"```

zihengL1, 2022-08-12:
Our install_win script:
```
# Helper functions

function probe_modules() {
    modprobe pci_stub
}

function offline_cpus() {
    # Each parameter of this function is considered the APIC ID (as is reported in /proc/cpuinfo, in decimal) of a CPU
    # assigned to a post-launched RTVM.
    for i in $*; do
        processor_id=$(grep -B 15 "apicid.*: ${i}$" /proc/cpuinfo | grep "^processor" | head -n 1 | cut -d ' ' -f 2)
        if [ -z ${processor_id} ]; then
            continue
        fi
        cpu_path="/sys/devices/system/cpu/cpu${processor_id}"
        if [ -f ${cpu_path}/online ]; then
            online=`cat ${cpu_path}/online`
            echo cpu${processor_id} online=${online} >> /dev/stderr
            if [ "${online}" = "1" ] && [ "${processor_id}" != "0" ]; then
                echo 0 > ${cpu_path}/online
                online=`cat ${cpu_path}/online`
                # during boot time, cpu hotplug may be disabled by pci_device_probe during a pci module insmod
                while [ "${online}" = "1" ]; do
                    sleep 1
                    echo 0 > ${cpu_path}/online
                    online=`cat ${cpu_path}/online`
                done
                echo ${processor_id} > /sys/devices/virtual/misc/acrn_hsm/remove_cpu
            fi
        fi
    done
}

function unbind_device() {
    physical_bdf=$1

    vendor_id=$(cat /sys/bus/pci/devices/${physical_bdf}/vendor)
    device_id=$(cat /sys/bus/pci/devices/${physical_bdf}/device)

    echo $(printf "%04x %04x" ${vendor_id} ${device_id}) > /sys/bus/pci/drivers/pci-stub/new_id
    echo ${physical_bdf} > /sys/bus/pci/devices/${physical_bdf}/driver/unbind
    echo ${physical_bdf} > /sys/bus/pci/drivers/pci-stub/bind
}

function create_tap() {
    # create a unique tap device for each VM
    tap=$1
    tap_exist=$(ip a show dev $tap)
    if [ "$tap_exist"x != "x" ]; then
        echo "$tap TAP device already available, reusing it."
    else
        ip tuntap add dev $tap mode tap
    fi

    # if acrn-br0 exists, add VM's unique tap device under it
    br_exist=$(ip a | grep acrn-br0 | awk '{print $1}')
    if [ "$br_exist"x != "x" -a "$tap_exist"x = "x" ]; then
        echo "acrn-br0 bridge already exists, adding new $tap TAP device to it..."
        ip link set "$tap" master acrn-br0
        ip link set dev "$tap" down
        ip link set dev "$tap" up
    fi
}

function mount_partition() {
    partition=$1

    tmpdir=`mktemp -d`
[... 140 lines cut ...]
# Invoking ACRN device model
dm_params=(
    --windows
    `add_cpus                                 0 2`
    -m 4096M
    --ovmf /home/tianshu/work/OVMF.fd
    `add_virtual_device                       1:0 lpc`
    `add_virtual_device                       0:0 hostbridge`
    `add_virtual_device                       3 virtio-net tap=WaaG,mac_seed=${mac:0:17}-POST_STD_VM1`
    `add_virtual_device                       5 ahci /home/tianshu/work/Windows10.iso `
    `add_virtual_device                       2 ahci /home/tianshu/work/winvirtio.iso `
    `add_virtual_device                       4 virtio-blk /home/tianshu/work/win10-ltsc.img`
    #`add_virtual_device                       2 passthru 0/2/0 `
    #`add_virtual_device 			8 passthru 0/14/0 d3hot_reset`
    #`add_passthrough_device                   2 0000:00:02.1 igd-vf`
    `add_virtual_device                       6 virtio-gpu,geometry=1024x768+100+100 ./`
    `add_logger_settings                      console=4 kmsg=3 disk=5`
    POST_STD_VM1
)

echo "Launch device model with parameters: ${dm_params[@]}"
acrn-dm "${dm_params[@]}"

# Deinitializing
```

zihengL1, 2022-08-13:
[image]

gvancuts, 2022-08-18:
Can you share the output of `pgrep acrn-dm` when this is running?

You do have this file on your machine, don't you? `/home/tianshu/work/winvirtio.iso`