projectacrn/acrn-hypervisor #8010: Access acrn-configurator from network

liyimeng, 2022-08-17:
The acrn-configurator seem based no nodejs, can the configuration GUI be accessed from network? 

If not, it would be nice to make it work like that, developer then can perform configuration from a web browser.

gvancuts, 2022-08-18:
What do you mean by "accessed from the network"? Do you mean via a web browser?

I do not believe accessing the Configurator via the network is possible but I'll let the engineers tell me if I'm wrong (cc/ @junjiemao1 )

junjiemao1, 2022-08-18:
While based on js, the configurator is packed using [Tauri](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwiC4sieydH5AhWoIkQIHUY2Dn8QFnoECAUQAQ&url=https%3A%2F%2Ftauri.app%2F&usg=AOvVaw1oLQfZbMy3_vhS-tC3CXIH) as a desktop application.

An alternative is to build the configurator on your local machine which you would like to have the browser open on.

liyimeng, 2022-08-19:
Hi  @gvancuts  junjiemao1 get my point perfectly.  With today tauri based app, the app itself need hundreds of MBs  space, even worse, it need an desktop, it will pull in GBs of dependencies, and install them into my development machine. it is  fine in early development phase, but bad when moving to scale. 

A simple solution would be using the already installed python, start a simple web server backend, replacing the rust backend it used today.  With this solution, development get much more flexible use case.: Config locally from the development machine If there is a web browser there, or remotely from another machine which have nice GUI.

Again, not a big deal in early phase, but make huge different when go scale, e.g. auto-deployment in thousands of IoT devices.

gvancuts, 2022-08-19:
Hi @liyimeng , thanks for the description and explanation, I'm not yet sure I fully understand your use-case. When you talk about auto-deployment of thousands of IoT devices, do you envision that each will be different and will need a specific ACRN config?

The ACRN Configurator is really used much before the deployment phase and is no longer needed once you have a binary package that you want to deploy on your systems. I would expect in a typical development cycle that first a prototype is created, then the ACRN configuration is defined and tweaked during the development phase, then full testing and validation is done for a device (or set of devices) and the binary package(s) get deployed on the devices. But I do not see why you would still need the ACRN Configurator in that later phase. Can you explain some more?

gvancuts, 2022-08-22:
> @gvancuts Thank for attention. I am likely too new for ACRN. Since all the tutorial as developer to build everything from scratch, so I get an impression that everything need to rebuild right before deployment.

Not a problem at all @liyimeng , there are no right or wrong questions and we welcome all questions and new use-cases too!

> ACRN is a hypervisor, which likely used in cloud environment. Since we are not expecting homogeneous hardware setup cross the entire cloud, we likely need to rebuild the binaries for each setup. If the configurator tool use server-browser mode, it can be used as a service, developers can just upload board spec files and get back the configuration.Not sure if this add much value though.

The primary focus of ACRN tody is actually not cloud environments but embedded, IoT and Edge devices. While the hardware landscape is not completely homogeneous, it is expected that the number of HW variants will be restricted and ACRN will have a pre-defined configuration for each of those.