projectacrn/acrn-hypervisor #8081: Errno 11. Board configuration tool

SIA-77, 2022-08-29:
Hello,

I get the blocking IO error when using Board configuration tool. System is build from the scratch according to the guide. Details below.
Generated XML file is broken according to ACRN configurator. Any ideas how to fix this?

**Platform**
iPC, core i5 8th gen, 256 NVMe, 8Gb, 2 Eth, 6 serial, wi-fi

[**Logs**]
```
Generating board XML my_board. This may take a few minutes...
Traceback (most recent call last):
  File "/usr/bin/extractors/40-acpi-tables.py", line 29, in extract_gsi_number
    line =  os.read(fd, 512).decode("utf-8")
BlockingIOError: [Errno 11] Resource temporarily unavailable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/board_inspector.py", line 240, in <module>
    main(args.board_name, board_xml, args)
  File "/usr/bin/board_inspector.py", line 191, in main
    module.extract(args, board_etree)
  File "/usr/bin/extractors/40-acpi-tables.py", line 59, in extract
    extract_topology(ioapics_node, tables)
  File "/usr/bin/extractors/40-acpi-tables.py", line 48, in extract_topology
    extract_gsi_number(ioapic_node, apic_id)
  File "/usr/bin/extractors/40-acpi-tables.py", line 37, in extract_gsi_number
    add_child(ioapic_node, "gsi_number", DEFAULT_MAX_IOAPIC_LINES)
  File "/usr/bin/extractors/helpers.py", line 10, in add_child
    child.text = text
  File "src/lxml/etree.pyx", line 1039, in lxml.etree._Element.text.__set__
  File "src/lxml/apihelpers.pxi", line 747, in lxml.etree._setNodeText
  File "src/lxml/apihelpers.pxi", line 735, in lxml.etree._createTextNode
  File "src/lxml/apihelpers.pxi", line 1538, in lxml.etree._utf8
TypeError: Argument must be bytes or unicode, got 'int'
```

**Additional info from generated XML:**
```
<acrn-config board="my_board">
	<BIOS_INFO>
	BIOS Information
	Vendor: American Megatrends Inc.
	Version: 5.12
	Release Date: 11/25/2021
	BIOS Revision: 5.12
	</BIOS_INFO>

	<BASE_BOARD_INFO>
	Base Board Information
	Manufacturer: Default string
	Product Name: Default string
	Version: Default string
	</BASE_BOARD_INFO>

	<PCI_DEVICE>
	00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 08)
	00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07)
	Region 0: Memory at de000000 (64-bit, non-prefetchable) [size=16M]
	Region 2: Memory at c0000000 (64-bit, prefetchable) [size=256M]
```
.....
```
<CPU_BRAND>
	"Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz"
	</CPU_BRAND>
```
....
```
<BLOCK_DEVICE_INFO>
	/dev/nvme0n1p2: TYPE=&quot;ext4&quot;
	</BLOCK_DEVICE_INFO>

	<TTYS_INFO>
	seri:/dev/ttyS0 type:portio base:0x3F8 irq:4
	seri:/dev/ttyS1 type:portio base:0x2F8 irq:3
	seri:/dev/ttyS2 type:portio base:0x3E8 irq:7
	seri:/dev/ttyS3 type:portio base:0x2E8 irq:7
	seri:/dev/ttyS4 type:portio base:0x2F0 irq:7
	seri:/dev/ttyS5 type:portio base:0x2E0 irq:7
	</TTYS_INFO>

	<AVAILABLE_IRQ_INFO>
	5, 6, 10, 11, 13, 14, 15
	</AVAILABLE_IRQ_INFO>

	<TOTAL_MEM_INFO>
	8046792 kB
	</TOTAL_MEM_INFO>

	<CPU_PROCESSOR_INFO>
	0, 1, 2, 3, 4, 5, 6, 7
	</CPU_PROCESSOR_INFO>

	<MAX_MSIX_TABLE_NUM>
	16
	</MAX_MSIX_TABLE_NUM>
</acrn-config>
```

SIA-77, 2022-09-01:
Fixed by changing DEFAULT_MAX_IOAPIC_LINES parameter in 40-acpi-tables.py from 240 to  '240'. Extract_gsi_number function went by 'except' branch and coursed error due to the types mismatch.

gvancuts, 2022-09-05:
That's the line in question, isn't it? https://github.com/projectacrn/acrn-hypervisor/blob/master/misc/config_tools/board_inspector/extractors/40-acpi-tables.py#L14

Assuming this is the right fix, we need to fix this in our upstream code, would you be able to send a Pull Request for this please?

gvancuts, 2022-09-05:
You can use this Github issue in the `Tracked-On` field of your commit message. Thanks!

SIA-77, 2022-09-07:
Yes? it's that line. But in the latest version the bug was recently fixed (in a different way):
` while True:
        try:
            line =  os.read(fd, 512).decode("utf-8")
            m = re.match(r"\s*\d+,\d+,\d+,-;IOAPIC\[[\d+]\]:\s+apic_id\s+{},\s+version\s+\d+,\s+address\s+0x[0-9a-f]+,\s+GSI\s+(\d+)-(\d+)".format(apic_id), line)
            if m:
                previous_max = int(m.group(1))
                current_max = int(m.group(2)) + 1
                add_child(ioapic_node, "gsi_number", str(current_max - previous_max))
                break
        except:
            add_child(ioapic_node, "gsi_number", **str**(DEFAULT_MAX_IOAPIC_LINES))
            break
    os.close(fd)`
We can close the issue.

closed 2022-09-07 by gvancuts