projectacrn/acrn-hypervisor #8131: Not able to build qemu board

liyimeng, 2022-09-12:
**Describe the bug**
To try out qemu, I am following this instruction https://projectacrn.github.io/latest/tutorials/acrn_on_qemu.html 
when running 
```
/acrn-work/acrn-hypervisor# make BOARD=qemu SCENARIO=shared
make[1]: Entering directory '/acrn-work/acrn-hypervisor/hypervisor'
Board XML is fetched from /acrn-work/acrn-hypervisor/misc/config_tools/data/qemu/qemu.xml
Scenario XML is being fetched from /acrn-work/acrn-hypervisor/misc/config_tools/data/qemu/shared.xml
WARNING:root:The sum 0.0 of memory configured in post launch VMs should not be larger than the calculated total hugepages -3 of service VMs. Please update the configuration in post launch VMs
/acrn-work/acrn-hypervisor/build/hypervisor/configs/allocation.xml generated
/acrn-work/acrn-hypervisor/build/hypervisor/configs/unified.xml generated
/acrn-work/acrn-hypervisor/build/hypervisor/configs/config.mk generated
Traceback (most recent call last):
  File "/acrn-work/acrn-hypervisor/misc/config_tools/scenario_config/elementpath_overlay.py", line 100, in evaluate_number_of_clos_id_needed
    return len(rdt.get_policy_list(op.elem)) if op else 0
  File "/acrn-work/acrn-hypervisor/misc/config_tools/scenario_config/../library/rdt.py", line 163, in get_policy_list
    init_cache2_id_list(scenario_etree)
  File "/acrn-work/acrn-hypervisor/misc/config_tools/scenario_config/../library/rdt.py", line 189, in init_cache2_id_list
    cache2_id_list = scenario_etree.xpath("//CACHE_ALLOCATION[CACHE_LEVEL = 2]/CACHE_ID/text()")
AttributeError: 'Xsd11Element' object has no attribute 'xpath'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/xmlschema/validators/assertions.py", line 106, in build
    self.token = self.parser.parse(self.path)
  File "/usr/local/lib/python3.8/dist-packages/elementpath/xpath2/xpath2_parser.py", line 440, in parse
    for _ in root_token.select(context):
  File "/usr/local/lib/python3.8/dist-packages/elementpath/xpath_token.py", line 116, in select
    item = self.evaluate(context)
  File "/acrn-work/acrn-hypervisor/misc/config_tools/scenario_config/elementpath_overlay.py", line 137, in evaluate
    for results in copy(context).iter_product(selectors, varnames):
  File "/usr/local/lib/python3.8/dist-packages/elementpath/xpath_context.py", line 281, in iter_product
    value = next(iterators[k])
  File "/usr/local/lib/python3.8/dist-packages/elementpath/xpath_token.py", line 116, in select
    item = self.evaluate(context)
  File "/acrn-work/acrn-hypervisor/misc/config_tools/scenario_config/elementpath_overlay.py", line 102, in evaluate_number_of_clos_id_needed
    raise self.error('XPTY0004', err)
elementpath.exceptions.ElementPathTypeError: [err:XPTY0004] 'Xsd11Element' object has no attribute 'xpath'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/acrn-work/acrn-hypervisor/hypervisor/../misc/config_tools/scenario_config/validator.py", line 309, in <module>
    main(args)
  File "/acrn-work/acrn-hypervisor/hypervisor/../misc/config_tools/scenario_config/validator.py", line 286, in main
    validator_construction_pipeline.run(obj)
  File "/acrn-work/acrn-hypervisor/misc/config_tools/scenario_config/pipeline.py", line 68, in run
    stage.run(obj)
  File "/acrn-work/acrn-hypervisor/hypervisor/../misc/config_tools/scenario_config/validator.py", line 182, in run
    validator = ScenarioValidator(obj.get("schema_etree"), obj.get("datachecks_etree"))
  File "/acrn-work/acrn-hypervisor/hypervisor/../misc/config_tools/scenario_config/validator.py", line 74, in __init__
    self.datachecks = xmlschema.XMLSchema11(datachecks_etree) if datachecks_etree else None
  File "/usr/local/lib/python3.8/dist-packages/xmlschema/validators/schemas.py", line 521, in __init__
    self.maps.build()
  File "/usr/local/lib/python3.8/dist-packages/xmlschema/validators/global_maps.py", line 664, in build
    obj.build()
  File "/usr/local/lib/python3.8/dist-packages/xmlschema/validators/assertions.py", line 108, in build
    self.parse_error(err)
  File "/usr/local/lib/python3.8/dist-packages/xmlschema/validators/xsdbase.py", line 190, in parse_error
    raise error
xmlschema.validators.exceptions.XMLSchemaParseError: [err:XPTY0004] 'Xsd11Element' object has no attribute 'xpath':

Schema:

  <xs:assert xmlns:xs="http://www.w3.org/2001/XMLSchema" xmlns:acrn="https://projectacrn.org" xmlns:xi="http://www.w3.org/2003/XInclude" test="every $needed in number-of-clos-id-needed(/acrn-config) satisfies                    every $capacity in min(//caches/cache/capability[@id='CAT']/clos_number) satisfies                    $needed &lt; $capacity" xml:base="../../../../../../../misc/config_tools/schema/checks/rdt_support.xsd"><xs:annotation acrn:severity="error" acrn:report-on="/acrn-config/hv/CACHE_REGION"><xs:documentation>The current CAT configuration requires {string($needed)} CLOS IDs, which exceeds the capacity of the platform which supports {string($capacity)} at most.</xs:documentation></xs:annotation></xs:assert>

Path: /xs:schema/xs:element/xs:complexType/xs:assert[26]

make[1]: Leaving directory '/acrn-work/acrn-hypervisor/hypervisor'
make -C /acrn-work/acrn-hypervisor/hypervisor BOARD=qemu SCENARIO=shared HV_OBJDIR=/acrn-work/acrn-hypervisor/build/hypervisor RELEASE=
make[1]: Entering directory '/acrn-work/acrn-hypervisor/hypervisor'
/acrn-work/acrn-hypervisor/build/hypervisor/include/config.h generated
Start pre-build static check ...
make -C ../misc/hv_prebuild BOARD=qemu SCENARIO=shared TARGET_DIR=/acrn-work/acrn-hypervisor/build/hypervisor/configs
make[2]: Entering directory '/acrn-work/acrn-hypervisor/misc/hv_prebuild'
make[2]: *** No rule to make target '/acrn-work/acrn-hypervisor/build/hypervisor/configs/scenarios/shared/vm_configurations.c', needed by 'default'.  Stop.
make[2]: Leaving directory '/acrn-work/acrn-hypervisor/misc/hv_prebuild'
make[1]: *** [Makefile:419: pre_build] Error 2
make[1]: Leaving directory '/acrn-work/acrn-hypervisor/hypervisor'
make: *** [Makefile:130: hypervisor] Error 2
```
**Platform**
qemu  + ubuntu 20.04
**Codebase**

acrn 3.0 

**Scenario**

Not sure 

**To Reproduce**
Flow instructions in https://projectacrn.github.io/latest/tutorials/acrn_on_qemu.html step by step. 
When build acrn hypervisor, error happen.

**Expected behavior**

The tutorial should work

**Additional context**
Attach the following logs if available:

- native Linux boot kernel message log
- ACRN serial port log (when issues happens)
- Service VM kernel message when issue happens
- acrn-dm log: all files in /var/log/acrn-tm
- console log when running script to launch guest VM

fuzhongl, 2022-09-13:
@liyimeng 
Can't reproduce this issue with acrn 3.0 on my side:
```
ACRN Configuration Summary:
Build directory: /home/acrn/acrn-hypervisor/build/hypervisor
This build directory is configured with the settings below.
- BOARD = qemu
- SCENARIO = shared
- RELEASE = n
dpkg-deb: building package 'acrn-hypervisor' in 'acrn-qemu-shared-3.01.deb'.
ACRN debian package acrn-qemu-shared-3.01.deb was successfully created in the /home/acrn/acrn-hypervisor/build.
```
As said in GSG: https://projectacrn.github.io/latest/getting-started/getting-started.html#
Do you install the necessary ACRN build tools?

```
sudo apt install -y gcc git make vim libssl-dev libpciaccess-dev uuid-dev \
     libsystemd-dev libevent-dev libxml2-dev libxml2-utils libusb-1.0-0-dev \
     python3 python3-pip python3.8-venv libblkid-dev e2fslibs-dev \
     pkg-config libnuma-dev libcjson-dev liblz4-tool flex bison \
     xsltproc clang-format bc libpixman-1-dev libsdl2-dev libegl-dev \
     libgles-dev libdrm-dev gnu-efi libelf-dev
```

liyimeng, 2022-09-13:
I do miss out two of them 'gnu-efi libelf-dev' 

I install them, same error.  I then try to detect a new board config, 

```
root@acrn-dev:/acrn-work/acrn-hypervisor# board_inspector.py myboad
Generating board XML myboad. This may take a few minutes...
2022-09-13 07:36:23-root-ERROR:-Board inspector is running inside an unsupported Virtual Machine (VM). Only KVM or QEMU is supported. Unexpected results may occur.
Traceback (most recent call last):
  File "/usr/bin/board_inspector.py", line 240, in <module>
    main(args.board_name, board_xml, args)
  File "/usr/bin/board_inspector.py", line 191, in main
    module.extract(args, board_etree)
  File "/usr/bin/extractors/95-usb.py", line 27, in extract
    with open(devpath + '/port/firmware_node/path') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/bus/usb/devices/1-1/port/firmware_node/path'
root@acrn-dev:/acrn-work/acrn-hypervisor# 
```

I am wondering if the VM I created miss something.  my qemu version is like:
```
qemu-system-x86_64 -version
QEMU emulator version 6.1.1
Copyright (c) 2003-2021 Fabrice Bellard and the QEMU Project developers
```

With the error above, could it be a bug? My understanding is that not all device have separated firmware. Board inspector seem assume there is alway one, right?

fuzhongl, 2022-09-13:
It is 4.2.1 on my ubuntu 20.04.
```
root@acrn-desktop:/home/acrn# qemu-system-x86_64 -version
QEMU emulator version 4.2.1 (Debian 1:4.2-3ubuntu6.23)
Copyright (c) 2003-2019 Fabrice Bellard and the QEMU Project developers
root@acrn-desktop:/home/acrn# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.5 LTS
Release:        20.04
Codename:       focal
```

liyimeng, 2022-09-13:
@fuzhongl I update it manually on my machine because other app need higher version.

liyimeng, 2022-09-13:
```
> 
> root@acrn-dev:/acrn-work/acrn-hypervisor# board_inspector.py myboad
> Generating board XML myboad. This may take a few minutes...
> 2022-09-13 07:36:23-root-ERROR:-Board inspector is running inside an unsupported Virtual Machine (VM). Only KVM or QEMU is supported. Unexpected results may occur.
> Traceback (most recent call last):
>   File "/usr/bin/board_inspector.py", line 240, in <module>
>     main(args.board_name, board_xml, args)
>   File "/usr/bin/board_inspector.py", line 191, in main
>     module.extract(args, board_etree)
>   File "/usr/bin/extractors/95-usb.py", line 27, in extract
>     with open(devpath + '/port/firmware_node/path') as f:
> FileNotFoundError: [Errno 2] No such file or directory: '/sys/bus/usb/devices/1-1/port/firmware_node/path'
> root@acrn-dev:/acrn-work/acrn-hypervisor# 
> 
> 
> With the error above, could it be a bug? My understanding is that not all device have separated firmware. Board inspector seem assume there is alway one, right?
```
Any comments on this?

fuzhongl, 2022-09-14:
> ```
> ```
>
> Any comments on this?

@liyimeng Please run `board_inspector.py` under native environment.

`> > 2022-09-13 07:36:23-root-ERROR:-Board inspector is running inside an unsupported Virtual Machine (VM). Only KVM or QEMU is supported. Unexpected results may occur.`