projectacrn/acrn-hypervisor #8135: VM configuration check file. [3.0 hybride mode]

lalala332, 2022-09-13:
**Describe the bug**
After using acrn-configurator to create the relevant xml and .sh files of hybrid mode according to Getting start, use acrn-hypervisor to compile and report an error: VM Configuration check fail. Please do help!

[scenario.txt](https://github.com/projectacrn/acrn-hypervisor/files/9554227/scenario.txt)
[my_board_dell.txt](https://github.com/projectacrn/acrn-hypervisor/files/9554228/my_board_dell.txt)

fuzhongl, 2022-09-19:
@NanlinXie Please help to check this build issue.
Thanks!

lalala332, 2022-09-19:
Hello，I solved this question. Maybe I can shared my detail of all the process.
First, due to the error message. I find the VM Configure file based on the error message. I see the prompt in the shell to “add reboot=acpi to the bootstrap of VM0”， so  I use **acrn-configurator to revise my config file by add "reboot=acpi" to the bootstrap of VM0.**
Then, the error message still there,  I see the line " prebuild error:255“ , so I check the MakeFile in misc/hv_prebuild.h finally find the funchtion "sanitize_vm_config", and check in main.c, "GUEST_FLAG_LAPIC_PASSTHROUGH must be set if we have GUEST_FLAG_RT set in guest_flags", so I **back to acrn-configurator and select "LAPIC_PASSTHROUGH" in VM0 advenced settings**. 
Then VM configure check pass.

gvancuts, 2022-09-19:
Thanks for sharing the solution @lalala332 . @NanlinXie @junjiemao1 that sounds like something which should be fixed in the ACRN Configurator directly, thoughts?

junjiemao1, 2022-09-20:
@gvancuts Yes, I think adding an assertion on that is sufficient to capture the report the error in the configurator.

@lalala332 The ACPI_VM0.bin can be found under the build directiry, i.e., `build/hypervisor/acpi/ACPI_VM0.bin`, instead of the source directory.