projectacrn/acrn-hypervisor #8178: Failed in eglCreateImageKHR when using  Nvidia GeForce

davidlamblin, 2022-09-20:
**Platform**
i7
lspci  -s 01:00.0 -v
01:00.0 VGA compatible controller: NVIDIA Corporation TU116 [GeForce GTX 1660 Ti] (rev a1) (prog-if 00 [VGA controller])
	Subsystem: Gigabyte Technology Co., Ltd TU116 [GeForce GTX 1660 Ti]
	Flags: bus master, fast devsel, latency 0, IRQ 152
	Memory at a3000000 (32-bit, non-prefetchable) [size=16M]
	Memory at 90000000 (64-bit, prefetchable) [size=256M]
	Memory at a0000000 (64-bit, prefetchable) [size=32M]
	I/O ports at 3000 [size=128]
	Expansion ROM at 000c0000 [disabled] [size=128K]
	Capabilities: [60] Power Management version 3
	Capabilities: [68] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Capabilities: [78] Express Legacy Endpoint, MSI 00
	Capabilities: [100] Virtual Channel
	Capabilities: [250] Latency Tolerance Reporting
	Capabilities: [258] L1 PM Substates
	Capabilities: [128] Power Budgeting <?>
	Capabilities: [420] Advanced Error Reporting
	Capabilities: [600] Vendor Specific Information: ID=0001 Rev=1 Len=024 <?>
	Capabilities: [900] Secondary PCI Express
	Capabilities: [bb0] Resizable BAR <?>
	Kernel driver in use: nvidia
	Kernel modules: nvidia_drm, nvidia

uname -r
5.15.44-acrn-service-vm-00001-gf4572566512e-dirty

modinfo nvidia
filename:       /lib/modules/5.15.44-acrn-service-vm-00001-gf4572566512e-dirty/kernel/drivers/video/nvidia.ko
firmware:       nvidia/515.65.01/gsp.bin
import_ns:      DMA_BUF
alias:          char-major-195-*
version:        515.65.01
supported:      external
license:        Dual MIT/GPL
srcversion:     EBA89555F4C275079F30609
alias:          pci:v000010DEd*sv*sd*bc06sc80i00*
alias:          pci:v000010DEd*sv*sd*bc03sc02i00*
alias:          pci:v000010DEd*sv*sd*bc03sc00i00*
depends:        
retpoline:      Y
name:           nvidia
vermagic:       5.15.44-acrn-service-vm-00001-gf4572566512e-dirty SMP preempt mod_unload modversions 

 modinfo nvidia-drm 
filename:       /lib/modules/5.15.44-acrn-service-vm-00001-gf4572566512e-dirty/kernel/drivers/video/nvidia-drm.ko
version:        515.65.01
supported:      external
license:        Dual MIT/GPL
srcversion:     002F7F9887A84FCCF254171
alias:          pci:v000010DEd*sv*sd*bc06sc80i00*
alias:          pci:v000010DEd*sv*sd*bc03sc02i00*
alias:          pci:v000010DEd*sv*sd*bc03sc00i00*
depends:        nvidia-modeset
retpoline:      Y
name:           nvidia_drm
vermagic:       5.15.44-acrn-service-vm-00001-gf4572566512e-dirty SMP preempt mod_unload modversions 
parm:           modeset:Enable atomic kernel modesetting (1 = enable, 0 = disable (default)) (bool)

glxinfo | grep version
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
OpenGL core profile version string: 4.6.0 NVIDIA 515.65.01
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL version string: 4.6.0 NVIDIA 515.65.01
OpenGL shading language version string: 4.60 NVIDIA
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 515.65.01
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
    GL_EXT_shader_group_vote, GL_EXT_shader_implicit_conversions, 

**Codebase**
git branch is  origin/release_3.1

**Scenario**
Describe the scenario: shared.

**Describe the bug**
vdpy_surface_set function call 
		egl_img = gl_ops->eglCreateImageKHR(vdpy.eglDisplay,
				EGL_NO_CONTEXT,
				EGL_LINUX_DMA_BUF_EXT,
				NULL, attrs); 
fail, and print  Failed in eglCreateImageKHR.

use eglGetError find that , errno is  EGL_BAD_ALLOC (0x3003)

1. using  https://gitlab.com/blaztinn/dma-buf-texture-sharing to test:
Failed  when modprobe nvidia-drm modeset=0
success  when modprobe nvidia-drm modeset=1
3. acrn-dm always fail  no matter "modprobe nvidia-drm modeset=1" or  "modprobe nvidia-drm modeset=0"
4. When  modprobe nvidia-drm modeset=1, there are some kernel errors as follow:
[ 7697.312839] NVRM osCreateOsDescriptorFromDmaBufPtr: osCreateOsDescriptorFromDmaBufPtr(): Error (86) while trying to import dma_buf!
[ 7697.313126] [drm:nv_drm_gem_export_nvkms_memory_ioctl [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to lookup NVKMS gem object for export: 0x00000001
[ 7697.313446] acrn_dm: Failed in eglCreateImageKHR.

gvancuts, 2022-09-20:
Can you share the launch script that you use here?

Please note that we do not support Nvidia graphics cards.

ywan170, 2022-09-21:
@yakuizhao @lifeix to help take a look.

yakuizhao, 2022-09-21:
@davidlamblin , Agree with what Gvancuts said. The Nvidia card is not covered by acrn-dm.
And the message log shows that the NV driver fails to create the EGLImage from the DMABuf-sharing.

davidlamblin, 2022-09-21:
The launch script is as follows.

dm_params=(
    `add_cpus                                 0 2 4 6 8 10 12 14`
    -m 4096M
    --ovmf /usr/share/acrn/bios/OVMF.fd
    `add_virtual_device                       1:0 lpc`
    `add_virtual_device                       0:0 hostbridge`
    `add_virtual_device                       3 xhci 1-1`
    `add_virtual_device                       4 virtio-input Device name: USB OPTICAL MOUSE, Device physical path: usb-0000:00:14.0-10/input0`
    `add_virtual_device                       5 virtio-console @stdio:stdio_port`
    `add_virtual_device                       6 virtio-blk /home/acrn-work/ubuntu_uservm.img`
    `add_virtual_device                       7 virtio-gpu geometry=800x600+0+0`
    `add_logger_settings                      console=4 kmsg=3 disk=5`
    VM1
)

yakuizhao, 2022-09-21:
From the log it seems that it tries to start the virtio-gpu on NV card. And this is not supported by ACRN-DM.
At the same time the issue in log shows that it is related with NV driver.
If the NV card is really needed, please use the passthru.

closed 2022-09-28 by fuzhongl