projectacrn/acrn-hypervisor #8202: [HV][qemu] [v3.1]HV fail to boot acrn on qemu

fuzhongl, 2022-09-23:
**Describe the bug**
ACRN on qemu can't boot up with v3.1

**Platform**
ADL

**Codebase**
v3.1

Check out v3.1
commit d459990443505430e77448c3c850fed6677876fb (HEAD -> rel_3.1, origin/release_3.1)

Multiboot Bootloader: GRUB 2.02-2ubuntu8.23
ACRN Hypervisor
[1333886028814us][cpu=0][(null)][sev=2][seq=1]:calibrate_tsc: tsc_khz = 421
[1333891648446us][cpu=0][(null)][sev=2][seq=2]:HV version 3.1-unstable-acrn-2022ww36.5-180000p-1-gd45999044-dirty-2022-09-05 05:35:05-d45999044-dirty DBG (daily tag:acrn-2022ww36.5-180000p) shared@qemu build by root, start time 1333838273327us
[1333909305059us][cpu=0][(null)][sev=2][seq=3]:API version 1.0
[1333913975054us][cpu=0][(null)][sev=2][seq=4]:Detect processor: Intel Atom Processor (Denverton)
[1333921099973us][cpu=0][(null)][sev=1][seq=5]:SECURITY WARNING!!!!!!
[1333926233928us][cpu=0][(null)][sev=1][seq=6]:Please apply the latest CPU uCode patch!
[1334843038771us][cpu=1][(null)][sev=4][seq=7]:Skipping VM configuration check which should be done before building HV binary.
[1334844702574us][cpu=0][(null)][sev=1][seq=8]:Secondary CPU2 failed to come up
[1334845123133us][cpu=2][(null)][sev=4][seq=9]:Skipping VM configuration check which should be done before building HV binary.
[1334864162560us][cpu=3][(null)][sev=4][seq=10]:Skipping VM configuration check which should be done before building HV binary.

[Debug info]
ACRN on qemu can boot up if revert following patch:
commit 97a2919138def45c3b728988189de56b6d79a1d6

Date: Tue Jul 5 15:36:41 2022 +0800

hv: tsc: calibrate TSC by HPET

time-river, 2022-12-22:
I have the same problem with U: acrn v3.1 can't boot on qemu.

Here is my qemu boot script:
```
qemu-system-x86_64 \
	-name acrn \
	-drive file=/usr/share/OVMF/OVMF_CODE.fd,if=pflash,format=raw,unit=0,readonly=on \
	-smbios type=0,uefi=on	\
	-cpu host,kvmclock=off,+invtsc	\
	-rtc base=localtime,clock=host	\
	-machine q35	\
	-accel kvm,kernel-irqchip=split \
	-smp 1 \
	-m 10240 \
	-drive if=none,file=ubuntu-22.04-minimal-cloudimg-amd64.qcow2,format=qcow2,id=rootfs	\
	-device virtio-blk-pci,drive=rootfs,bootindex=1	\
	-netdev bridge,id=net0,br=acrn-br0 \
	-device virtio-net-pci,netdev=net0,id=net0,mac=52:54:00:c9:2d:4f \
	-object rng-random,filename=/dev/hwrng,id=rng0 \
	-device virtio-rng-pci,rng=rng0	\
	-serial stdio	\
	-device virtio-vga-gl	\
	-device qemu-xhci,id=usb0	\
	-device usb-kbd	\
	-device usb-mouse	\
	-device usb-tablet	\
	-device intel-iommu,intremap=on,caching-mode=on,aw-bits=48	\
	-drive if=none,file=vda.qcow2,format=qcow2,id=vda	\
	-device virtio-blk-pci,drive=vda,bootindex=2	\
	-drive if=none,file=vdb.qcow2,format=qcow2,id=vdb	\
	-device virtio-blk-pci,drive=vdb,bootindex=3	\
	-drive if=none,file=vdc.qcow2,format=qcow2,id=vdc	\
	-device virtio-blk-pci,drive=vdc,bootindex=4	\
	-drive if=none,file=vdd.qcow2,format=qcow2,id=vdd	\
	-device virtio-blk-pci,drive=vdd,bootindex=5	\
	-device virtio-balloon-pci,id=balloon0	\
	-display sdl,gl=on	\
	-gdb tcp::1234	\
	-monitor telnet::1235,server,nowait
```

But the different is that I revert that commit, it still can't boot. @fuzhongl

time-river, 2023-01-05:
#8132 handles the problem, the same as #8113