projectacrn/acrn-hypervisor #8229: TPM passthrough not recognized in windows guest

johnmave126, 2022-10-13:
**Describe the bug**
Platform TPM is not recognized in windows guest when passing through via `--acpidev_pt MSFT0101`

**Platform**
Intel RNUC12WSHi50000

**Codebase**
acrn-hypervisor tag v3.1
acrn-kernel tag acrn-v3.1
acrn-edk2 tag ovmf-acrn-v3.0

**Scenario**
Windows VM is a post-launched user VM with GPU passthrough. OVMF is built with GOP driver ver. 21.0.1054 and vbt extracted from `/sys/kernel/debug/dri/0/i915_vbt`. After launching windows, TPM cannot be found in device manager.

**To Reproduce**
Follow the steps in the documentation to make a windows guest vm and custom OVMF. Modify the launch script and add `--acpidev_pt MSFT0101` to `acrn-dm`.

**Expected behavior**
TPM device found in device manager of windows.

**Additional context**
Attach the following logs if available:
<details>
<summary>acrn-dm log</summary>

```
[2022-10-13 01:37:14][   36.697680] Found default iasl path: /usr/sbin/iasl
[2022-10-13 01:37:14][   36.700580] iasl version: 20210105
[2022-10-13 01:37:14][   36.700775] vm_create: Windows
[2022-10-13 01:37:14][   36.701392] vm_setup_memory: size=0x600000000
[2022-10-13 01:37:14][   36.701435] level 0 free/need pages:0/9 page size:0x200000
[2022-10-13 01:37:14][   36.701451] level 1 free/need pages:28/24 page size:0x40000000
[2022-10-13 01:37:14][   36.701455] to reserve more free pages:
[2022-10-13 01:37:14][   36.702099] to reserve pages (+orig 0): echo 9 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages
[2022-10-13 01:37:14][   36.702119] now enough free pages are reserved!
[2022-10-13 01:37:14][   36.702123] 
try to setup hugepage with:
[2022-10-13 01:37:14][   36.702127]     level 0 - lowmem 0x0, fbmem 0x1000000, biosmem 0x200000, highmem 0x0
[2022-10-13 01:37:14][   36.702132]     level 1 - lowmem 0x80000000, fbmem 0x0, biosmem 0x0, highmem 0x580000000
[2022-10-13 01:37:14][   36.702135] total_size 0x6c0000000

[2022-10-13 01:37:14][   36.702142] mmap ptr 0x0x7eff78f8b000 -> baseaddr 0x0x7eff80000000
[2022-10-13 01:37:14][   36.702155] mmap 0x80000000@0x7eff80000000
[2022-10-13 01:37:14][   36.702159] touch 2 pages with pagesz 0x40000000
[2022-10-13 01:37:14][   36.878293] mmap 0x580000000@0x7f0080000000
[2022-10-13 01:37:14][   36.878315] touch 22 pages with pagesz 0x40000000
[2022-10-13 01:37:16][   38.780485] mmap 0x200000@0x7f007fe00000
[2022-10-13 01:37:16][   38.780502] touch 1 pages with pagesz 0x200000
[2022-10-13 01:37:16][   38.780708] mmap 0x1000000@0x7f007ee00000
[2022-10-13 01:37:16][   38.780716] touch 8 pages with pagesz 0x200000
[2022-10-13 01:37:16][   38.782137] 
really setup hugepage with:
[2022-10-13 01:37:16][   38.782151]     level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
[2022-10-13 01:37:16][   38.782156]     level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x580000000
[2022-10-13 01:37:16][   38.813353] vm_init_vdevs
[2022-10-13 01:37:16][   38.814226] No pm notify channel given
[2022-10-13 01:37:16][   38.814538] start monitor interrupt data...
[2022-10-13 01:37:16][   38.814767] mmiodev[0] hpa:0xfed40000 gpa:0xfed40000 size:0x5000 err:0
[2022-10-13 01:37:16][   38.816281] pci init passthru
[2022-10-13 01:37:16][   38.926328] modify_bar_registration: bypass for pci-passthru 0:2.0
[2022-10-13 01:37:16][   38.926344] modify_bar_registration: bypass for pci-passthru 0:2.0
[2022-10-13 01:37:16][   38.926363] modify_bar_registration: bypass for pci-passthru 0:2.0
[2022-10-13 01:37:16][   38.926680] pci init passthru
[2022-10-13 01:37:16][   39.030251] modify_bar_registration: bypass for pci-passthru 0:8.0
[2022-10-13 01:37:16][   39.030305] pci init passthru
[2022-10-13 01:37:16][   39.030482] MSI not supported for PCI 0/a/0[2022-10-13 01:37:16][   39.030517] modify_bar_registration: bypass for pci-passthru 0:a.0
[2022-10-13 01:37:16][   39.030569] pci init passthru
[2022-10-13 01:37:16][   39.030852] modify_bar_registration: bypass for pci-passthru 0:d.0
[2022-10-13 01:37:16][   39.030904] pci init passthru
[2022-10-13 01:37:16][   39.031140] modify_bar_registration: bypass for pci-passthru 0:d.2
[... 66 lines cut ...]
Listening 0...
polling 37...
mmiodev[0] hpa:0xfed40000 gpa:0xfed40000 size:0x5000 err:0
pci init passthru
pci init passthru
pci init passthru
pci init passthru
pci init passthru
pci init passthru
pci init passthru
pci init passthru
pci init passthru
pci init passthru
pci init hostbridge
pci init lpc
pci init virtio-net
pci init virtio-blk
pci init igd-lpc
ACRN_IOCTL_SET_PTDEV_INTR ioctl() returned an error: Operation not permitted
ACRN_IOCTL_SET_PTDEV_INTR ioctl() returned an error: Operation not permitted
tpm: init_vtpm2:Invalid socket path!
acrn_sw_load
add_cpu
```
</details>

johnmave126, 2022-10-19:
Tried to compile acrn-edk2 with additional `-DTPM2_ENABLE`, now user VM fails to boot.

weichenli-intel, 2022-10-27:
Pls check your board xml and find the element of TPM:

<device id="xxxxxxx" description="TPM 2.0 Device">

You could set the device id "xxxxxx" as the HID parameter of acpidev_pt.

fuzhongl, 2022-11-02:
@johnmave126 
To find the HID of TPM, please run following command in Service VM:
$sudo cp /sys/firmware/acpi/tables/SSDT* ./

$sudo iasl -d SSDT*

$  grep -irn  -C3 "TPM 2.0" --include=*.dsl

Following log is the output on my side:
$ grep -irn  -C3 "TPM 2.0" --include=*.dsl
SSDT4.dsl-25-        Device (TPM)
SSDT4.dsl-26-        {
SSDT4.dsl-27-            Name (_HID, "**INTC6001**")  // _HID: Hardware ID
SSDT4.dsl:28:            Name (_CID, "MSFT0101" /* TPM 2.0 Security Device */)  // _CID: Compatible ID
SSDT4.dsl:29:            Name (_STR, Unicode ("TPM 2.0 Device"))  // _STR: Description String
SSDT4.dsl-30-            OperationRegion (SMIP, SystemIO, 0xB2, One)
SSDT4.dsl-31-            Field (SMIP, ByteAcc, NoLock, Preserve)
SSDT4.dsl-32-            {

In this case, INTC6001 is the HID of TPM.

johnmave126, 2022-11-03:
@fuzhongl 

Here is what found
```
   Device (_SB.TPM)
    {
        Name (TMRQ, 0x00000000)
        Name (TLVL, 0xFFFFFFFF)
        Method (_HID, 0, NotSerialized)  // _HID: Hardware ID
        {
            If (TCMF)
            {
                Return (0x01013469)
            }
            ElseIf ((TTDP == Zero))
            {
                Return (0x310CD041)
            }
            Else
            {
                Return ("MSFT0101")
            }
        }

        OperationRegion (TMMB, SystemMemory, 0xFED40000, 0x5000)
        Field (TMMB, ByteAcc, Lock, Preserve)
        {
```

And `grep -i "tpm"` could **not** find anything within SSDT, this is found in DSDT.

johnmave126, 2022-11-03:
TPM shows up in board.xml as
```xml
      <device id="MSFT0101" description="TPM 2.0 Device">
        <acpi_object>\_SB_.TPM_</acpi_object>
        <acpi_uid>1</acpi_uid>
        <aml_template>5b8246065c2e5f53425f54504d5f085f5354410a0f085f4849440d4d5346543031303100085f55494401085f53545211210a1e540050004d00200032002e00300020004400650076006900630065000000085f43525311110a0e860900010000d4fe005000007900</aml_template>
        <status>
          <present>y</present>
          <enabled>y</enabled>
          <functioning>y</functioning>
        </status>
        <resource id="res0" type="memory" min="0xfed40000" max="0xfed44fff" len="0x5000"/>
        <capability id="control_area">
          <address_of_control_area>0xfed40040</address_of_control_area>
        </capability>
        <capability id="start_method">
          <value>0x7</value>
          <parameter>0x0</parameter>
          <parameter>0x0</parameter>
          <parameter>0x0</parameter>
          <parameter>0x0</parameter>
          <parameter>0x0</parameter>
          <parameter>0x0</parameter>
          <parameter>0x0</parameter>
          <parameter>0x0</parameter>
          <parameter>0x0</parameter>
          <parameter>0x0</parameter>
          <parameter>0x0</parameter>
          <parameter>0x0</parameter>
        </capability>
        <capability id="log_area">
          <log_area_minimum_length>0x10000</log_area_minimum_length>
          <log_area_start_address>0x56dd4000</log_area_start_address>
        </capability>
      </device>
```

fuzhongl, 2022-11-04:
@johnmave126 Should you please share the whole board.xml?
Which kernel are you using when generate the board.xml? 
Thanks!

johnmave126, 2022-11-04:
@fuzhongl 
Actually I just compiled a DEBUG OVMF and inspect its serial output. It seems that UEFI does recognize TPM2 device on boot, but with warnings like
```
WARNING: TPM2 Event log has HashAlg unsupported by PCR bank (0x4)
WARNING: TPM2 Event log has HashAlg unsupported by PCR bank (0xC)
```
I think this may more likely be a Windows bug. I will try to spin up a Linux VM to see whether Linux is able to recognize it.

johnmave126, 2022-11-04:
OK, `tpm2-tools` on a Linux VM is able to read and extend TPM digests just fine. I now believe this is a Windows restriction.

closed 2022-11-04 by johnmave126