projectacrn/acrn-hypervisor #8323: User VM fail to get ip after configurating vhost

liudlong, 2022-11-17:
HW/Board: adl-s

Commit:  4e77e44542a63f4f85cb3c4e530efbdf308074ce (master)

Step:

build acrn with the default XML file (board xml: https://github.com/intel-innersource/virtualization.hypervisors.acrn.acrn-dev.acrn-offline-patch/blob/master/boards/adl-s-crb/adl-s-crb.xml scenario xml: https://github.com/intel-innersource/virtualization.hypervisors.acrn.acrn-dev.acrn-offline-patch/blob/master/boards/adl-s-crb/shared_launch_6user_vm_gpu_sriov.xml)
generate the launch script for post_rt_vm with config tool 
boot acrn, and launch RTVM.
no IP address in RTVM.
Expected result:

User VM can get IP.

Actual result:

Error message is found when launched RTVM.

out instr on NMI port (0x61) not supported
out instr on NMI port (0x61) not supported
out instr on NMI port (0x61) not supported
ACRN_IOCTL_INJECT_MSI ioctl() returned an error: Operation not permitted
ACRN_IOCTL_INJECT_MSI ioctl() returned an error: Operation not permitted
ACRN_IOCTL_INJECT_MSI ioctl() returned an error: Operation not permitted
ACRN_IOCTL_INJECT_MSI ioctl() returned an error: Operation not permitted
ACRN_IOCTL_INJECT_MSI ioctl() returned an error: Operation not permitted
ACRN_IOCTL_INJECT_MSI ioctl() returned an error: Operation not permitted
ACRN_IOCTL_INJECT_MSI ioctl() returned an error: Operation not permitted
ACRN_IOCTL_INJECT_MSI ioctl() returned an error: Operation not permitted
ACRN_IOCTL_INJECT_MSI ioctl() returned an error: Operation not permitted
[ioeventfd: 0][0x3010@2][flags: 0xb][data: 0x0]
ACRN_IOCTL_IOEVENTFD ioctl() returned an error: Invalid argument
vm_ioeventfd failed rc = -1, errno = 22
[ioeventfd: 0][0x3010@2][flags: 0xb][data: 0x1]
ACRN_IOCTL_IOEVENTFD ioctl() returned an error: Invalid argument
vm_ioeventfd failed rc = -1, errno = 22
ACRN_IOCTL_INJECT_MSI ioctl() returned an error: Operation not permitted
ACRN_IOCTL_INJECT_MSI ioctl() returned an error: Operation not permitted
ACRN_IOCTL_INJECT_MSI ioctl() returned an error: Operation not permitted

Reproduce rate:

100%

liudlong, 2022-11-17:
[External_System_ID] ACRN-9456

huongnguyenthi, 2025-05-30:
Hello,
I'm encountering a similar issue when trying to run a real-time VM. The VM fails to display the login screen, and the following messages appear in the console:
<pre>
out instr on NMI port (0x61) not supported
out instr on NMI port (0x61) not supported
out instr on NMI port (0x61) not supported
ACRN_IOCTL_SET_IRQLINE ioctl() returned an error: Operation not permitted
ACRN_IOCTL_INJECT_MSI ioctl() returned an error: Operation not permitted
ACRN_IOCTL_INJECT_MSI ioctl() returned an error: Operation not permitted
ACRN_IOCTL_INJECT_MSI ioctl() returned an error: Operation not permitted
ACRN_IOCTL_INJECT_MSI ioctl() returned an error: Operation not permitted
</pre>

Has anyone run into this before or have any suggestions on how to resolve it?

[launch-vm-error.log](https://github.com/user-attachments/files/20516904/launch-vm-error.log)

Please check the full log attached for more details.

Thanks in advance!