projectacrn/acrn-hypervisor #8360: Cannot run acrn-configurator on Debian Bullseye

tttech-industrial-buchsbaum, 2022-12-09:
**Describe the bug**
I have to use Debian Bullseye as development environment to build and configure ACRN. I installed the acrn-configurator according to GSG (https://projectacrn.github.io/latest/getting-started/getting-started.html#generate-a-scenario-configuration-file-and-launch-script). When running the acrn-configurator I get the following error message:

    user@host:~/acrn-work$ acrn-configurator 
    acrn-configurator: error while loading shared libraries: libssl.so.3: cannot open shared object file: No such file or directory

**Platform**
Standard Debian Bullseye installation on HP EliteBook laptop.

**Codebase**
https://github.com/projectacrn/acrn-hypervisor/releases/download/v3.1/acrn-configurator-3.2-unstable.deb

**Scenario**
not applicable

**To Reproduce**
Steps to reproduce the behavior on a Debian Bullseye host:
1. Eventually remove a previously installed version of acrn-configurator:

        sudo apt purge acrn-configurator
3. Download the latest Debian package of acrn-configurator:

        wget https://github.com/projectacrn/acrn-hypervisor/releases/download/v3.1/acrn-configurator-3.2-unstable.deb -P /tmp
4. Install it:

        sudo apt install -y /tmp/acrn-configurator-3.2-unstable.deb
5. Running it produce the error message mentioned above!

**Expected behavior**
acrn-configurator starts up and works as expected.

**Additional context**
There are in fact two problems:
1. The provided Debian package is missing the dependency  on openssl >= 3. WIth correct dependencies already the package install would have failed. Please also re-check all other runtime dependencies that might also be missing. Th e debian/control file of the package only lists libwebkit2gtk-4.0-37, libgtk-3-0, but the binary has a lot of dependencies, so I doubt this can be correct:

        user@host:~/acrn-work$ ldd /usr/bin/acrn-configurator  | wc -l
        144
2. No idea where this openssl dependency comes from. Is it really required by acrn-configurator? I am not aware of any crypt related functions for configuring an ACRN scenario! If openssl is really required, I ask for providing a openssl-2 based package which then can be used e.g. on Debian Bullseye. Right now, creating some Ubuntu VM just for running acrn-configurator is not the way I expect configuration to work! BTW, I could not find any Debian source package for acrn-configurator, otherwise I would have tried to rebuild my own Bullseye package ;-)

NanlinXie, 2022-12-12:
acrn-configurator-3.2-unstable.deb was built from ubuntu 22.04, maybe caused your current issue, suggest to rebuild on your own Bullseye, below doc is for your reference. 
https://projectacrn.github.io/latest/tutorials/acrn_configurator_tool.html#build-acrn-configurator-from-source-code

gvancuts, 2022-12-12:
I think the main point of this bug is the fact the `.deb` is not correctly indicating what the dependencies are, this is what needs to be fixed (and a good look at those to make they are legit would be good too).