projectacrn/acrn-hypervisor #8380: Cannot generate xml file for target system

MaiM1ng, 2023-04-12:
**Describe the bug**
A bug was encountered while going through the manual steps to create the board xml file.
When I execute the following command
```Bash
sudo acrn-board-inspector my_board
```
error 
```Bash
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/usr/lib/python3/dist-packages/acrn_board_inspector/board_inspector.py", line 240, in <module>
    main(args.board_name, board_xml, args)
  File "/usr/lib/python3/dist-packages/acrn_board_inspector/board_inspector.py", line 186, in main
    module.extract(args, board_etree)
  File "/usr/lib/python3/dist-packages/acrn_board_inspector/extractors/10-processors.py", line 174, in extract
    extract_topology(processors_node)
  File "/usr/lib/python3/dist-packages/acrn_board_inspector/extractors/10-processors.py", line 128, in extract_topology
    extract_model(processors_node, cpu_id, family_id, model_id, core_type, native_model_id)
  File "/usr/lib/python3/dist-packages/acrn_board_inspector/extractors/10-processors.py", line 75, in extract_model
    msr_data = msr_reg.rdmsr(cpu_id)
  File "/usr/lib/python3/dist-packages/acrn_board_inspector/cpuparser/platformbase.py", line 144, in rdmsr
    r = msr_reader.read(8)
OSError: [Errno 5] Input/output error
```

**Platform**
NUC6CAYH

**Expected behavior**
https://projectacrn.github.io/latest/getting-started/getting-started.html
This document should work just fine

NanlinXie, 2023-04-13:
It’s likely that the TURBO_RATIO_LIMIT and TURBO_ACTIVATION_RATIO MSRs are not present on your Apollo Lake platform. .

Unfortunately there is no other way to work that around without code changes. Possible solution is to capture the IOError exception in extract_model() in 10-processors.py to gracefully handle such cases.

MaiM1ng, 2023-04-13:
> It’s likely that the TURBO_RATIO_LIMIT and TURBO_ACTIVATION_RATIO MSRs are not present on your Apollo Lake platform. .
>
> Unfortunately there is no other way to work that around without code changes. [...]
@NanlinXie ，Hello，Is Apollo lake not supported in the latest ACRN？Can you tell me which version of acrn supports apollo lake？thanks

closed 2023-04-18 by MaiM1ng

Kunhui-Li, 2023-05-04:
[External_System_ID] ACRN-9917