projectacrn/acrn-hypervisor #8413: hypervisor: 'vm_config' may be used uninitialized [-Werror=maybe-uninitialized]

saininav, 2023-05-29:
| In function 'partition_epc',
|     inlined from 'init_sgx' at arch/x86/sgx.c:116:19:
| arch/x86/sgx.c:86:91: error: 'vm_config' may be used uninitialized [-Werror=maybe-uninitialized]
|    86 |                         vm_epc_maps[mid][vm_id].gpa = vm_config->epc.base + vm_config->epc.size - vm_request_size;
|       |                                                                             ~~~~~~~~~~~~~~^~~~~
| arch/x86/sgx.c: In function 'init_sgx':
| arch/x86/sgx.c:61:32: note: 'vm_config' was declared here
|    61 |         struct acrn_vm_config *vm_config;
|       |                                ^~~~~~~~~
| cc1: all warnings being treated as errors

This is reported with GCC-13.

jiaqingz-intel, 2023-06-01:
[External_System_ID] ACRN-10120

jiaqingz-intel, 2023-06-07:
Fixed in [e4e25f0](https://github.com/projectacrn/acrn-hypervisor/commit/e4e25f076ca374004de9383a89b76c308f71c809)

NanlinXie, 2023-06-07:
@saininav , please help to check merged fix and close this issue, thanks

saininav, 2023-06-13:
It fixes the issue. Any plans to backport in release_3.1 branch ?

saininav, 2023-06-14:
This issue has been fixed in release_3.2.

closed 2023-06-14 by saininav