projectacrn/acrn-hypervisor #8443: Run Debian_ Build_ SH script error

H-chun, 2023-07-05:
**Describe the bug**
A clear and concise description of what the bug is.

**Platform**
Describe the platform information, 

**Codebase**
Describe which codebase is used.

**Scenario**
Describe the scenario: Hybrid, Industry or logical_partition

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Attach the following logs if available:

- native Linux boot kernel message log
- ACRN serial port log (when issues happens)
- Service VM kernel message when issue happens
- acrn-dm log: all files in /var/log/acrn-tm
- console log when running script to launch guest VM

Hello, when I was in the process of building ACRN and reached Debian/Debian_ Build.sh clean&&debian/debian_ The following error occurred in build.sh - c~/acrn work/MyConfiguration. May I ask what the problem is and what I should do? Thank you

CONFIGDIRS        = 'misc/config_tools/data'
gbp:info: Changelog last touched at 'ed9cc9409a49c762273775314ae4fdcfff56543d'
gbp:info: Continuing from commit 'ed9cc9409a49c762273775314ae4fdcfff56543d'
gbp:info: Changelog unknown~1.gbpfeb1af (snapshot #1) prepared up to feb1afb
debuild -eCONFIGDIRS='misc/config_tools/data' -- clean
 dpkg-buildpackage --rules-target clean -us -uc -ui
dpkg-buildpackage: info: 源码包 acrn-hypervisor
dpkg-buildpackage: info: 源码版本 3.2-unstable~1.gbpfeb1af
dpkg-buildpackage: info: source distribution UNRELEASED
dpkg-buildpackage: info: 源码修改者 123 <[projectacrn@gmail.com](mailto:projectacrn@gmail.com)>
 fakeroot debian/rules clean
   debian/rules override_dh_auto_clean
Building ACRN 3.2 for boards: adl-asrock, cfl-k700-i7, generic_board, nuc11tnbi5, qemu, tgl-vecow-spc-7100-Corei7, whl-ipc-i5
The scenarios used are:
  adl-asrock: shared
  cfl-k700-i7: hybrid, hybrid_launch_2user_vm, hybrid_rt, partitioned, shared, shared_launch_6user_vm
  generic_board: hybrid, partitioned, shared
  nuc11tnbi5: hybrid, hybrid_launch_2user_vm, partitioned, shared, shared_launch_6user_vm
  qemu: shared
  tgl-vecow-spc-7100-Corei7: hybrid, partitioned, shared, shared_launch_2user_vm
  whl-ipc-i5: hybrid, hybrid_rt, hybrid_rt_launch_1user_vm_waag, partitioned, shared, shared_launch_1user_vm_hardrt, shared_launch_1user_vm_vxworks, shared_launch_1user_vm_waag, shared_launch_2user_vm, shared_launch_6user_vm
   override_dh_auto_clean: CLEAN
   dh_clean
/home/abc/acrn-work/acrn-hypervisor
CONFIGDIRS        = '/home/abc/acrn-work/MyConfiguration'
gbp:info: Changelog last touched at 'ed9cc9409a49c762273775314ae4fdcfff56543d'
gbp:info: Continuing from commit 'ed9cc9409a49c762273775314ae4fdcfff56543d'
gbp:info: Changelog unknown~1.gbpfeb1af (snapshot #1) prepared up to feb1afb
debuild -eCONFIGDIRS='/home/abc/acrn-work/MyConfiguration' -- binary
 dpkg-buildpackage --rules-target binary -us -uc -ui
dpkg-buildpackage: info: 源码包 acrn-hypervisor
dpkg-buildpackage: info: 源码版本 3.2-unstable~1.gbpfeb1af
dpkg-buildpackage: info: source distribution UNRELEASED
dpkg-buildpackage: info: 源码修改者 123 <[projectacrn@gmail.com](mailto:projectacrn@gmail.com)>
dpkg-buildpackage: info: 主机架构 amd64
 fakeroot debian/rules binary
   dh_update_autotools_config
   dh_autoreconf
   dh_auto_configure
   debian/rules override_dh_auto_build-arch
   override_dh_auto_build-arch: BUILD life_mngr
WARN: x86_64-w64-mingw32-gcc not installed, skipping life_mngr_win.exe
   debian/rules override_dh_auto_build-indep
   build/documentation/html/index.html: BUILD documentation
   override_dh_auto_build-indep: BUILD acrn-board-inspector
I: pybuild base:239: /usr/bin/python3 setup.py build 
package init file '../../misc/config_tools/board_inspector/__init__.py' not found (or not a regular file)
package init file '../../misc/config_tools/board_inspector/acpiparser/aml/__init__.py' not found (or not a regular file)
package init file '../../misc/config_tools/board_inspector/extractors/__init__.py' not found (or not a regular file)
package init file '../../misc/config_tools/board_inspector/inspectorlib/__init__.py' not found (or not a regular file)
package init file '../../misc/config_tools/board_inspector/legacy/__init__.py' not found (or not a regular file)
package init file '../../misc/config_tools/board_inspector/schema/__init__.py' not found (or not a regular file)
   dh_auto_test
   create-stamp debian/debhelper-build-stamp
   dh_testroot
   dh_prep
   dh_installdirs
   debian/rules override_dh_auto_install-arch
   debian/acrn-hypervisor.install: CREATE debian/acrn-hypervisor.install
   override_dh_auto_install-arch: BUILD & INSTALL hypervisor for my_board:MyConfiguration
scripts/makefile/[config.mk:191](http://config.mk:191/): The command line sets RELEASE to be 'n', but an existing build is configured with 'y'
scripts/makefile/[config.mk:191](http://config.mk:191/): The configuration will be modified for RELEASE=n
ERROR:root:/acrn-config/hv/BUILD_TYPE: Select a hypervisor serial console port when Build type is Debug, or use Build type Release if no serial console is available.
ERROR:root:Board my_board and scenario MyConfiguration are inconsistent: 0 syntax errors, 1 data errors, 0 warnings.
make[3]: *** 没有规则可制作目标“/home/abc/acrn-work/acrn-hypervisor/build/my_board/MyConfiguration/hypervisor/configs/.validation.timestamp”，由“/home/abc/acrn-work/acrn-hypervisor/build/my_board/MyConfiguration/hypervisor/configs/.configfiles.timestamp” 需求。 停止。
make[2]: *** [Makefile:126：hypervisor] 错误 2
make[1]: *** [debian/rules:260：override_dh_auto_install-arch] 错误 2
make: *** [debian/rules:411：binary] 错误 2
dpkg-buildpackage: 错误: fakeroot debian/rules binary subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage --rules-target binary -us -uc -ui failed

jiaqingz-intel, 2023-07-06:
`SERIAL_CONSOLE` must be specified in scenario file if you are building a debug build. Please upload your board.xml and scenario.xml for details.

H-chun, 2023-07-06:
[my_board.zip](https://github.com/projectacrn/acrn-hypervisor/files/11963632/my_board.zip)
[scenario.zip](https://github.com/projectacrn/acrn-hypervisor/files/11963633/scenario.zip)
Hello, could you help me take a look? Thank you

H-chun, 2023-07-06:
Hello, can you help me take a look when I have time

lonnietc, 2024-10-27:
@H-chun Hello,  I am exploring ACRN on an Intel NUC7 (target system) and Intel NUC13 (build system with Ubuntu 22.04) and have run into this same effort.

Were you ever able to find a solution?

thanks in advance